coolsnowwolf/lede #12084: 有没有openwrt 23.05.03release版本的源码呢？

lysgwl, 2024-04-19:
### 详细叙述

openwrt官方发布了，openwrt23.05.3的release版本，coolsnowwolf有没有提供23.05的release版本，看了tag，最新版本是20230609的版本。

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

Rockchip r2s

### 详细日志

关于openwrt源码release版本的提供。

WYC-2020, 2024-04-19:
什么玩意，lede 怎么可能会有openwrt那边的分支，需要自己去 openwrt官方可弄就是了，你怕是lede和openwrt的区别都不知道吧
https://github.com/openwrt/openwrt 去这里找

lysgwl, 2024-04-19:
> 什么玩意，lede 怎么可能会有openwrt那边的分支，需要自己去 openwrt官方可弄就是了，你怕是lede和openwrt的区别都不知道吧 https://github.com/openwrt/openwrt 去这里找

真不知道啊，给普及下。你知道吗？

caixin1218, 2024-04-19:
> > 什么玩意，lede 怎么可能会有openwrt那边的分支，需要自己去 openwrt官方可弄就是了，你怕是lede和openwrt的区别都不知道吧 https://github.com/openwrt/openwrt 去这里找
> 
> 真不知道啊，给普及下。你知道吗？
git clone https://github.com/openwrt/openwrt.git  openwrt
cd openwrt
git checkout v23.05.3
git pull 
./scripts/feeds update -a   
./scripts/feeds install -a
git pull 
./scripts/feeds update -a   
./scripts/feeds install -a
make menuconfig  #第一次编译尽量保持默认
make download -j8 V=s
make download -j1 V=s 
make -j$(nproc) V=s
按照这个一条条执行，应该可以.

lysgwl, 2024-04-19:
> 什么玩意，lede 怎么可能会有openwrt那边的分支，需要自己去 openwrt官方可弄就是了，你怕是lede和openwrt的区别都不知道吧 https://github.com/openwrt/openwrt 去这里找
>
>
> 真不知道啊，给普及下。你知道吗？
> git clone https://github.com/openwrt/openwrt.git  openwrt
> cd [...]

谢谢，不过你理解错我意思了。coolsnowwolf代码也是基于openwrt主线版本，意思是官方发布了23.05.3 release版本，coolsnowwolf有没有类似的23.05.3tag呢？

caixin1218, 2024-04-20:
> 谢谢，不过你理解错我意思了。coolsnowwolf代码也是基于openwrt主线版本，意思是官方发布了23.05.3 release版本，coolsnowwolf有没有类似的23.05.3tag呢？

这是两不一样的东西，但都是基于openwrt内核，Lede的app更丰富，功能更强大。如果Lede有23.05.3 release版本  ,那还要官方的做什么呢。除了Lede，还有很大佬的版本，都有各自的tag

lysgwl, 2024-04-20:
> 这是两不一样的东西，但都是基于openwrt内核，Lede的app更丰富，功能更强大。如果Lede有23.05.3 release版本 ,那还要官方的做什么呢。除了Lede，还有很大佬的版本，都有各自的tag

额，明白了，就是基于最新内核的开发版。Alpha版本，是这个意思吧，没理解错吧。

dukmonki, 2024-04-20:
建议大佬搞两个版本，一个是现在这个 lede,天天不断追新。一个基于V23稳定版本V23，几个月更新一下，

caixin1218, 2024-04-20:
> 建议大佬搞两个版本，一个是现在这个 lede,天天不断追新。一个基于V23稳定版本V23，几个月更新一下，

那不多此一举，要V23 直接去官网git

WYC-2020, 2024-04-22:
> > 什么玩意，lede 怎么可能会有openwrt那边的分支，需要自己去 openwrt官方可弄就是了，你怕是lede和openwrt的区别都不知道吧 https://github.com/openwrt/openwrt 去这里找
> 
> 真不知道啊，给普及下。你知道吗？

不知道自己去看esir的前世今生，别人说的很清楚

closed 2024-04-22 by lysgwl