coolsnowwolf/lede #12086: ERROR: igmpproxy

ywt114, 2024-04-20:
### 详细叙述

ERROR: package/feeds/packages/igmpproxy failed to build.
automake-1.15: command not found.
现在的automake版本是1.16，是否只能通过创建一个软连接解决？
sudo ln -s /usr/bin/automake /usr/bin/automake-1.15

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

2024-04-19T20:59:09.8738547Z make[3]: Entering directory '/home/runner/work/test/test/openwrt/feeds/packages/net/igmpproxy'
2024-04-19T20:59:10.0271798Z rm -f /home/runner/work/test/test/openwrt/build_dir/target-x86_64_musl/igmpproxy-0.2.1/.built
2024-04-19T20:59:10.0300672Z touch /home/runner/work/test/test/openwrt/build_dir/target-x86_64_musl/igmpproxy-0.2.1/.built_check
2024-04-19T20:59:10.0325757Z cd /home/runner/work/test/test/openwrt/build_dir/target-x86_64_musl/igmpproxy-0.2.1 && aclocal
2024-04-19T20:59:10.3849494Z make -C /home/runner/work/test/test/openwrt/build_dir/target-x86_64_musl/igmpproxy-0.2.1/src CC="x86_64-openwrt-linux-musl-gcc" CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/runner/work/test/test/openwrt/build_dir/target-x86_64_musl/igmpproxy-0.2.1=igmpproxy-0.2.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -Dlog=igmpproxy_log -std=gnu99"
2024-04-19T20:59:10.3910012Z make[4]: Entering directory '/home/runner/work/test/test/openwrt/build_dir/target-x86_64_musl/igmpproxy-0.2.1/src'
2024-04-19T20:59:10.3937995Z make[5]: Entering directory '/home/runner/work/test/test/openwrt/build_dir/target-x86_64_musl/igmpproxy-0.2.1'
2024-04-19T20:59:10.3949250Z  cd . && /bin/bash /home/runner/work/test/test/openwrt/build_dir/target-x86_64_musl/igmpproxy-0.2.1/missing automake-1.15 --gnu
2024-04-19T20:59:10.3967168Z /home/runner/work/test/test/openwrt/build_dir/target-x86_64_musl/igmpproxy-0.2.1/missing: line 81: automake-1.15: command not found
2024-04-19T20:59:10.3992644Z WARNING: 'automake-1.15' is missing on your system.
2024-04-19T20:59:10.3993380Z          You should only need it if you modified 'Makefile.am' or
2024-04-19T20:59:10.3994572Z          'configure.ac' or m4 files included by 'configure.ac'.
2024-04-19T20:59:10.3995569Z          The 'automake' program is part of the GNU Automake package:
2024-04-19T20:59:10.3996121Z          <http://www.gnu.org/software/automake>
2024-04-19T20:59:10.3996603Z          It also requires GNU Autoconf, GNU m4 and Perl in order to run:
2024-04-19T20:59:10.3997076Z          <http://www.gnu.org/software/autoconf>
2024-04-19T20:59:10.3997444Z          <http://www.gnu.org/software/m4/>
2024-04-19T20:59:10.3997994Z          <http://www.perl.org/>
2024-04-19T20:59:10.3998572Z make[5]: *** [Makefile:338: Makefile.in] Error 1
2024-04-19T20:59:10.3999504Z make[5]: Leaving directory '/home/runner/work/test/test/openwrt/build_dir/target-x86_64_musl/igmpproxy-0.2.1'
2024-04-19T20:59:10.4003145Z make[4]: *** [Makefile:289: Makefile.in] Error 1
2024-04-19T20:59:10.4004410Z make[4]: Leaving directory '/home/runner/work/test/test/openwrt/build_dir/target-x86_64_musl/igmpproxy-0.2.1/src'
2024-04-19T20:59:10.4005918Z make[3]: *** [Makefile:83: /home/runner/work/test/test/openwrt/build_dir/target-x86_64_musl/igmpproxy-0.2.1/.built] Error 2
2024-04-19T20:59:10.4006918Z make[3]: Leaving directory '/home/runner/work/test/test/openwrt/feeds/packages/net/igmpproxy'
2024-04-19T20:59:10.4009554Z time: package/feeds/packages/igmpproxy/compile#0.43#0.12#0.53
2024-04-19T20:59:10.4017070Z     ERROR: package/feeds/packages/igmpproxy failed to build.
2024-04-19T20:59:10.4020475Z make[2]: *** [package/Makefile:116: package/feeds/packages/igmpproxy/compile] Error 1
2024-04-19T20:59:10.4024809Z make[2]: Leaving directory '/home/runner/work/test/test/openwrt'
2024-04-19T20:59:10.4029483Z make[1]: *** [package/Makefile:110: /home/runner/work/test/test/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
2024-04-19T20:59:10.4034759Z make[1]: Leaving directory '/home/runner/work/test/test/openwrt'
2024-04-19T20:59:10.4043555Z make: *** [/home/runner/work/test/test/openwrt/include/toplevel.mk:231: world] Error 2
2024-04-19T20:59:10.4060810Z ##[error]Process completed with exit code 2.

wention, 2024-04-21:
同样问题 git rev: 849b8c3dbcb8529ae782589f1aac25ebe3030957

wention, 2024-04-21:
跟了一下日志，igmpproxy 的 patch 修改了 automake 脚本，应该在 make 之前执行一下 autoreconf 重新生成 makefile 才对。但我对 openwrt 不太了解不知道怎么修改

ywt114, 2024-04-22:
> 跟了一下日志，igmpproxy 的 patch 修改了 automake 脚本，应该在 make 之前执行一下 autoreconf 重新生成 makefile 才对。但我对 openwrt 不太了解不知道怎么修改

我是添加了一个软连接解决 但不知道怎么在源码处修改

WYC-2020, 2024-04-22:
[igmpproxy.tar.gz](https://github.com/coolsnowwolf/lede/files/15056427/igmpproxy.tar.gz)
替换原来的

ywt114, 2024-04-22:
> [igmpproxy.tar.gz](https://github.com/coolsnowwolf/lede/files/15056427/igmpproxy.tar.gz)
> 
> 替换原来的

这社区不能没有你啊 哈哈

closed 2024-04-22 by ywt114