coolsnowwolf/lede #12087: alist报错

ywt114, 2024-04-20:
### 详细叙述

ERROR: package/lean/alist/alist failed to build.
提示缺少了fuse.h文件，但依赖里也包含了libfuse-dev，

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

2024-04-20T14:03:12.9164707Z time: package/kernel/mt76/compile#20.31#4.19#24.29
2024-04-20T14:03:12.9285820Z make[3]: Entering directory '/home/runner/work/test/test/openwrt/package/lean/alist/alist'
2024-04-20T14:03:13.0746714Z rm -f /home/runner/work/test/test/openwrt/build_dir/target-x86_64_musl/alist-3.33.0/.built
2024-04-20T14:03:13.0776002Z touch /home/runner/work/test/test/openwrt/build_dir/target-x86_64_musl/alist-3.33.0/.built_check
2024-04-20T14:03:13.0821285Z CONFIG_GOLANG_MOD_CACHE_WORLD_READABLE="" GO_BUILD_CACHE_DIR="/home/runner/work/test/test/openwrt/tmp/go-build" GO_MOD_CACHE_DIR="/home/runner/work/test/test/openwrt/dl/go-mod-cache" GO_MOD_ARGS="-modcacherw" GO_PKG="github.com/alist-org/alist" GO_INSTALL_EXTRA="" GO_INSTALL_ALL="" GO_SOURCE_ONLY="" GO_BUILD_PKG="github.com/alist-org/alist/..." GO_EXCLUDES="" GO_GO_GENERATE="" GO_INSTALL_BIN_PATH="/usr/bin" BUILD_DIR="/home/runner/work/test/test/openwrt/build_dir/target-x86_64_musl/alist-3.33.0" GO_BUILD_DIR="/home/runner/work/test/test/openwrt/build_dir/target-x86_64_musl/alist-3.33.0/.go_work/build" GO_BUILD_BIN_DIR="/home/runner/work/test/test/openwrt/build_dir/target-x86_64_musl/alist-3.33.0/.go_work/build/bin" GO_BUILD_DEPENDS_PATH="/usr/share/gocode" GO_BUILD_DEPENDS_SRC="/home/runner/work/test/test/openwrt/staging_dir/target-x86_64_musl/usr/share/gocode/src" GOOS="linux" GOARCH="amd64" GO386="" GOAMD64="v1" GOARM="" GOMIPS="" GOMIPS64="" GOPPC64="" CGO_ENABLED=1 CC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" CGO_CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/runner/work/test/test/openwrt/build_dir/target-x86_64_musl/alist-3.33.0=alist-3.33.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -D_LARGEFILE64_SOURCE" CGO_CPPFLAGS="-I/home/runner/work/test/test/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/include -I/home/runner/work/test/test/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include/fortify -I/home/runner/work/test/test/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include" CGO_CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/runner/work/test/test/openwrt/build_dir/target-x86_64_musl/alist-3.33.0=alist-3.33.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -D_LARGEFILE64_SOURCE" CGO_LDFLAGS="-L/home/runner/work/test/test/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/lib -L/home/runner/work/test/test/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib -znow -zrelro" GOPATH="/home/runner/work/test/test/openwrt/build_dir/target-x86_64_musl/alist-3.33.0/.go_work/build" GOCACHE="/home/runner/work/test/test/openwrt/tmp/go-build" GOMODCACHE="/home/runner/work/test/test/openwrt/dl/go-mod-cache" GOENV=off GOTOOLCHAIN=local /usr/bin/env bash /home/runner/work/test/test/openwrt/feeds/packages/lang/golang//golang-build.sh build -v -buildvcs=false -trimpath -ldflags "all=-buildid '1713543053' -linkmode external -extldflags '-L/home/runner/work/test/test/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/lib -L/home/runner/work/test/test/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib -Wl,-z,now -Wl,-z,relro'"        -ldflags "-w -s -X github.com/alist-org/alist/v3/internal/conf.Version=v3.33.0-x86_64 -X github.com/alist-org/alist/v3/internal/conf.WebVersion=3.33.0 -buildid '1713543053' -linkmode external -extldflags '-L/home/runner/work/test/test/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/lib -L/home/runner/work/test/test/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib -Wl,-z,now -Wl,-z,relro'"  
2024-04-20T14:03:13.0859577Z Finding targets
2024-04-20T14:03:13.4116712Z 
2024-04-20T14:03:13.4118904Z Building targets
2024-04-20T14:03:13.7957575Z github.com/winfsp/cgofuse/fuse
2024-04-20T14:03:13.8527481Z # github.com/winfsp/cgofuse/fuse
2024-04-20T14:03:13.8529628Z ../../../../../dl/go-mod-cache/github.com/winfsp/cgofuse@v1.5.1-0.20230130140708-f87f5db493b5/fuse/host_cgo.go:117:10: fatal error: fuse.h: No such file or directory
2024-04-20T14:03:13.8531026Z   117 | #include <fuse.h>
2024-04-20T14:03:13.8531471Z       |          ^~~~~~~~
2024-04-20T14:03:13.8536643Z compilation terminated.
2024-04-20T14:03:16.4719655Z 
2024-04-20T14:03:16.6200201Z make[3]: *** [Makefile:92: /home/runner/work/test/test/openwrt/build_dir/target-x86_64_musl/alist-3.33.0/.built] Error 1
2024-04-20T14:03:16.6202336Z make[3]: Leaving directory '/home/runner/work/test/test/openwrt/package/lean/alist/alist'
2024-04-20T14:03:16.6205501Z time: package/lean/alist/alist/compile#2.52#1.80#3.70
2024-04-20T14:03:16.6212685Z     ERROR: package/lean/alist/alist failed to build.
2024-04-20T14:03:16.6216287Z make[2]: *** [package/Makefile:116: package/lean/alist/alist/compile] Error 1
2024-04-20T14:03:16.6221646Z make[2]: Leaving directory '/home/runner/work/test/test/openwrt'
2024-04-20T14:03:16.6228556Z make[1]: *** [package/Makefile:110: /home/runner/work/test/test/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
2024-04-20T14:03:16.6232316Z make[1]: Leaving directory '/home/runner/work/test/test/openwrt'
2024-04-20T14:03:16.6239011Z make: *** [/home/runner/work/test/test/openwrt/include/toplevel.mk:231: world] Error 2
2024-04-20T14:03:16.6257333Z ##[error]Process completed with exit code 2.

WYC-2020, 2024-04-22:
自己搜类似issue fuse关键字

ywt114, 2024-04-22:
> 自己搜类似issue fuse关键字

可是我的依赖环境已经安装了libfuse-dev了，所以不明白原因

WYC-2020, 2024-04-22:
> > 自己搜类似issue fuse关键字
> 
> 可是我的依赖环境已经安装了libfuse-dev了，所以不明白原因

那就改成依赖 我就是改成的依赖 等会我发给你

WYC-2020, 2024-04-22:
我原来 就是靠安装这玩意 但是 始终觉得 依赖宿主机这玩意不靠谱

WYC-2020, 2024-04-22:
[image]
如此

ywt114, 2024-04-22:
> ![image](https://private-user-images.githubusercontent.com/34025388/324369946-280fd897-3737-4112-aec0-592e2a01eb39.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

太看得起我了，我都不知道你改的这是哪个文件

WYC-2020, 2024-04-22:
> 太看得起我了，我都不知道你改的这是哪个文件

你自己搜索 package 里面 引用的那个 地方的alist啊 然后修改makefile 这一行就行了，我自己仓库是引用的kenzok8仓库的alist，不是luci 是alist
[image]

ywt114, 2024-04-22:
> 你自己搜索 package 里面 引用的那个 地方的alist啊 然后修改makefile 这一行就行了，我自己仓库是引用的kenzok8仓库的alist，不是luci 是alist [...]

sed -i 's/golang\/host/golang\/host fuse/g' package/luci-app-alist/alist/Makefile

ywt114, 2024-04-22:
> sed -i 's/golang\/host/golang\/host fuse/g' package/luci-app-alist/alist/Makefile

WYC-2020, 2024-04-22:
改成 fuse/host把 这样感觉更加合理一些
[image]
至于如何改 看你自己 我是克隆到有仓库 所以我是直接仓库改的

closed 2024-04-22 by ywt114

ywt114, 2024-04-23:
> 改成 fuse/host把 这样感觉更加合理一些 [...]

终极原因找到了 是github修改了部分规则导致raw文件链接识别失效导致 我之前安装环境编译依赖用的是sudo -E apt -yqq install $(curl -fsSL https://raw.githubusercontent.com/xxxxxx)之前都是可以正常识别https://raw.githubusercontent.com/xxxxxx的链接 规则修改后识别成https://raw.githubusercontent.com/xxxxxx) 带括号，不能正常读取造成依赖没用被安装造成 我修改了代码sudo -E apt -yqq install $(curl -fsSL https://raw.githubusercontent.com/xxxxxx | tr '\n' ' ') 解决 很抱歉 我太粗心没有关注到fuse没有被正常安装 所以实际上源码并没有问题 Makefile不需要修改