coolsnowwolf/lede #12094: mt7621+mt7613e mediatek/mt7663pr2h.bin not found, switching to mediatek/mt7663pr2h_rebb.bin

trackoftime, 2024-04-24:
### 详细叙述

看内核没有 mt7613e 的选项，就选了 mt7615e 如下 
配置为 d-team_newifi-d2 产品
CONFIG_TARGET_ramips_mt7621_DEVICE_d-team_newifi-d2=y

# CONFIG_PACKAGE_kmod-mt7603e is not set
CONFIG_PACKAGE_kmod-mt7615-common=y
CONFIG_PACKAGE_kmod-mt7615-firmware=y
# CONFIG_PACKAGE_kmod-mt7615d is not set
# CONFIG_PACKAGE_kmod-mt7615d_dbdc is not set
CONFIG_PACKAGE_kmod-mt7615e=y

驱动有加载，中间报错没找到  mediatek/mt7663pr2h.bin not found, switching to mediatek/mt7663pr2h_rebb.bin
[   49.458659] mt7621-pci 1e140000.pcie: bus=2 slot=1 irq=27
[   49.469837] mt7615e 0000:02:00.0: enabling device (0000 -> 0002)
[   49.484894] mt7615e 0000:02:00.0: Invalid MAC address, using random address 3a:59:25:37:ec:6a
[   52.069988] mt7615e 0000:02:00.0: mediatek/mt7663pr2h.bin not found, switching to mediatek/mt7663pr2h_rebb.bin
[   54.074202] mt7615e 0000:02:00.0: failed to load mediatek/mt7663pr2h_rebb.bin
[   76.228963] mtk_soc_eth 1e100000.ethernet eth0: configuring for fixed/rgmii link mode
[   76.237457] mtk_soc_eth 1e100000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx

驱动信息
PKG_NAME:=mt76
PKG_RELEASE=5

PKG_LICENSE:=GPLv2
PKG_LICENSE_FILES:=

PKG_SOURCE_URL:=https://github.com/openwrt/mt76
PKG_SOURCE_PROTO:=git
PKG_SOURCE_DATE:=2022-12-22
PKG_SOURCE_VERSION:=5b509e80384ab019ac11aa90c81ec0dbb5b0d7f2
PKG_MIRROR_HASH:=6fc25df4d28becd010ff4971b23731c08b53e69381a9e4c868091899712f78a9

把 root@OpenWrt:~# ls /lib/firmware/mediatek/mt7615_
mt7615_cr4.bin        mt7615_n9.bin         mt7615_rom_patch.bin
重命名为 mt7663pr2h.bin 提示

[  352.309923] mt7615e 0000:02:00.0: HW/SW Version: 0x77cf7722, Build Time: H#)b
[  372.357023] mt7615e 0000:02:00.0: Message 00000007 (seq 4) timeout
[  372.363318] mt7615e 0000:02:00.0: Failed to start patch
[  392.837043] mt7615e 0000:02:00.0: Message 00000010 (seq 5) timeout
[  392.843233] mt7615e 0000:02:00.0: Failed to release patch semaphore
[  392.849622] mt7615e 0000:02:00.0: mediatek/mt7663pr2h.bin not found, switching to mediatek/mt7663pr2h_rebb.bin
[  392.879755] mt7615e 0000:02:00.0: failed to load mediatek/mt7663pr2h_rebb.bin

root@OpenWrt:~# ifconfig wlan0
wlan0     Link encap:Ethernet  HWaddr 2A:64:F8:97:EE:D5  
          BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

root@OpenWrt:~# 
root@OpenWrt:/lib/firmware/mediatek# ifconfig wlan0 up
ifconfig: SIOCSIFFLAGS: I/O error
root@OpenWrt:/lib/firmware/mediatek# 

接口不能正常工作，大佬，帮忙看下

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

mt7621

### 详细日志

编码没问题

benboguan, 2024-04-24:
这玩意不是7612吗？

trackoftime, 2024-04-24:
> 这玩意不是7612吗？

硬件用的是 mt7613e

aiamadeus, 2024-04-24:
你要选 kmod-mt7663-firmware-ap

trackoftime, 2024-04-25:
> 你要选 kmod-mt7663-firmware-ap
选了 kmod-mt7663-firmware-ap ，目录下看是有 对应的 fw,但还是出现 Failed to get patch semaphore

[   43.404162] mt7615e 0000:02:00.0: Message 00000010 (seq 1) timeout
[   43.410337] mt7615e 0000:02:00.0: Failed to get patch semaphore
[   43.416453] mt7615e 0000:02:00.0: mediatek/mt7663pr2h.bin not found, switching to mediatek/mt7663pr2h_rebb.bin
[   63.884180] mt7615e 0000:02:00.0: Message 00000010 (seq 2) timeout
[   63.890355] mt7615e 0000:02:00.0: Failed to get patch semaphore
[   63.896417] mt7615e 0000:02:00.0: failed to load mediatek/mt7663pr2h_rebb.bin
[   70.193226] mtk_soc_eth 1e100000.ethernet eth0: configuring for fixed/rgmii link mode
[   70.201595] mtk_soc_eth 1e100000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[   70.210588] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   70.223835] br-lan: port 1(eth0.1) entered blocking state
[   70.229351] br-lan: port 1(eth0.1) entered disabled state
[   70.235572] device eth0.1 entered promiscuous mode
[   70.240382] device eth0 entered promiscuous mode
[   70.249476] br-lan: port 1(eth0.1) entered blocking state
[   70.255009] br-lan: port 1(eth0.1) entered forwarding state
[   72.367320] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   84.364167] mt7615e 0000:02:00.0: Message 00000010 (seq 3) timeout
[   84.370341] mt7615e 0000:02:00.0: Failed to get patch semaphore
[   84.376480] mt7615e 0000:02:00.0: mediatek/mt7663pr2h.bin not found, switching to mediatek/mt7663pr2h_rebb.bin
[  104.844160] mt7615e 0000:02:00.0: Message 00000010 (seq 4) timeout
[  104.850335] mt7615e 0000:02:00.0: Failed to get patch semaphore
[  104.856426] mt7615e 0000:02:00.0: failed to load mediatek/mt7663pr2h_rebb.bin
[  125.324179] mt7615e 0000:02:00.0: Message 00000010 (seq 5) timeout
[  125.330354] mt7615e 0000:02:00.0: Failed to get patch semaphore
[  125.336466] mt7615e 0000:02:00.0: mediatek/mt7663pr2h.bin not found, switching to mediatek/mt7663pr2h_rebb.bin
[  145.804182] mt7615e 0000:02:00.0: Message 00000010 (seq 6) timeout
[  145.810355] mt7615e 0000:02:00.0: Failed to get patch semaphore
[  145.816451] mt7615e 0000:02:00.0: failed to load mediatek/mt7663pr2h_rebb.bin
[  166.284174] mt7615e 0000:02:00.0: Message 00000010 (seq 7) timeout
[  166.290348] mt7615e 0000:02:00.0: Failed to get patch semaphore
[  166.296454] mt7615e 0000:02:00.0: mediatek/mt7663pr2h.bin not found, switching to mediatek/mt7663pr2h_rebb.bin
[  186.764167] mt7615e 0000:02:00.0: Message 00000010 (seq 8) timeout
[  186.770341] mt7615e 0000:02:00.0: Failed to get patch semaphore
[  186.776409] mt7615e 0000:02:00.0: failed to load mediatek/mt7663pr2h_rebb.bin
[  207.244188] mt7615e 0000:02:00.0: Message 00000010 (seq 9) timeout
[  207.250363] mt7615e 0000:02:00.0: Failed to get patch semaphore
[  207.256475] mt7615e 0000:02:00.0: mediatek/mt7663pr2h.bin not found, switching to mediatek/mt7663pr2h_rebb.bin
[  227.724180] mt7615e 0000:02:00.0: Message 00000010 (seq 10) timeout
[  227.730439] mt7615e 0000:02:00.0: Failed to get patch semaphore
[  227.736513] mt7615e 0000:02:00.0: failed to load mediatek/mt7663pr2h_rebb.bin
[  248.204321] mt7615e 0000:02:00.0: Message 00000010 (seq 11) timeout
[  248.210624] mt7615e 0000:02:00.0: Failed to get patch semaphore
[  248.216801] mt7615e 0000:02:00.0: mediatek/mt7663pr2h.bin not found, switching to mediatek/mt7663pr2h_rebb.bin
[  268.684176] mt7615e 0000:02:00.0: Message 00000010 (seq 12) timeout
[  268.690436] mt7615e 0000:02:00.0: Failed to get patch semaphore
[  268.696490] mt7615e 0000:02:00.0: failed to load mediatek/mt7663pr2h_rebb.bin
[  289.164180] mt7615e 0000:02:00.0: Message 00000010 (seq 13) timeout
[  289.170440] mt7615e 0000:02:00.0: Failed to get patch semaphore
[  289.176558] mt7615e 0000:02:00.0: mediatek/mt7663pr2h.bin not found, switching to mediatek/mt7663pr2h_rebb.bin
[  309.644188] mt7615e 0000:02:00.0: Message 00000010 (seq 14) timeout
[  309.650447] mt7615e 0000:02:00.0: Failed to get patch semaphore
[  309.656512] mt7615e 0000:02:00.0: failed to load mediatek/mt7663pr2h_rebb.bin
[  330.124304] mt7615e 0000:02:00.0: Message 00000010 (seq 15) timeout
[  330.130565] mt7615e 0000:02:00.0: Failed to get patch semaphore
[  330.136685] mt7615e 0000:02:00.0: mediatek/mt7663pr2h.bin not found, switching to mediatek/mt7663pr2h_rebb.bin
[  350.604185] mt7615e 0000:02:00.0: Message 00000010 (seq 1) timeout
[  350.610358] mt7615e 0000:02:00.0: Failed to get patch semaphore
[  350.616442] mt7615e 0000:02:00.0: failed to load mediatek/mt7663pr2h_rebb.bin
[  371.084190] mt7615e 0000:02:00.0: Message 00000010 (seq 2) timeout
[  371.090365] mt7615e 0000:02:00.0: Failed to get patch semaphore
[  371.096474] mt7615e 0000:02:00.0: mediatek/mt7663pr2h.bin not found, switching to mediatek/mt7663pr2h_rebb.bin
[  391.564195] mt7615e 0000:02:00.0: Message 00000010 (seq 3) timeout
[  391.570372] mt7615e 0000:02:00.0: Failed to get patch semaphore
[  391.576410] mt7615e 0000:02:00.0: failed to load mediatek/mt7663pr2h_rebb.bin
[  412.044259] mt7615e 0000:02:00.0: Message 00000010 (seq 4) timeout
[  412.050438] mt7615e 0000:02:00.0: Failed to get patch semaphore
[  412.056632] mt7615e 0000:02:00.0: mediatek/mt7663pr2h.bin not found, switching to mediatek/mt7663pr2h_rebb.bin
[  432.524198] mt7615e 0000:02:00.0: Message 00000010 (seq 5) timeout
[  432.530374] mt7615e 0000:02:00.0: Failed to get patch semaphore
[  432.536457] mt7615e 0000:02:00.0: failed to load mediatek/mt7663pr2h_rebb.bin
[... cut ...]
[  502.115332] jffs2_scan_eraseblock(): End of filesystem marker found at 0x0
[  502.122241] jffs2_build_filesystem(): unlocking the mtd device... 
[  502.122274] done.
[  502.130506] jffs2_build_filesystem(): erasing all blocks after the end marker... 
[  527.077850] done.
[  527.087320] jffs2: notice: (9431) jffs2_build_xattr_subsystem: complete building xattr subsystem, 0 of xdatum (0 unchecked, 0 orphan) and 0 of xref (0 dead, 0 orphan) found.
[  527.971731] overlayfs: upper fs does not support tmpfile.

trackoftime, 2024-04-25:
root@OpenWrt:(unknown)# ls /lib/firmware/mediatek/ -l
-rw-r--r--    1 root     root        122644 Jan  9 13:45 mt7615_cr4.bin
-rw-r--r--    1 root     root        457992 Jan  9 13:45 mt7615_n9.bin
-rw-r--r--    1 root     root         11102 Jan  9 13:45 mt7615_rom_patch.bin
-rw-r--r--    1 root     root        346216 Jan  9 13:45 mt7663_n9_rebb.bin
-rw-r--r--    1 root     root        440568 Jan  9 13:45 mt7663_n9_v3.bin
-rw-r--r--    1 root     root        278558 Jan  9 13:45 mt7663pr2h.bin
-rw-r--r--    1 root     root        211518 Jan  9 13:45 mt7663pr2h_rebb.bin
ash: getcwd: No such file or directory
root@OpenWrt:(unknown)#

trackoftime, 2024-04-25:
更新了内核版本为 5.10 看起来没问题了

wlan0     Link encap:Ethernet  HWaddr 68:F2:00:22:69:E4  
          inet6 addr: fe80::6af2:ff:fe22:69e4/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:45 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:5711 (5.5 KiB)

wlan1     Link encap:Ethernet  HWaddr 2E:4A:F3:D2:FB:FD  
          inet6 addr: fe80::2c4a:f3ff:fed2:fbfd/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:45 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:5711 (5.5 KiB)

closed 2024-05-28 by aiamadeus