coolsnowwolf/lede #12100: config.guess: cannot create a temporary directory in /tmp

wangweitung, 2024-04-27:
### 详细叙述

```
config.guess: cannot create a temporary directory in /tmp
make -s -C scripts/config mconf: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [/home/openwrt/include/toplevel.mk:106: scripts/config/mconf] Error 1
```

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

X86 64

### 详细日志

```
conf.c:915:1: fatal error: error writing to /tmp/ccT0Wovj.s: No space left on device
  915 | }
      | ^
compilation terminated.
make[2]: *** [<builtin>: conf.o] Error 1
make[2]: Leaving directory '/home/Disk/4Tsdd/Github/lede/scripts/config'
make[1]: *** [/home/Disk/4Tsdd/Github/lede/include/toplevel.mk:106: scripts/config/conf] Error 2
make: *** [/home/Disk/4Tsdd/Github/lede/include/toplevel.mk:229: world] Error 2
```

wangweitung, 2024-04-27:
```
git clone https://github.com/coolsnowwolf/lede
cd lede
make menuconfig
```

aiamadeus, 2024-04-27:
显然你硬盘空间不够了。。。

wangweitung, 2024-04-27:
> 显然你硬盘空间不够了。。。

[image]

实际还有2T可用，我再检查下。。。谢谢大佬回复。。。

wangweitung, 2024-04-27:
> 显然你硬盘空间不够了。。。

确实是。。。系统盘爆了。。。编译的盘还够。。

closed 2024-04-27 by wangweitung