coolsnowwolf/lede #12101: mwan3 2.11 显示error 15

GreatMichaelLee, 2024-04-28:
### 详细叙述

如标题所示，查了一下网上，这个供参考 https://blog.csdn.net/lvxiangyu11/article/details/131408454 ，我试着手工重启mwan3 恢复了，但是每次重启路由器拨号完就会error 15, 之前2.8.8不会，大佬帮忙看看是不是有办法给mwan3弄个判断脚本，每次多拨完成后自动重启一下mwan3?

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

[image]

zuni11, 2024-05-04:
同样的问题，并且大几率mwan3 restart不能恢复，已经回退2.8.8

GreatMichaelLee, 2024-05-04:
> 同样的问题，并且大几率mwan3 restart不能恢复，已经回退2.8.8

没人理，呵呵，之前求爷爷告奶奶的求大佬们升级个mwan3,结果一堆大佬告诉我，那玩意儿升级了“很危险，慢一点哦，再慢一点”（林志玲高德导航BGM :) ) 现在用的好好的稳定的，干嘛要升.......我一想，大佬说的也对，哎，那就不升了把？ 就不再说这个事了，以为2.8.8养老了。结果，突然某天，大佬就给你一下就，啪，给你升了，哈哈。有问题，谁叫你要升的？是你之前一直叫唤要升的，对不对，结果升了有问题，你又来叫唤了对不对....哈哈。

WYC-2020, 2024-05-06:
> > 同样的问题，并且大几率mwan3 restart不能恢复，已经回退2.8.8
> 
> 没人理，呵呵，之前求爷爷告奶奶的求大佬们升级个mwan3,结果一堆大佬告诉我，那玩意儿升级了“很危险，慢一点哦，再慢一点”（林志玲高德导航BGM :) ) 现在用的好好的稳定的，干嘛要升.......我一想，大佬说的也对，哎，那就不升了把？ 就不再说这个事了，以为2.8.8养老了。结果，突然某天，大佬就给你一下就，啪，给你升了.....嗯？有问题？谁叫你要升的？是你之前一直叫唤要升的，对不对，结果升了有问题，你又来叫唤了对不对....哈哈。

知道就好哈哈 这个我以前就说过，只要不是什么有特别新意的比较好的特性 完全没必要升级，玩的就是稳定，我软路由都稳定好久了哈哈，这是众口难调，所以我都是自己克隆仓库 自己搞一份 自己玩

King-Of-Knights, 2024-05-14:
@GreatMichaelLee @WYC-2020 两位大佬好啊，我是光猫拨号，OP二级路由，在2.8.8，路由器刚起来的时候，还是有error但是tracking is active，加入重启脚本后，ssh里面显示正常了，但是luci里面界面不正常，请问有解嘛？
[image]

ramondelee, 2024-05-16:
这个问题在21.02里也有，以下给个推测和两个解决方法
推测：应该是mwan3先于wan启动，导致没有识别到，2.8.8版本应该有检测或刷新代码所以正常，2.11应该少了这个
解决方案一：系统-启动项-本地脚本，在exit 0前插入一行代码 mwan3 restart ，重启OK
解决方案二：编译时，修改源码files/etc/init.d/mwan3中START=19为START=96，生成固件重新刷机
我自己测试有效

coolsnowwolf, 2026-09-05:
当前 feeds/packages 的 mwan3 已是 2.12.2，且 init/hotplug 文件与 OpenWrt 主线一致；该 issue 报告的是更早的 2.11 启动时序问题。日志/讨论给出的 START=96 workaround 属于旧版本启动顺序调整，当前代码已同步上游处理。无需新增补丁，issue 已过时，现关闭。

closed 2026-09-05 by coolsnowwolf