coolsnowwolf/lede #12102: 新增设备支持时编译报错，求大佬们指导还有哪没添加到。

SpaHuang, 2024-04-28:
### 详细叙述

手上有个新硬件设备，根据H68K的硬件信息添加支持，最后编译找不到dts文件，反复对比了H68K的文件好像也没有漏掉什么，求大佬们指导一下，感谢感谢。

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

roceos_roc-x4s

### 详细日志

h68k targets的相关文件
pidge@ubuntu:~/lede$ grep -lri h68k ./target/
./target/linux/rockchip/armv8/base-files/etc/board.d/01_leds
./target/linux/rockchip/armv8/base-files/etc/board.d/02_network
./target/linux/rockchip/files/arch/arm64/boot/dts/rockchip/rk3568-opc-h68k.dts
./target/linux/rockchip/patches-5.15/210-rockchip-rk356x-add-support-for-new-boards.patch
./target/linux/rockchip/image/armv8.mk
./target/linux/rockchip/patches-6.6/210-rockchip-rk356x-add-support-for-new-boards.patch
./target/linux/rockchip/patches-6.1/210-rockchip-rk356x-add-support-for-new-boards.patch

h68k package的相关文件
pidge@ubuntu:~/lede$ grep -lri h68k ./package/
./package/boot/uboot-rockchip/Makefile
./package/boot/uboot-rockchip/patches/315-rockchip-rk3568-Add-support-for-radxa_e25.patch
./package/boot/uboot-rockchip/patches/316-rockchip-rk3568-Add-support-for-lyt_t68m.patch
./package/boot/uboot-rockchip/patches/312-rockchip-rk3568-Add-support-for-hinlink-opc-h68k.patch
./package/boot/uboot-rockchip/patches/313-rockchip-rk3568-Add-support-for-fastrhino-r66s.patch
./package/boot/uboot-rockchip/patches/314-rockchip-rk3568-Add-support-for-photonicat.patch

x4s targets的相关文件
pidge@ubuntu:~/lede$ grep -lri roc-x4s ./target/
./target/linux/rockchip/armv8/base-files/etc/board.d/02_network
./target/linux/rockchip/files/arch/arm64/boot/dts/rockchip/rk3568-roc-x4s.dts
./target/linux/rockchip/patches-5.15/210-rockchip-rk356x-add-support-for-new-boards.patch
./target/linux/rockchip/image/armv8.mk
./target/linux/rockchip/patches-6.6/210-rockchip-rk356x-add-support-for-new-boards.patch
./target/linux/rockchip/patches-6.1/210-rockchip-rk356x-add-support-for-new-boards.patch

x4s package的相关文件
pidge@ubuntu:~/lede$ grep -lri roc-x4s ./package/
./package/boot/uboot-rockchip/Makefile
./package/boot/uboot-rockchip/patches/318-rockchip-rk3568-Add-support-for-advantech-rsb4810.patch
./package/boot/uboot-rockchip/patches/319-rockchip-rk3568-Add-support-for-roceos-roc-x4s.patch

编译最后报错信息，报错说是找不到-roc-x4s.dts，但所新增的设备的那个文件都没有单独定义-roc-x4s.dts的名称。
rm -f /home/pidge/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3568-nanopi-r5s.dtb.tmp
aarch64-openwrt-linux-musl-cpp -nostdinc -x assembler-with-cpp -DDTS_NO_LEGACY -I/home/pidge/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-6.1.87/arch/arm64/boot/dts -I/home/pidge/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-6.1.87/arch/arm64/boot/dts/include -I/home/pidge/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-6.1.87/include/ -undef -D__DTS__  -o /home/pidge/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3568-roc-pc.dtb.tmp /home/pidge/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-6.1.87/arch/arm64/boot/dts/rockchip/rk3568-roc-pc.dts
/home/pidge/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-6.1.87/scripts/dtc/dtc -O dtb -i/home/pidge/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-6.1.87/arch/arm64/boot/dts/rockchip/ -Wno-unit_address_vs_reg -Wno-simple_bus_reg -Wno-unit_address_format -Wno-pci_bridge -Wno-pci_device_bus_num -Wno-pci_device_reg -Wno-avoid_unnecessary_addr_size -Wno-alias_paths -Wno-graph_child_address -Wno-graph_port -Wno-unique_unit_address    -o /home/pidge/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3568-roc-pc.dtb /home/pidge/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3568-roc-pc.dtb.tmp
rm -f /home/pidge/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3568-roc-pc.dtb.tmp
aarch64-openwrt-linux-musl-cpp -nostdinc -x assembler-with-cpp -DDTS_NO_LEGACY -I/home/pidge/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-6.1.87/arch/arm64/boot/dts -I/home/pidge/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-6.1.87/arch/arm64/boot/dts/include -I/home/pidge/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-6.1.87/include/ -undef -D__DTS__  -o /home/pidge/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image--roc-x4s.dtb.tmp /home/pidge/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-6.1.87/arch/arm64/boot/dts/rockchip/-roc-x4s.dts
cc1: fatal error: /home/pidge/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-6.1.87/arch/arm64/boot/dts/rockchip/-roc-x4s.dts: No such file or directory
compilation terminated.
make[5]: *** [Makefile:82: /home/pidge/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image--roc-x4s.dtb] Error 1
make[5]: Leaving directory '/home/pidge/lede/target/linux/rockchip/image'
make[4]: *** [Makefile:24: install] Error 2
make[4]: Leaving directory '/home/pidge/lede/target/linux/rockchip'
make[3]: *** [Makefile:11: install] Error 2
make[3]: Leaving directory '/home/pidge/lede/target/linux'
time: target/linux/install#12.26#5.98#19.08
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/install] Error 1
make[2]: Leaving directory '/home/pidge/lede'
make[1]: *** [target/Makefile:24: /home/pidge/lede/staging_dir/target-aarch64_generic_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/home/pidge/lede'
make: *** [/home/pidge/lede/include/toplevel.mk:231: world] Error 2

附DTS头信息
/dts-v1/;

/ {
        model = "ROCEOS ROC-X4S";
        compatible = "roceos,roc-x4s", "rockchip,rk3568";

aiamadeus, 2024-04-29:
`define Device/`缺了`SOC :=`

SpaHuang, 2024-04-29:
> `define Device/`缺了`SOC :=`

感谢指导，已经编译完成。

closed 2024-04-29 by SpaHuang