coolsnowwolf/lede #12112: 拉取最新源码编译6.6内核x86固件报错

bi4tms, 2024-04-30:
### 详细叙述

拉取最新源码编译6.6内核x86固件报错
@Bin

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

Applying /home/x86.6.6/lede/target/linux/generic/pending-6.6/901-usb-add-more-modem-support.patch using plaintext: 
patching file drivers/net/usb/qmi_wwan.c
patching file drivers/usb/serial/option.c
Hunk #1 FAILED at 2278.
1 out of 1 hunk FAILED -- saving rejects to file drivers/usb/serial/option.c.rej
Patch failed!  Please fix /home/x86.6.6/lede/target/linux/generic/pending-6.6/901-usb-add-more-modem-support.patch!
make[4]: *** [Makefile:34: /home/x86.6.6/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.29/.prepared_c26641d1b0d9504d84493ef4ea8a1191] Error 1
make[4]: Leaving directory '/home/x86.6.6/lede/target/linux/x86'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/home/x86.6.6/lede/target/linux'
time: target/linux/compile#19.33#15.57#38.85
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/compile] Error 1
make[2]: Leaving directory '/home/x86.6.6/lede'
make[1]: *** [target/Makefile:23: /home/x86.6.6/lede/staging_dir/target-x86_64_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/x86.6.6/lede'
make: *** [/home/x86.6.6/lede/include/toplevel.mk:231：world] 错误 2

kxdn3, 2024-04-30:
一样出错

Beginner-Go, 2024-04-30:
https://github.com/coolsnowwolf/lede/commit/54bb95de0f4cfa07e63d3a668e1a2a4fcaf8c99f  这个commit引起的 我来修一下

GreatMichaelLee, 2024-04-30:
一样错误

mumuou, 2024-05-01:
901-usb-add-more-modem-support.patch 补丁包有问题，删除就好了

kxdn3, 2024-05-01:
还没修好

mumuou, 2024-05-01:
> > 还没修好
> 
> 按照pr里面的自己去修改就行了

那破补丁也是一顿瞎改，全角。都出来了，合并一样翻车。

tonyzzt001, 2024-05-01:
同样报错，等修复。

tonyzzt001, 2024-05-01:
> 901-usb-add-more-modem-support.patch 补丁包有问题，删除就好了

删除可解决？

mumuou, 2024-05-01:
901-usb-add-more-modem-support.patch 安装我上面这个改，直接覆盖复制覆盖就行了~~~

tonyzzt001, 2024-05-01:
> 901-usb-add-more-modem-support.patch 按照我上面这个改，直接覆盖复制覆盖就行了~~~
感谢，试试

mumuou, 2024-05-01:
好像不用了，直接更新就行了，合并的代码把全角。，啥的都改好了，你自己试试吧

tonyzzt001, 2024-05-01:
啊？好的，在更一下。

Beginner-Go, 2024-05-01:
> > > 还没修好
> > 
> > 
> > 按照pr里面的自己去修改就行了
> 
> 那破补丁也是一顿瞎改，全角。都出来了，合并一样翻车。

要不您来更新？