coolsnowwolf/lede #12120: 拉取最新源码编译5.15内核x86固件报错

bi4tms, 2024-05-02:
### 详细叙述

拉取最新源码报错
@Beginner-Go 

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

  CC [M]  net/ipv6/ip6_udp_tunnel.o
  CC      net/packet/af_packet.o
  AR      net/packet/built-in.a
  AR      net/bridge/netfilter/built-in.a
  CC      net/bridge/br.o
  CC      net/bridge/br_device.o
  CC      net/bridge/br_fdb.o
  CC      net/bridge/br_forward.o
  CC      net/bridge/br_if.o
  CC      net/bridge/br_input.o
net/bridge/br_input.c: In function 'br_handle_frame_finish':
net/bridge/br_input.c:127:24: error: too few arguments to function 'br_pass_frame_up'
  127 |                 return br_pass_frame_up(skb);
      |                        ^~~~~~~~~~~~~~~~
net/bridge/br_input.c:34:12: note: declared here
   34 | static int br_pass_frame_up(struct sk_buff *skb, bool promisc)
      |            ^~~~~~~~~~~~~~~~
make[7]: *** [scripts/Makefile.build:289: net/bridge/br_input.o] Error 1
make[6]: *** [scripts/Makefile.build:552: net/bridge] Error 2
make[5]: *** [Makefile:1907: net] Error 2
make[5]: Leaving directory '/home/x86.5.15/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.157'
make[4]: *** [Makefile:28: /home/x86.5.15/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.157/.modules] Error 2
make[4]: Leaving directory '/home/x86.5.15/lede/target/linux/x86'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/home/x86.5.15/lede/target/linux'
time: target/linux/compile#2266.64#182.20#2460.10
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/compile] Error 1
make[2]: Leaving directory '/home/x86.5.15/lede'
make[1]: *** [target/Makefile:23: /home/x86.5.15/lede/staging_dir/target-x86_64_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/x86.5.15/lede'
make: *** [/home/x86.5.15/lede/include/toplevel.mk:231：world] 错误 2

Beginner-Go, 2024-05-02:
明天我看看

dukmonki, 2024-05-03:
一样的问题，MT7621

luxiaok, 2024-05-03:
mt798x

WYC-2020, 2024-05-04:
都五一谁没事天天盯到这玩意修，急用就自己回退呗

closed 2024-05-05 by coolsnowwolf