coolsnowwolf/lede #12121: 编译到node这块，老是遇到这个错误，怎么也搞不定，求助！

GreatMichaelLee, 2024-05-03:
### 详细叙述

现在编译，都rf -rm ./tmp && make clean(distclean)什么都试过了，每次必定编译到node这块失败，报这个错，到底什么鬼啊，老是说头文件没有包含啥，类型不匹配或未定义？？有没有大佬帮我看看，现在搞得我只能恢复较早能编译过的vmware镜像再git pull一次编译过。之前的快照是可以编译过的，也是邪门。

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

  g++  -o /home/builder/lede_x86/build_dir/target-x86_64_musl/node-v14.21.3/out/Release/obj.host/v8_libbase/deps/v8/src/base/bounded-page-allocator.o ../deps/v8/src/base/bounded-page-allocator.cc '-DNODE_OPENSSL_CONF_NAME=nodejs_conf' '-DICU_NO_USER_DATA_OVERRIDE' '-DV8_GYP_BUILD' '-DV8_TYPED_ARRAY_MAX_SIZE_IN_HEAP=64' '-D__STDC_FORMAT_MACROS' '-DOPENSSL_NO_PINSHARED' '-DOPENSSL_THREADS' '-DV8_TARGET_ARCH_X64' '-DV8_EMBEDDER_STRING="-node.88"' '-DENABLE_DISASSEMBLER' '-DV8_PROMISE_INTERNAL_FIELD_COUNT=1' '-DENABLE_MINOR_MC' '-DOBJECT_PRINT' '-DV8_CONCURRENT_MARKING' '-DV8_ARRAY_BUFFER_EXTENSION' '-DV8_ENABLE_LAZY_SOURCE_POSITIONS' '-DV8_USE_SIPHASH' '-DDISABLE_UNTRUSTED_CODE_MITIGATIONS' '-DV8_WIN64_UNWINDING_INFO' '-DV8_ENABLE_REGEXP_INTERPRETER_THREADED_DISPATCH' '-DV8_SNAPSHOT_COMPRESSION' '-DBUILDING_V8_BASE_SHARED' -I/home/builder/lede_x86/staging_dir/target-x86_64_musl/usr/include -I../deps/v8 -I../deps/v8/include  -pthread -Wno-unused-parameter -m64 -Wno-return-type -fno-strict-aliasing -m64 -O3 -fno-omit-frame-pointer -fdata-sections -ffunction-sections -O3 -fno-rtti -fno-exceptions -std=gnu++1y -MMD -MF /home/builder/lede_x86/build_dir/target-x86_64_musl/node-v14.21.3/out/Release/.deps//home/builder/lede_x86/build_dir/target-x86_64_musl/node-v14.21.3/out/Release/obj.host/v8_libbase/deps/v8/src/base/bounded-page-allocator.o.d.raw   -c
  g++  -o /home/builder/lede_x86/build_dir/target-x86_64_musl/node-v14.21.3/out/Release/obj.host/v8_libbase/deps/v8/src/base/cpu.o ../deps/v8/src/base/cpu.cc '-DNODE_OPENSSL_CONF_NAME=nodejs_conf' '-DICU_NO_USER_DATA_OVERRIDE' '-DV8_GYP_BUILD' '-DV8_TYPED_ARRAY_MAX_SIZE_IN_HEAP=64' '-D__STDC_FORMAT_MACROS' '-DOPENSSL_NO_PINSHARED' '-DOPENSSL_THREADS' '-DV8_TARGET_ARCH_X64' '-DV8_EMBEDDER_STRING="-node.88"' '-DENABLE_DISASSEMBLER' '-DV8_PROMISE_INTERNAL_FIELD_COUNT=1' '-DENABLE_MINOR_MC' '-DOBJECT_PRINT' '-DV8_CONCURRENT_MARKING' '-DV8_ARRAY_BUFFER_EXTENSION' '-DV8_ENABLE_LAZY_SOURCE_POSITIONS' '-DV8_USE_SIPHASH' '-DDISABLE_UNTRUSTED_CODE_MITIGATIONS' '-DV8_WIN64_UNWINDING_INFO' '-DV8_ENABLE_REGEXP_INTERPRETER_THREADED_DISPATCH' '-DV8_SNAPSHOT_COMPRESSION' '-DBUILDING_V8_BASE_SHARED' -I/home/builder/lede_x86/staging_dir/target-x86_64_musl/usr/include -I../deps/v8 -I../deps/v8/include  -pthread -Wno-unused-parameter -m64 -Wno-return-type -fno-strict-aliasing -m64 -O3 -fno-omit-frame-pointer -fdata-sections -ffunction-sections -O3 -fno-rtti -fno-exceptions -std=gnu++1y -MMD -MF /home/builder/lede_x86/build_dir/target-x86_64_musl/node-v14.21.3/out/Release/.deps//home/builder/lede_x86/build_dir/target-x86_64_musl/node-v14.21.3/out/Release/obj.host/v8_libbase/deps/v8/src/base/cpu.o.d.raw   -c
In file included from ../deps/v8/src/base/macros.h:12,
                 from ../deps/v8/src/base/cpu.h:17,
                 from ../deps/v8/src/base/cpu.cc:5:
../deps/v8/src/base/logging.h: In function 'typename std::enable_if<(std::is_enum<_Tp>::value && (! v8::base::has_output_operator<T>::value)), std::__cxx11::basic_string<char> >::type v8::base::PrintCheckOperand(T)':
../deps/v8/src/base/logging.h:170:34: error: 'uint8_t' was not declared in this scope
  170 |       std::is_same<underlying_t, uint8_t>::value, uint16_t,
      |                                  ^~~~~~~
../deps/v8/src/base/logging.h:16:1: note: 'uint8_t' is defined in header '<cstdint>'; did you forget to '#include <cstdint>'?
   15 | #include "src/base/template-utils.h"
  +++ |+#include <cstdint>
   16 |
../deps/v8/src/base/logging.h:170:41: error: template argument 2 is invalid
  170 |       std::is_same<underlying_t, uint8_t>::value, uint16_t,
      |                                         ^
../deps/v8/src/base/logging.h:170:51: error: 'uint16_t' was not declared in this scope
  170 |       std::is_same<underlying_t, uint8_t>::value, uint16_t,
      |                                                   ^~~~~~~~
../deps/v8/src/base/logging.h:170:51: note: 'uint16_t' is defined in header '<cstdint>'; did you forget to '#include <cstdint>'?
../deps/v8/src/base/logging.h:172:61: error: template argument 1 is invalid
  172 |                                 int16_t, underlying_t>::type>::type;
      |                                                             ^
../deps/v8/src/base/logging.h:172:61: error: template argument 2 is invalid
../deps/v8/src/base/logging.h:172:62: error: '<declaration error>' is not a template [-fpermissive]
  172 |                                 int16_t, underlying_t>::type>::type;
      |                                                              ^~
../deps/v8/src/base/logging.h:173:40: error: 'int_t' does not name a type; did you mean 'wint_t'?
  173 |   return PrintCheckOperand(static_cast<int_t>(static_cast<underlying_t>(val)));
      |                                        ^~~~~
      |                                        wint_t
../deps/v8/src/base/macros.h: At global scope:
../deps/v8/src/base/macros.h:338:25: error: 'intptr_t' has not been declared
  338 | inline T RoundDown(T x, intptr_t m) {
      |                         ^~~~~~~~
../deps/v8/src/base/macros.h:344:11: error: 'intptr_t' has not been declared
  344 | template <intptr_t m, typename T>
      |           ^~~~~~~~
../deps/v8/src/base/macros.h: In function 'constexpr T RoundDown(T)':
../deps/v8/src/base/macros.h:348:17: error: 'm' was not declared in this scope
  348 |   STATIC_ASSERT(m != 0 && ((m & (m - 1)) == 0));
      |                 ^
../deps/v8/src/base/macros.h:200:43: note: in definition of macro 'STATIC_ASSERT'
  200 | #define STATIC_ASSERT(test) static_assert(test, #test)
      |                                           ^~~~
../deps/v8/src/base/macros.h: At global scope:
../deps/v8/src/base/macros.h:354:23: error: 'intptr_t' has not been declared
  354 | inline T RoundUp(T x, intptr_t m) {
      |                       ^~~~~~~~
../deps/v8/src/base/macros.h:358:11: error: 'intptr_t' has not been declared
  358 | template <intptr_t m, typename T>
      |           ^~~~~~~~
../deps/v8/src/base/macros.h: In function 'constexpr T RoundUp(T)':
../deps/v8/src/base/macros.h:361:20: error: 'm' was not declared in this scope
  361 |   return RoundDown<m, T>(static_cast<T>(x + (m - 1)));
      |                    ^
../deps/v8/src/base/macros.h: In function 'void* AlignedAddress(void*, size_t)':
../deps/v8/src/base/macros.h:372:51: error: 'uintptr_t' does not name a type
  372 |   return reinterpret_cast<void*>(reinterpret_cast<uintptr_t>(address) &
      |                                                   ^~~~~~~~~
../deps/v8/src/base/macros.h:13:1: note: 'uintptr_t' is defined in header '<cstdint>'; did you forget to '#include <cstdint>'?
   12 | #include "src/base/logging.h"
  +++ |+#include <cstdint>
   13 |
../deps/v8/src/base/macros.h:373:47: error: 'uintptr_t' does not name a type
  373 |                                  ~static_cast<uintptr_t>(alignment - 1));
      |                                               ^~~~~~~~~
../deps/v8/src/base/macros.h:373:47: note: 'uintptr_t' is defined in header '<cstdint>'; did you forget to '#include <cstdint>'?
make[5]: *** [tools/v8_gypfiles/v8_libbase.host.mk:159: /home/builder/lede_x86/build_dir/target-x86_64_musl/node-v14.21.3/out/Release/obj.host/v8_libbase/deps/v8/src/base/cpu.o] Error 1
make[4]: *** [Makefile:109: node] Error 2
make[4]: Leaving directory '/home/builder/lede_x86/build_dir/target-x86_64_musl/node-v14.21.3'
make[3]: *** [Makefile:225: /home/builder/lede_x86/build_dir/target-x86_64_musl/node-v14.21.3/.built] Error 2
make[3]: Leaving directory '/home/builder/lede_x86/feeds/packages/lang/node'
time: package/feeds/packages/node/compile#147.70#51.01#195.18
    ERROR: package/feeds/packages/node failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/node/compile] Error 1
make[2]: Leaving directory '/home/builder/lede_x86'
make[1]: *** [package/Makefile:110: /home/builder/lede_x86/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/builder/lede_x86'
make: *** [/home/builder/lede_x86/include/toplevel.mk:231：world] 错误 2
builder@Build-Server:/home/builder/lede_x86$
builder@Build-Server:/home/builder/lede_x86$
builder@Build-Server:/home/builder/lede_x86$

ChaingTsung, 2024-05-03:
+1同样的问题

GreatMichaelLee, 2024-05-03:
> +1同样的问题

我还以为真的是我的环境问题呢，看来不是啊！

GreatMichaelLee, 2024-05-03:
PS: 网上好像说跟gcc版本有关，我的是13.2，这里有个issue不知道是不是跟我的这个相关。

https://github.com/envoyproxy/envoy/pull/29425

hzjnet, 2024-05-03:
源码切换成node 18试试

GreatMichaelLee, 2024-05-04:
> 源码切换成node 18试试

不懂怎么切换 -_-!

GreatMichaelLee, 2024-05-04:
> 源码切换成node 18试试

我看网上是推荐切换到node 20. 刚试了把gcc alter成gcc 11然后make clean && make也还是报错，见了鬼，我把上面那几个fix apply试试

GreatMichaelLee, 2024-05-04:
搞不定， apply了这两个fix，把cstdint头文件包进去，之前的error没有了
[image]
[image]
[image]

但不料想结果又报这个模板定义错
[image]

按网上说的apply这个fix （不确定这个是否可用于Node)
[image]
[image]

结果又报这个，......看意思是说disjunction这个玩意要指定C++17 以上版本，这个就真不知道怎么搞了....水平太菜，求大佬帮忙。
[image]

ChaingTsung, 2024-05-04:
node20更不行，https://github.com/openwrt/openwrt/issues/15373看我的这个。

GreatMichaelLee, 2024-05-04:
> node20更不行，https://github.com/openwrt/openwrt/issues/15373看我的这个。

你这个感觉是编译机的问题

closed 2024-05-05 by coolsnowwolf

WYC-2020, 2024-05-06:
[001-fix-build-with-gcc13.patch](https://github.com/coolsnowwolf/lede/files/15217141/001-fix-build-with-gcc13.patch)
[image]
HOST_CXXFLAGS += -fpermissive
老规矩自己替换 @GreatMichaelLee