coolsnowwolf/lede #12122: 5.15.157报错

ywt114, 2024-05-03:
### 详细叙述

net/bridge/br_input.c:127:24: error: too few arguments to function 'br_pass_frame_up'

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

2024-05-03T10:04:03.0006556Z make[5]: Entering directory '/home/runner/work/test/test/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.157'
2024-05-03T10:04:03.4060476Z net/sched/Kconfig:45: warning: menuconfig statement without prompt
2024-05-03T10:04:03.8628930Z #
2024-05-03T10:04:03.8629433Z # No change to .config
2024-05-03T10:04:03.8629944Z #
2024-05-03T10:04:03.8659259Z make[5]: Leaving directory '/home/runner/work/test/test/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.157'
2024-05-03T10:04:03.8665487Z make -C /home/runner/work/test/test/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.157 KCFLAGS="-fmacro-prefix-map=/home/runner/work/test/test/openwrt/build_dir/target-x86_64_musl=target-x86_64_musl -fno-caller-saves " HOSTCFLAGS="-O2 -I/home/runner/work/test/test/openwrt/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="x86_64-openwrt-linux-musl-" ARCH="x86" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="test" KBUILD_BUILD_HOST="test" KBUILD_BUILD_TIMESTAMP="Thu May  2 12:01:21 2024" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/home/runner/work/test/test/openwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRELEASE=5.15.157 CC="x86_64-openwrt-linux-musl-gcc" bzImage modules
2024-05-03T10:04:03.8693752Z make[5]: Entering directory '/home/runner/work/test/test/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.157'
2024-05-03T10:04:04.3160232Z   CALL    scripts/checksyscalls.sh
2024-05-03T10:04:04.3169008Z   CALL    scripts/atomic/check-atomics.sh
2024-05-03T10:04:04.3461185Z   DESCEND objtool
2024-05-03T10:04:04.5130815Z   CHK     include/generated/compile.h
2024-05-03T10:04:14.7072405Z   CC      net/bridge/br_input.o
2024-05-03T10:04:15.1633656Z net/bridge/br_input.c: In function 'br_handle_frame_finish':
2024-05-03T10:04:15.1635080Z net/bridge/br_input.c:127:24: error: too few arguments to function 'br_pass_frame_up'
2024-05-03T10:04:15.1636186Z   127 |                 return br_pass_frame_up(skb);
2024-05-03T10:04:15.1636905Z       |                        ^~~~~~~~~~~~~~~~
2024-05-03T10:04:15.1637602Z net/bridge/br_input.c:34:12: note: declared here
2024-05-03T10:04:15.1638493Z    34 | static int br_pass_frame_up(struct sk_buff *skb, bool promisc)
2024-05-03T10:04:15.1639324Z       |            ^~~~~~~~~~~~~~~~
2024-05-03T10:04:15.1829861Z make[7]: *** [scripts/Makefile.build:289: net/bridge/br_input.o] Error 1
2024-05-03T10:04:15.1832665Z make[6]: *** [scripts/Makefile.build:552: net/bridge] Error 2
2024-05-03T10:04:15.1834925Z make[5]: *** [Makefile:1907: net] Error 2
2024-05-03T10:04:15.1836517Z make[5]: Leaving directory '/home/runner/work/test/test/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.157'
2024-05-03T10:04:15.1839119Z make[4]: *** [Makefile:28: /home/runner/work/test/test/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.157/.modules] Error 2
2024-05-03T10:04:15.1840619Z make[4]: Leaving directory '/home/runner/work/test/test/openwrt/target/linux/x86'
2024-05-03T10:04:15.1841476Z make[3]: *** [Makefile:11: compile] Error 2
2024-05-03T10:04:15.1842098Z make[3]: Leaving directory '/home/runner/work/test/test/openwrt/target/linux'
2024-05-03T10:04:15.1843905Z time: target/linux/compile#8.92#3.66#12.49
2024-05-03T10:04:15.1851808Z     ERROR: target/linux failed to build.
2024-05-03T10:04:15.1855803Z make[2]: *** [target/Makefile:30: target/linux/compile] Error 1
2024-05-03T10:04:15.1860869Z make[2]: Leaving directory '/home/runner/work/test/test/openwrt'
2024-05-03T10:04:15.1890248Z make[1]: *** [target/Makefile:23: /home/runner/work/test/test/openwrt/staging_dir/target-x86_64_musl/stamp/.target_compile] Error 2
2024-05-03T10:04:15.1896905Z make[1]: Leaving directory '/home/runner/work/test/test/openwrt'
2024-05-03T10:04:15.1928017Z make: *** [/home/runner/work/test/test/openwrt/include/toplevel.mk:231: world] Error 2
2024-05-03T10:04:15.1943827Z ##[error]Process completed with exit code 2.

Nicepaul, 2024-05-04:
同样报错
net/bridge/br_input.c:34:12: note: declared here
   34 | static int br_pass_frame_up(struct sk_buff *skb, bool promisc)
      |            ^~~~~~~~~~~~~~~~
make[7]: *** [scripts/Makefile.build:289: net/bridge/br_input.o] Error 1
make[6]: *** [scripts/Makefile.build:552: net/bridge] Error 2
make[5]: *** [Makefile:1907: net] Error 2
make[5]: Leaving directory '/home/julien/360t7/lede/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/linux-5.15.157'
make[4]: *** [Makefile:21: /home/julien/360t7/lede/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/linux-5.15.157/.modules] Error 2
make[4]: Leaving directory '/home/julien/360t7/lede/target/linux/mediatek'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/home/julien/360t7/lede/target/linux'
time: target/linux/compile#2015.54#242.18#2284.30
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/compile] Error 1
make[2]: Leaving directory '/home/julien/360t7/lede'
make[1]: *** [target/Makefile:23: /home/julien/360t7/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/julien/360t7/lede'
make: *** [/home/julien/360t7/lede/include/toplevel.mk:231: world] Error 2

bi4tms, 2024-05-04:
同样报错 @Beginner-Go

closed 2024-05-05 by coolsnowwolf

WYC-2020, 2024-05-06:
如果不回退的话看这里 https://github.com/coolsnowwolf/lede/commit/23a35d1ef0971f50251552c21c12a75e1441c4ce#commitcomment-141702403
是能编译过，我怀疑就是吧打patch的函数搞错了

WYC-2020, 2024-05-06:
> 同样报错 net/bridge/br_input.c:34:12: note: declared here 34 | static int br_pass_frame_up(struct sk_buff *skb, bool promisc) | ^~~~~~~~~~~~~~~~ make[7]: *** [scripts/Makefile.build:289: [...]

[image]
不回退内核的话 这样改

ywt114, 2024-05-10:
几天前 lean已经修复了 还是麻烦@WYC-2020费心了