coolsnowwolf/lede #12124: 第一次编译时卡在下载，手动下载却hash不符，疑似新的投毒？

hugeWrongMistake, 2024-05-03:
### 详细叙述

新人第一次编译，make download 一切正常，在make -j1 V=s卡住，发现是下载argp-standalone-1.3.tar.gz。于是手动下载这个压缩包放入dl文件夹中，再次make却报哈希不符。

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

RT-ACRH17

### 详细日志

SHELL= flock /home/aa/lede/tmp/.argp-standalone-1.3.tar.gz.flock -c '   /home/aa/lede/scripts/download.pl "/home/aa/lede/dl" "argp-standalone-1.3.tar.gz" "dec79694da1319acd2238ce95df57f3680fea2482096e483323fddf3d818d8be" "" "http://www.lysator.liu.se/~nisse/misc/"    '
Hash of the local file argp-standalone-1.3.tar.gz does not match (file: 6982d2cf387b20eed8eee8aa0deac213b60f45a6a9005b1c9d1bfa770ed785f4, requested: dec79694da1319acd2238ce95df57f3680fea2482096e483323fddf3d818d8be) - deleting download.
make[3]: *** [Makefile:61: /home/aa/lede/dl/argp-standalone-1.3.tar.gz] Error 255
make[3]: Leaving directory '/home/aa/lede/package/libs/argp-standalone'
time: package/libs/argp-standalone/compile#0.19#0.09#0.24
    ERROR: package/libs/argp-standalone failed to build.
make[2]: *** [package/Makefile:116: package/libs/argp-standalone/compile] Error 1
make[2]: Leaving directory '/home/aa/lede'
make[1]: *** [package/Makefile:110: /home/aa/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/aa/lede'
make: *** [/home/aa/lede/include/toplevel.mk:231: world] Error 2

hugeWrongMistake, 2024-05-03:
[argp-standalone-1.3.tar.gz](https://github.com/coolsnowwolf/lede/files/15201946/argp-standalone-1.3.tar.gz)
这是我这边下载的此压缩包，大佬看看哪里不对了

wlshdjj, 2024-05-04:
官方改用了apk包构建，hash校验与之前的ipk包构建的不同，自己用改成skip就算了

hugeWrongMistake, 2024-05-04:
奇怪的地方是，官方网站显示这个文件的最后修改时间是2004-02-23 ，根据历史提交，这个hash很久以前就是了。也就是说，这个文件的最后修改时间没变，但hash变了，这点不可疑吗？
而且官网上https://www.[lysator.liu.se/~nisse/misc/argp-standalone-1.3.tar.gz.asc](https://www.lysator.liu.se/~nisse/misc/argp-standalone-1.3.tar.gz.asc)这个文件目前无法下载。

wlshdjj, 2024-05-04:
我就是说的现在官方要使用apk构建，hash检验和以往不同，源码包还是以前的包，但hash不是以前的hash，不单止这一个包，很多package的hash都被改了，也就有人疑问这个hash是怎么得来的，和自己生成的根本不同

获取Outlook for Android<https://aka.ms/AAb9ysg>

而且官网上https://www.[lysator.liu.se/~nisse/misc/argp-standalone-1.3.tar.gz.asc](https://www.lysator.liu.se/~nisse/misc/argp-standalone-1.3.tar.gz.asc)这个文件目前无法下载。<https://www.%5Blysator.liu.se/~nisse/misc/argp-standalone-1.3.tar.gz.asc%5D(https://www.lysator.liu.se/~nisse/misc/argp-standalone-1.3.tar.gz.asc)%E8%BF%99%E4%B8%AA%E6%96%87%E4%BB%B6%E7%9B%AE%E5%89%8D%E6%97%A0%E6%B3%95%E4%B8%8B%E8%BD%BD%E3%80%82>

wlshdjj, 2024-05-04:
> 奇怪的地方是，官方网站显示这个文件的最后修改时间是2004-02-23 ，根据历史提交，这个hash很久以前就是了。也就是说，这个文件的最后修改时间没变，但hash变了，这点不可疑吗？ 而且官网上 https://www.lysator.liu.se/~nisse/misc/argp-standalone-1.3.tar.gz.asc 这个文件目前无法下载。

我就是说的现在官方要使用apk构建，hash检验和以往不同，源码包还是以前的包，但hash不是以前的hash，不单止这一个包，很多package的hash都被改了，也就有人疑问这个hash是怎么得来的，和自己生成的根本不同，下面是两个改动最多的commit：
https://git.nju.edu.cn/nju/openwrt/-/commit/c02a2db05e941c49ba3d073f537c2d101c7e48b0
https://git.nju.edu.cn/nju/openwrt/-/commit/e8725a932e16eaf6ec51add8c084d959cbe32ff2

hugeWrongMistake, 2024-05-05:
在github上找到一个dl备份的仓库：https://github.com/TOLQY/openwrt-lede-dl
经过对比之后可以确定文件变了。备份的argp-standalone-1.3.tar.gz大小为130,255 字节，现在下载的大小为573,440 字节。
这么明显的文件大小差别我居然没发现？人家网站上直接写了
argp-standalone-1.3.tar.gz  	127.2 kB  	application/gzip  	2004-02-23
经过右键属性，发现是压缩率变了导致的文件大小和hash变化。现在下载的压缩率是100%，但文件内容没有变化。
请问这作者在搞什么？悄无声息的重新打包？重新打包就算了，还往大了压？
服了，现在哪里有能正常编译的dl文件夹镜像？

WYC-2020, 2024-05-06:
说这么多 为啥我这边一切正常，并且并不是官方没法下载 是你自己链接就不对
[image]
https://www.lysator.liu.se/~nisse/misc/argp-standalone-1.3.tar.gz.asc 长点心吧 为啥最后有一个asc

hugeWrongMistake, 2024-05-06:
> 说这么多 为啥我这边一切正常，并且并不是官方没法下载 是你自己链接就不对 [...]

我是说无asc的hash不对，顺带提了一句asc无法下载。
话说为什么你的脚本里有镜像链接？sources.openwrt.org

WYC-2020, 2024-05-06:
> > 说这么多 为啥我这边一切正常，并且并不是官方没法下载 是你自己链接就不对 ![image](https://private-user-
> 
> 我是说无asc的hash不对，顺带提了一句asc无法下载。

哪里不对自己操作不对吧
[image]
[image]
 谁没事回去投毒哦，并且asc 也是能下载的 ，还是多分析分析自己网络环境以及检查hash的方式吧

> 我是说无asc的hash不对，顺带提了一句asc无法下载。

[image]
[image]
asc是可以下载的，并且那个包源码下载地址 会有好几个 他们不同是很正常的，以前qt经常会这样 ，这又啥好奇怪的，openwrt下载地址是有一个优先级的，只有前几个无法下载的时候 才会继续从下一个地址下载，多检查自己的网络环境，十有八九是你自己网络没法从前几个网址下载下来 然后重拾到这个网址去下载，下载下来hash不对不是很正常，自己如果没法改网那就自己吧hash skip掉，别再问我为啥这几个网址的会不同了那你只能去问他们收集的了，别纠结这个

WYC-2020, 2024-05-06:
顺便再和你说下 这些更新hash的 基本都是基于能出去的方式下载的包 进行的hash更新 如果你自己不是基于这个 而是从国内的网址下载的包 遇到这种情况在正常不过