coolsnowwolf/lede #12134: 划分vlan导致pppoe拨号失败

hugeWrongMistake, 2024-05-05:
### 详细叙述

我这边拨号无需设置vlan id，wan口为默认的eth1。

第0种情况：无其他vlan时能正常拨号。（仅默认的vlan1，4LAN+CPU (eth0)五个端口全为未标记）
第1种情况：添加vlan但不绑定任何端口时正常。（五个端口全设为关）
滴2种情况：例如新的vlan号为2，将一个 LAN 口 vlan 1 设为关，vlan 2 设为未标记，拨号超时。
第3种情况：五个端口 vlan 1 全为未标记，任意端口的 vlan 2 设为已标记，拨号超时。

难道是我玩vlan的姿势不对？

固件版本：	OpenWrt R24.4.4 / LuCI Master (git-24.119.32332-a710a25)

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

RT-ACRH17

### 详细日志

系统日志：
Mon May  6 01:36:40 2024 daemon.notice netifd: wan (23318): uci: Entry not found
Mon May  6 01:36:40 2024 daemon.notice netifd: wan (23318): sh: out of range
Mon May  6 01:36:40 2024 daemon.info pppd[23352]: Plugin pppoe.so loaded.
Mon May  6 01:36:40 2024 daemon.info pppd[23352]: PPPoE plugin from pppd 2.4.9
Mon May  6 01:36:40 2024 daemon.notice pppd[23352]: pppd 2.4.9 started by root, uid 0
Mon May  6 01:36:55 2024 daemon.warn pppd[23352]: Timeout waiting for PADO packets
Mon May  6 01:36:55 2024 daemon.err pppd[23352]: Unable to complete PPPoE Discovery
Mon May  6 01:36:55 2024 daemon.info pppd[23352]: Exit.

coolsnowwolf, 2026-09-05:
从日志看 PPPoE 在 VLAN 修改后超时等待 PADO，说明二层 VLAN 配置已将 WAN/CPU 上联链路隔离；PPPoE 本身与 VLAN 兼容，交换机端口的 tagged/untagged、CPU port VLAN 成员及 PVID 必须保持正确。RT-ACRH17 使用旧 swconfig 配置，任意修改 VLAN1/新增 VLAN2 都可能移除 WAN 所需的 CPU 端口成员。当前主线没有针对该特定配置的通用修复，建议恢复 WAN 所在 VLAN 的 CPU tagged/端口 PVID 后再配置新 VLAN。该旧配置咨询 issue 现关闭。

closed 2026-09-05 by coolsnowwolf

llfj, 2026-10-05:
我的竞斗云 也碰到这样的问题，以前是eth1是整合在交换机里面的，也就是说有CPU  1 2 3 4 5.然后5单独作为一个VLAN为WAN，1~4单独做VLAN为LAN，可以正常使用。我编译了[LuCI openwrt-25.12 branch (26.277.30465~de2410c)](https://github.com/openwrt/luci) / [OpenWrt 24.10.5 ()](https://openwrt.org/)，这个版本的，现在是交换机里面是没有WAN这个口的，wan端口单独是eth1. 如果我在交换机中配置VLAN，WAN端口就不能使用，PPPOE无法拨号，设置DHCP也不能获取上游的IP地址。但是把交换机中的Vlan删除了，就可以正常拨号了。

llfj, 2026-10-05:
还有这个版本的多拨助手 也不能创建虚拟接口，开了多拨和没开多拨是一样的