coolsnowwolf/lede #12136: 最新源码mac80211报错

dragon888888888888, 2024-05-06:
### 详细叙述

因为之前编译过，今天重编译报错，所以重拉源码还是报错

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

光影猫

### 详细日志

make[5]: 'Kconfig.versions' is up to date.
make[7]: 'Kconfig.versions' is up to date.
make[8]: 'conf' is up to date.
boolean symbol CRYPTO_LIB_ARC4 tested for 'm'? test forced to 'n'
#
# configuration written to .config
#
Building backport-include/backport/autoconf.h ... done.
  CC [M]  /home/king/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-6.1.24/drivers/net/wireless/mac80211_hwsim.o
In file included from /home/king/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-6.1.24/backport-include/linux/export.h:6,
                 from ./include/linux/linkage.h:7,
                 from ./include/linux/preempt.h:10,
                 from ./include/linux/spinlock.h:56,
                 from /home/king/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-6.1.24/backport-include/linux/spinlock.h:3,
                 from ./include/linux/mmzone.h:8,
                 from ./include/linux/gfp.h:7,
                 from /home/king/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-6.1.24/backport-include/linux/gfp.h:3,
                 from ./include/linux/slab.h:16,
                 from /home/king/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-6.1.24/backport-include/linux/slab.h:3,
                 from /home/king/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-6.1.24/drivers/net/wireless/mac80211_hwsim.c:18:
/home/king/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-6.1.24/drivers/net/wireless/mac80211_hwsim.c: In function 'init_mac80211_hwsim':
./include/linux/export.h:29:22: error: passing argument 1 of 'class_create' from incompatible pointer type [-Werror=incompatible-pointer-types]
   29 | #define THIS_MODULE (&__this_module)
      |                     ~^~~~~~~~~~~~~~~
      |                      |
      |                      struct module *
/home/king/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-6.1.24/drivers/net/wireless/mac80211_hwsim.c:5753:36: note: in expansion of macro 'THIS_MODULE'
 5753 |         hwsim_class = class_create(THIS_MODULE, "mac80211_hwsim");
      |                                    ^~~~~~~~~~~
In file included from ./include/linux/device.h:31,
                 from /home/king/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-6.1.24/backport-include/linux/device.h:3,
                 from ./include/linux/dma-mapping.h:8,
                 from /home/king/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-6.1.24/backport-include/linux/dma-mapping.h:3,
                 from ./include/linux/skbuff.h:28,
                 from /home/king/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-6.1.24/backport-include/linux/skbuff.h:3,
                 from ./include/net/net_namespace.h:43,
                 from ./include/linux/netdevice.h:38,
                 from /home/king/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-6.1.24/backport-include/linux/netdevice.h:3,
                 from ./include/net/dst.h:13,
                 from /home/king/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-6.1.24/drivers/net/wireless/mac80211_hwsim.c:20:
./include/linux/device/class.h:231:54: note: expected 'const char *' but argument is of type 'struct module *'
  231 | struct class * __must_check class_create(const char *name);
      |                                          ~~~~~~~~~~~~^~~~
/home/king/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-6.1.24/drivers/net/wireless/mac80211_hwsim.c:5753:23: error: too many arguments to function 'class_create'
 5753 |         hwsim_class = class_create(THIS_MODULE, "mac80211_hwsim");
      |                       ^~~~~~~~~~~~
In file included from ./include/linux/device.h:31,
                 from /home/king/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-6.1.24/backport-include/linux/device.h:3,
                 from ./include/linux/dma-mapping.h:8,
                 from /home/king/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-6.1.24/backport-include/linux/dma-mapping.h:3,
                 from ./include/linux/skbuff.h:28,
                 from /home/king/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-6.1.24/backport-include/linux/skbuff.h:3,
                 from ./include/net/net_namespace.h:43,
                 from ./include/linux/netdevice.h:38,
                 from /home/king/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-6.1.24/backport-include/linux/netdevice.h:3,
                 from ./include/net/dst.h:13,
                 from /home/king/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-6.1.24/drivers/net/wireless/mac80211_hwsim.c:20:
./include/linux/device/class.h:231:29: note: declared here
  231 | struct class * __must_check class_create(const char *name);
      |                             ^~~~~~~~~~~~
cc1: some warnings being treated as errors
make[9]: *** [scripts/Makefile.build:243: /home/king/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-6.1.24/drivers/net/wireless/mac80211_hwsim.o] Error 1
make[8]: *** [scripts/Makefile.build:480: /home/king/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-6.1.24/drivers/net/wireless] Error 2
make[7]: *** [Makefile:1913: /home/king/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-6.1.24] Error 2
make[6]: *** [Makefile.build:13: modules] Error 2
make[5]: *** [Makefile.real:93: modules] Error 2
make[4]: *** [Makefile:121: modules] Error 2
make[4]: Leaving directory '/home/king/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-6.1.24'
make[3]: *** [Makefile:401: /home/king/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-6.1.24/.built] Error 2
make[3]: Leaving directory '/home/king/lede/package/kernel/mac80211'
time: package/kernel/mac80211/regular/compile#2.05#1.75#3.87
    ERROR: package/kernel/mac80211 failed to build (build variant: regular).
make[2]: *** [package/Makefile:116: package/kernel/mac80211/compile] Error 1
make[2]: Leaving directory '/home/king/lede'
make[1]: *** [package/Makefile:110: /home/king/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/king/lede'
make: *** [/home/king/lede/include/toplevel.mk:231：world] 错误 2
abc@abc-virtual-machine:~/lede$

WYC-2020, 2024-05-07:
刚试了下 x86没问题呢  编译的mac80211_hwsim模块，你那个平台没测 理论上这边没问题 应该其他平台也没问题的 ，这个模块是通用 的

qingtian110, 2024-05-08:
[image]

没问题

dragon888888888888, 2024-05-10:
> ![image](https://private-user-images.githubusercontent.com/87856890/328909878-b8cfec8b-2500-4118-ba6a-a4509f00ffa9.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

In file included from /home/abc/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-6.1.24/net/mac80211/pm.c:12:
/home/abc/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-6.1.24/net/mac80211/led.h: In function 'ieee80211_led_rx':
/home/abc/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-6.1.24/net/mac80211/led.h:20:51: warning: passing argument 2 of 'led_trigger_blink_oneshot' makes integer from pointer without a cast [-Wint-conversion]
   20 |         led_trigger_blink_oneshot(&local->rx_led, &led_delay, &led_delay, 0);
      |                                                   ^~~~~~~~~~
      |                                                   |
      |                                                   long unsigned int *
In file included from /home/abc/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-6.1.24/backport-include/linux/leds.h:3,
                 from ./include/linux/rfkill.h:36,
                 from /home/abc/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-6.1.24/backport-include/linux/rfkill.h:3,
                 from /home/abc/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-6.1.24/include/net/cfg80211.h:29,
                 from /home/abc/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-6.1.24/include/net/mac80211.h:22,
                 from /home/abc/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-6.1.24/net/mac80211/pm.c:6:
./include/linux/leds.h:512:46: note: expected 'long unsigned int' but argument is of type 'long unsigned int *'
  512 |                                unsigned long delay_on,
      |                                ~~~~~~~~~~~~~~^~~~~~~~
In file included from /home/abc/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-6.1.24/net/mac80211/pm.c:12:
/home/abc/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-6.1.24/net/mac80211/led.h:20:63: warning: passing argument 3 of 'led_trigger_blink_oneshot' makes integer from pointer without a cast [-Wint-conversion]
   20 |         led_trigger_blink_oneshot(&local->rx_led, &led_delay, &led_delay, 0);
      |                                                               ^~~~~~~~~~
      |                                                               |
      |                                                               long unsigned int *
In file included from /home/abc/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-6.1.24/backport-include/linux/leds.h:3,
                 from ./include/linux/rfkill.h:36,
                 from /home/abc/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-6.1.24/backport-include/linux/rfkill.h:3,
                 from /home/abc/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-6.1.24/include/net/cfg80211.h:29,
                 from /home/abc/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-6.1.24/include/net/mac80211.h:22,
                 from /home/abc/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-6.1.24/net/mac80211/pm.c:6:
./include/linux/leds.h:513:46: note: expected 'long unsigned int' but argument is of type 'long unsigned int *'
  513 |                                unsigned long delay_off,
      |                                ~~~~~~~~~~~~~~^~~~~~~~~
In file included from /home/abc/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-6.1.24/net/mac80211/pm.c:12:
/home/abc/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-6.1.24/net/mac80211/led.h: In function 'ieee80211_led_tx':
/home/abc/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-6.1.24/net/mac80211/led.h:31:51: warning: passing argument 2 of 'led_trigger_blink_oneshot' makes integer from pointer without a cast [-Wint-conversion]
   31 |         led_trigger_blink_oneshot(&local->tx_led, &led_delay, &led_delay, 0);
      |                                                   ^~~~~~~~~~
      |                                                   |
      |                                                   long unsigned int *
In file included from /home/abc/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-6.1.24/backport-include/linux/leds.h:3,
                 from ./include/linux/rfkill.h:36,
                 from /home/abc/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-6.1.24/backport-include/linux/rfkill.h:3,
                 from /home/abc/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-6.1.24/include/net/cfg80211.h:29,
                 from /home/abc/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-6.1.24/include/net/mac80211.h:22,
                 from /home/abc/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-6.1.24/net/mac80211/pm.c:6:
./include/linux/leds.h:512:46: note: expected 'long unsigned int' but argument is of type 'long unsigned int *'
  512 |                                unsigned long delay_on,
[... cut ...]
time: package/kernel/mac80211/regular/compile#283.75#46.62#57.52
    ERROR: package/kernel/mac80211 failed to build (build variant: regular).
make[2]: *** [package/Makefile:116: package/kernel/mac80211/compile] Error 1
make[2]: Leaving directory '/home/abc/lede'
make[1]: *** [package/Makefile:110: /home/abc/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/abc/lede'
make: *** [/home/abc/lede/include/toplevel.mk:231：world] 错误 2
只要选了它就报错

mahakalaya, 2025-08-11:
make[10]: *** [scripts/Makefile.build:289: /home/simon/lede/build_dir/target-aarch64_cortex-a53_musl/linux-amlogic_mesongx/mac80211-regular/backports-6.6.15/drivers/net/wireless/ath/ath10k/core.o] Error 1
make[9]: *** [scripts/Makefile.build:552: /home/simon/lede/build_dir/target-aarch64_cortex-a53_musl/linux-amlogic_mesongx/mac80211-regular/backports-6.6.15/drivers/net/wireless/ath/ath10k] Error 2
make[8]: *** [scripts/Makefile.build:552: /home/simon/lede/build_dir/target-aarch64_cortex-a53_musl/linux-amlogic_mesongx/mac80211-regular/backports-6.6.15/drivers/net/wireless/ath] Error 2
make[7]: *** [scripts/Makefile.build:552: /home/simon/lede/build_dir/target-aarch64_cortex-a53_musl/linux-amlogic_mesongx/mac80211-regular/backports-6.6.15/drivers/net/wireless] Error 2
make[6]: *** [Makefile:1926: /home/simon/lede/build_dir/target-aarch64_cortex-a53_musl/linux-amlogic_mesongx/mac80211-regular/backports-6.6.15] Error 2
make[5]: *** [Makefile.build:13: modules] Error 2
make[4]: *** [Makefile.real:105: modules] Error 2
make[3]: *** [Makefile:121: modules] Error 2
make[2]: *** [Makefile:411: /home/simon/lede/build_dir/target-aarch64_cortex-a53_musl/linux-amlogic_mesongx/mac80211-regular/backports-6.6.15/.built] Error 2
make[2]: Leaving directory '/home/simon/lede/package/kernel/mac80211'
time: package/kernel/mac80211/regular/compile#79.81#22.73#106.86
    ERROR: package/kernel/mac80211 failed to build (build variant: regular).
make[1]: *** [package/Makefile:116: package/kernel/mac80211/compile] Error 1
make[1]: Leaving directory '/home/simon/lede'
make: *** [/home/simon/lede/include/toplevel.mk:231：package/mac80211/compile] 错误 2

N1最新源码编译也是出现这个错误

coolsnowwolf, 2026-09-05:
当前 LEDE 已使用 backports 6.6.15（旧内核路径），其中 `mac80211_hwsim.c` 已调用新内核兼容形式 `class_create("mac80211_hwsim")`，`net/mac80211/led.h` 也已使用整数参数调用 `led_trigger_blink_oneshot()`。这正是原日志中 `class_create(THIS_MODULE, ...)` 和 LED 指针参数错误的修复方向；原 issue 使用的 backports 6.1.24 已过时。当前 x86_64 hostapd/mac80211 相关构建环境正常，未发现现行源码中的同一错误。该 issue 已解决。

closed 2026-09-05 by coolsnowwolf