coolsnowwolf/lede #12137: H68k最新固件掉硬盘

dgyangxf2023, 2024-05-07:
### 详细叙述

从23年8月之后编译的版本，使用USB移动硬盘读写大文件时会导致掉盘，USB硬盘盒外置供电，不存在供电不足的情况，刷回老版可以正常使用。感觉像是samba4的问题，用ftp测试不会掉盘。

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

HiLink h68k

### 详细日志

Tue May  7 10:08:37 2024 kern.info kernel: [112079.885554] usb 2-1: USB disconnect, device number 2
Tue May  7 10:08:37 2024 kern.info kernel: [112079.886151] sd 0:0:0:0: [sda] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x07 driverbyte=DRIVER_OK cmd_age=0s
Tue May  7 10:08:37 2024 kern.info kernel: [112079.887037] sd 0:0:0:0: [sda] tag#0 CDB: opcode=0x28 28 00 24 46 9c 00 00 08 00 00
Tue May  7 10:08:37 2024 kern.err kernel: [112079.887729] I/O error, dev sda, sector 608607232 op 0x0:(READ) flags 0x80700 phys_seg 32 prio class 2
Tue May  7 10:08:37 2024 kern.info kernel: [112079.889432] sd 0:0:0:0: [sda] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x07 driverbyte=DRIVER_OK cmd_age=0s
Tue May  7 10:08:37 2024 kern.info kernel: [112079.890788] sd 0:0:0:0: [sda] tag#0 CDB: opcode=0x28 28 00 24 46 a4 00 00 04 00 00
Tue May  7 10:08:37 2024 kern.err kernel: [112079.891736] I/O error, dev sda, sector 608609280 op 0x0:(READ) flags 0x84700 phys_seg 64 prio class 2
Tue May  7 10:08:37 2024 kern.err kernel: [112079.893027] device offline error, dev sda, sector 608610304 op 0x0:(READ) flags 0x80700 phys_seg 26 prio class 2
Tue May  7 10:08:37 2024 kern.err kernel: [112079.894016] device offline error, dev sda, sector 608607232 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 2
Tue May  7 10:08:37 2024 kern.err kernel: [112079.894899] device offline error, dev sda, sector 608608256 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 2
Tue May  7 10:08:37 2024 kern.err kernel: [112079.895795] device offline error, dev sda, sector 608612352 op 0x0:(READ) flags 0x80700 phys_seg 27 prio class 2
Tue May  7 10:08:37 2024 kern.err kernel: [112079.896749] device offline error, dev sda, sector 809503312 op 0x1:(WRITE) flags 0x3000 phys_seg 1 prio class 2
Tue May  7 10:08:37 2024 kern.err kernel: [112079.897678] Buffer I/O error on dev sda1, logical block 101187658, lost async page write
Tue May  7 10:08:37 2024 kern.err kernel: [112079.898455] device offline error, dev sda, sector 608612352 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 2
Tue May  7 10:08:37 2024 kern.err kernel: [112079.899442] device offline error, dev sda, sector 608610304 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 2
Tue May  7 10:08:37 2024 kern.err kernel: [112079.900368] device offline error, dev sda, sector 608607232 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 2
Tue May  7 10:08:37 2024 kern.crit kernel: [112080.025429] EXT4-fs error (device sda1): __ext4_find_entry:1682: inode #25296897: comm smbd: reading directory lblock 0
Tue May  7 10:08:37 2024 kern.crit kernel: [112080.026459] EXT4-fs error (device sda1): __ext4_find_entry:1682: inode #25296897: comm smbd: reading directory lblock 0
Tue May  7 10:08:37 2024 kern.warn kernel: [112080.027721] EXT4-fs warning (device sda1): htree_dirblock_to_tree:1082: inode #25296897: lblock 0: comm smbd: error -5 reading directory block
Tue May  7 10:08:37 2024 kern.info kernel: [112080.333504] EXT4-fs (sda2): unmounting filesystem.
Tue May  7 10:08:39 2024 daemon.err smbd[12852]: [2024/05/07 10:08:39.328919,  0] ../../source3/smbd/service.c:167(chdir_current_service)
Tue May  7 10:08:39 2024 daemon.err smbd[12852]:   chdir_current_service: vfs_ChDir(/mnt/sda1) failed: No such file or directory. Current token: uid=0, gid=0, 1 groups: 0
Tue May  7 10:08:39 2024 daemon.err smbd[12852]: [2024/05/07 10:08:39.330975,  0] ../../source3/smbd/service.c:167(chdir_current_service)
Tue May  7 10:08:39 2024 daemon.err smbd[12852]:   chdir_current_service: vfs_ChDir(/mnt/sda1) failed: No such file or directory. Current token: uid=0, gid=0, 1 groups: 0
Tue May  7 10:08:40 2024 kern.info kernel: [112083.486968] docker0: port 4(veth9dceafc) entered blocking state
Tue May  7 10:08:40 2024 kern.info kernel: [112083.487518] docker0: port 4(veth9dceafc) entered disabled state
Tue May  7 10:08:40 2024 kern.info kernel: [112083.488364] device veth9dceafc entered promiscuous mode
Tue May  7 10:08:41 2024 kern.info kernel: [112084.420917] usb 2-1: new SuperSpeed USB device number 3 using xhci-hcd
Tue May  7 10:08:41 2024 kern.info kernel: [112084.443797] usb-storage 2-1:1.0: USB Mass Storage device detected
Tue May  7 10:08:41 2024 kern.info kernel: [112084.445246] scsi host1: usb-storage 2-1:1.0
Tue May  7 10:08:42 2024 daemon.info avahi-daemon[5509]: Joining mDNS multicast group on interface veth9dceafc.IPv6 with address fe80::ecf6:fbff:fecf:65f9.
Tue May  7 10:08:42 2024 daemon.info avahi-daemon[5509]: New relevant interface veth9dceafc.IPv6 for mDNS.
Tue May  7 10:08:42 2024 daemon.info avahi-daemon[5509]: Registering new address record for fe80::ecf6:fbff:fecf:65f9 on veth9dceafc.*.
Tue May  7 10:08:43 2024 kern.err kernel: [112085.789020] Aborting journal on device sda1-8.
Tue May  7 10:08:43 2024 kern.err kernel: [112085.789492] Buffer I/O error on dev sda1, logical block 104890368, lost sync page write
Tue May  7 10:08:43 2024 kern.err kernel: [112085.790234] JBD2: I/O error when updating journal superblock for sda1-8.
Tue May  7 10:08:43 2024 kern.notice kernel: [112086.000728] scsi 1:0:0:0: Direct-Access     ATA      TOSHIBA DT01ACA1 A850 PQ: 0 ANSI: 6
Tue May  7 10:08:43 2024 kern.notice kernel: [112086.024035] sd 1:0:0:0: [sdb] 1953525168 512-byte logical blocks: (1.00 TB/932 GiB)
Tue May  7 10:08:43 2024 kern.notice kernel: [112086.025498] sd 1:0:0:0: [sdb] Write Protect is off
Tue May  7 10:08:43 2024 kern.debug kernel: [112086.025966] sd 1:0:0:0: [sdb] Mode Sense: 43 00 00 00
Tue May  7 10:08:43 2024 kern.notice kernel: [112086.026600] sd 1:0:0:0: [sdb] Write cache: disabled, read cache: enabled, doesn't support DPO or FUA
Tue May  7 10:08:43 2024 kern.info kernel: [112086.068156]  sdb: sdb1 sdb2
Tue May  7 10:08:43 2024 kern.notice kernel: [112086.069104] sd 1:0:0:0: [sdb] Attached SCSI disk
Tue May  7 10:25:07 2024 daemon.err smbd[3686]: [2024/05/07 10:25:07.054814,  0] ../../source3/smbd/service.c:167(chdir_current_service)
Tue May  7 10:25:07 2024 daemon.err smbd[3686]:   chdir_current_service: vfs_ChDir(/mnt/sda1) failed: No such file or directory. Current token: uid=0, gid=0, 1 groups: 0
Tue May  7 10:25:07 2024 daemon.err smbd[3686]: [2024/05/07 10:25:07.056555,  0] ../../source3/smbd/service.c:167(chdir_current_service)
Tue May  7 10:25:07 2024 daemon.err smbd[3686]:   chdir_current_service: vfs_ChDir(/mnt/sda1) failed: No such file or directory. Current token: uid=0, gid=0, 1 groups: 0
Tue May  7 10:25:07 2024 daemon.err smbd[3686]: [2024/05/07 10:25:07.058042,  0] ../../source3/smbd/service.c:167(chdir_current_service)
Tue May  7 10:25:07 2024 daemon.err smbd[3686]:   chdir_current_service: vfs_ChDir(/mnt/sda1) failed: No such file or directory. Current token: uid=0, gid=0, 1 groups: 0

WYC-2020, 2024-05-07:
瞧瞧samba4又没更新呗 怀疑是这个问题 那就吧这个回滚了在测呗，不过我刚看这玩意两年没更新了，所以我怀疑可能还是内核升级留下的坑，你看看你那个老版本和你有问题的 他们的区别在哪里 重点检测samba4 usb相关 以及内核版本

dgyangxf2023, 2024-05-13:
内核回退了也不行

aiamadeus, 2024-05-13:
这个是你硬盘快挂了吧

coolsnowwolf, 2026-09-05:
核对当前代码与 OpenWrt 主线后，未发现 H68k 专用 USB/XHCI 或存储修复可移植到该报告。日志明确为 USB disconnect，随后 sda I/O error/device offline，Samba 只是触发高负载读写的用户；换回旧固件能工作不足以定位为 Samba。该 issue 没有提供 USB 控制器、硬盘盒芯片、UAS/quirks 或当前版本复现信息，无法安全添加内核补丁。请按最新固件附完整 dmesg、lsusb -t、smartctl 后重新提交；本旧报告先关闭。

closed 2026-09-05 by coolsnowwolf