coolsnowwolf/lede #12139: GCC 13编译问题 zmq库断言不过

GreatMichaelLee, 2024-05-07:
### 详细叙述

编译zmq库的时候，gcc 13 的assertion不过

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

time: package/feeds/packages/protobuf-c/host-compile#0.05#0.08#0.11
make[3]: Entering directory '/home/builder/lede_x86/feeds/packages/libs/zmq'
rm -f /home/builder/lede_x86/build_dir/target-x86_64_musl/zeromq-nc/zeromq-4.3.4/.built
touch /home/builder/lede_x86/build_dir/target-x86_64_musl/zeromq-nc/zeromq-4.3.4/.built_check
MAKEFLAGS="" /home/builder/lede_x86/staging_dir/host/bin/ninja -v -j1 -C /home/builder/lede_x86/build_dir/target-x86_64_musl/zeromq-nc/zeromq-4.3.4/openwrt-build
ninja: Entering directory `/home/builder/lede_x86/build_dir/target-x86_64_musl/zeromq-nc/zeromq-4.3.4/openwrt-build'
[1/104] /home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-13.2.0_musl/bin/x86_64-openwrt-linux-musl-g++ -DZMQ_CUSTOM_PLATFORM_HPP -D_GNU_SOURCE -D_REENTRANT -D_THREAD_SAFE -I/home/builder/lede_x86/build_dir/target-x86_64_musl/zeromq-nc/zeromq-4.3.4/include -I/home/builder/lede_x86/build_dir/target-x86_64_musl/zeromq-nc/zeromq-4.3.4/openwrt-build -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/builder/lede_x86/build_dir/target-x86_64_musl/zeromq-nc/zeromq-4.3.4=zeromq-4.3.4 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -std=gnu++11 -Wno-tautological-constant-compare -DNDEBUG -fPIC -MD -MT CMakeFiles/objects.dir/src/curve_client.cpp.o -MF CMakeFiles/objects.dir/src/curve_client.cpp.o.d -o CMakeFiles/objects.dir/src/curve_client.cpp.o -c /home/builder/lede_x86/build_dir/target-x86_64_musl/zeromq-nc/zeromq-4.3.4/src/curve_client.cpp
FAILED: CMakeFiles/objects.dir/src/curve_client.cpp.o
/home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-13.2.0_musl/bin/x86_64-openwrt-linux-musl-g++ -DZMQ_CUSTOM_PLATFORM_HPP -D_GNU_SOURCE -D_REENTRANT -D_THREAD_SAFE -I/home/builder/lede_x86/build_dir/target-x86_64_musl/zeromq-nc/zeromq-4.3.4/include -I/home/builder/lede_x86/build_dir/target-x86_64_musl/zeromq-nc/zeromq-4.3.4/openwrt-build -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/builder/lede_x86/build_dir/target-x86_64_musl/zeromq-nc/zeromq-4.3.4=zeromq-4.3.4 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -std=gnu++11 -Wno-tautological-constant-compare -DNDEBUG -fPIC -MD -MT CMakeFiles/objects.dir/src/curve_client.cpp.o -MF CMakeFiles/objects.dir/src/curve_client.cpp.o.d -o CMakeFiles/objects.dir/src/curve_client.cpp.o -c /home/builder/lede_x86/build_dir/target-x86_64_musl/zeromq-nc/zeromq-4.3.4/src/curve_client.cpp
In file included from /home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-13.2.0_musl/x86_64-openwrt-linux-musl/include/c++/13.2.0/ext/alloc_traits.h:34,
                 from /home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-13.2.0_musl/x86_64-openwrt-linux-musl/include/c++/13.2.0/bits/stl_tree.h:67,
                 from /home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-13.2.0_musl/x86_64-openwrt-linux-musl/include/c++/13.2.0/map:62,
                 from /home/builder/lede_x86/build_dir/target-x86_64_musl/zeromq-nc/zeromq-4.3.4/src/metadata.hpp:33,
                 from /home/builder/lede_x86/build_dir/target-x86_64_musl/zeromq-nc/zeromq-4.3.4/src/msg.hpp:40,
                 from /home/builder/lede_x86/build_dir/target-x86_64_musl/zeromq-nc/zeromq-4.3.4/src/curve_client.cpp:35:
/home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-13.2.0_musl/x86_64-openwrt-linux-musl/include/c++/13.2.0/bits/alloc_traits.h: In instantiation of 'struct std::__allocator_traits_base::__rebind<zmq::secure_allocator_t<unsigned char>, unsigned char, void>':
/home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-13.2.0_musl/x86_64-openwrt-linux-musl/include/c++/13.2.0/bits/alloc_traits.h:94:11:   required by substitution of 'template<class _Alloc, class _Up> using std::__alloc_rebind = typename std::__allocator_traits_base::__rebind<_Alloc, _Up>::type [with _Alloc = zmq::secure_allocator_t<unsigned char>; _Up = unsigned char]'
/home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-13.2.0_musl/x86_64-openwrt-linux-musl/include/c++/13.2.0/bits/alloc_traits.h:228:8:   required by substitution of 'template<class _Alloc> template<class _Tp> using std::allocator_traits< <template-parameter-1-1> >::rebind_alloc = std::__alloc_rebind<_Alloc, _Tp> [with _Tp = unsigned char; _Alloc = zmq::secure_allocator_t<unsigned char>]'
/home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-13.2.0_musl/x86_64-openwrt-linux-musl/include/c++/13.2.0/ext/alloc_traits.h:126:65:   required from 'struct __gnu_cxx::__alloc_traits<zmq::secure_allocator_t<unsigned char>, unsigned char>::rebind<unsigned char>'
/home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-13.2.0_musl/x86_64-openwrt-linux-musl/include/c++/13.2.0/bits/stl_vector.h:88:21:   required from 'struct std::_Vector_base<unsigned char, zmq::secure_allocator_t<unsigned char> >'
/home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-13.2.0_musl/x86_64-openwrt-linux-musl/include/c++/13.2.0/bits/stl_vector.h:425:11:   required from 'class std::vector<unsigned char, zmq::secure_allocator_t<unsigned char> >'
/home/builder/lede_x86/build_dir/target-x86_64_musl/zeromq-nc/zeromq-4.3.4/src/curve_client_tools.hpp:64:76:   required from here
/home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-13.2.0_musl/x86_64-openwrt-linux-musl/include/c++/13.2.0/bits/alloc_traits.h:70:31: error: static assertion failed: allocator_traits<A>::rebind_alloc<A::value_type> must be A
   70 |                         _Tp>::value,
      |                               ^~~~~
/home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-13.2.0_musl/x86_64-openwrt-linux-musl/include/c++/13.2.0/bits/alloc_traits.h:70:31: note: 'std::integral_constant<bool, false>::value' evaluates to false
cc1plus: note: unrecognized command-line option '-Wno-tautological-constant-compare' may have been intended to silence earlier diagnostics
ninja: build stopped: subcommand failed.
make[3]: *** [Makefile:128: /home/builder/lede_x86/build_dir/target-x86_64_musl/zeromq-nc/zeromq-4.3.4/.built] Error 1
make[3]: Leaving directory '/home/builder/lede_x86/feeds/packages/libs/zmq'
time: package/feeds/packages/zmq/nc/compile#0.23#0.13#0.35
    ERROR: package/feeds/packages/zmq failed to build (build variant: nc).
make[2]: *** [package/Makefile:116: package/feeds/packages/zmq/compile] Error 1
make[2]: Leaving directory '/home/builder/lede_x86'
make[1]: *** [package/Makefile:110: /home/builder/lede_x86/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/builder/lede_x86'
make: *** [/home/builder/lede_x86/include/toplevel.mk:231：world] 错误 2

WYC-2020, 2024-05-07:
[030-gcc13-compatibility.patch](https://github.com/coolsnowwolf/lede/files/15230902/030-gcc13-compatibility.patch)

GreatMichaelLee, 2024-05-07:
> [030-gcc13-compatibility.patch](https://github.com/coolsnowwolf/lede/files/15230902/030-gcc13-compatibility.patch)

哎，大佬们能不能把这些gcc 13的compatibility issue都集中搞一搞呢。。。。

WYC-2020, 2024-05-07:
> > [030-gcc13-compatibility.patch](https://github.com/coolsnowwolf/lede/files/15230902/030-gcc13-compatibility.patch)
> 
> 哎，大佬们能不能把这些gcc 13的compatibility issue都集中搞一搞呢。。。。

没那个时间 也没那个精力，我只管我自己编译需要那些库就行了，其他用不上管他死活 遇到再说

GreatMichaelLee, 2024-05-07:
目前我选的package环境下就node, nginx和zeromq lib这三个有兼容性问题，都解决了可以build出image了。大佬们有时间就整上去吧

WYC-2020, 2024-05-07:
> 目前我选的package环境下就node, nginx和zeromq lib这三个有兼容性问题，目前按@WYC-2020 大佬给的patch，都解决了可以build出image了。大佬们有时间就整上去吧^^

你去提撒 我之所以不提 是因为我仓库相对于l的仓库变化的太多 懒得去单独提某个 你的应该和l的是一样的吧

fynhnt, 2024-05-11:
> [030-gcc13-compatibility.patch](https://github.com/coolsnowwolf/lede/files/15230902/030-gcc13-compatibility.patch)

大佬这个怎么用？

closed 2024-07-19 by GreatMichaelLee