coolsnowwolf/lede #12140: xiaobao-nas的uboot patch第811行+CONFIG_TEXT_BASE=0x00200000有错误

Lemon1151, 2024-05-07:
### 详细叙述

xiaobao-nas的uboot patch第811行+CONFIG_TEXT_BASE=0x00200000有错误，应该是+CONFIG_SYS_TEXT_BASE=0x00200000才对

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

xiaobao-nas

### 详细日志

CONFIG_SYS_TEXT_BASE []

WYC-2020, 2024-05-07:
你搜索会发现 其他很多机型也是 CONFIG_TEXT_BASE 同时也有的是CONFIG_SYS_TEXT_BASE，并且uboot-rockchip下面的patches下面的path 全都是CONFIG_TEXT_BASE，如果这个错了 那其他很多都是错的吧，坐等大佬确认

Lemon1151, 2024-05-07:
我编译每次到这个位置就不动了CONFIG_SYS_TEXT_BASE []

Lemon1151, 2024-05-07:
> 你搜索会发现 其他很多机型也是 CONFIG_TEXT_BASE 同时也有的是CONFIG_SYS_TEXT_BASE，并且uboot-rockchip下面的patches下面的path 全都是CONFIG_TEXT_BASE，如果这个错了 那其他很多都是错的吧，坐等大佬确认

我编译每次到这个位置就不动了CONFIG_SYS_TEXT_BASE []

WYC-2020, 2024-05-07:
> > 你搜索会发现 其他很多机型也是 CONFIG_TEXT_BASE 同时也有的是CONFIG_SYS_TEXT_BASE，并且uboot-rockchip下面的patches下面的path 全都是CONFIG_TEXT_BASE，如果这个错了 那其他很多都是错的吧，坐等大佬确认
> 
> 我编译每次到这个位置就不动了CONFIG_SYS_TEXT_BASE []

等大佬来看了哈哈

aiamadeus, 2024-05-07:
https://github.com/u-boot/u-boot/commit/984639039f4cfe32ec2cc531d6ace05326ac49eb uboot v23.01 就已经重命名了，你代码是 lede 最新的吗

Lemon1151, 2024-05-07:
> [u-boot/u-boot@9846390](https://github.com/u-boot/u-boot/commit/984639039f4cfe32ec2cc531d6ace05326ac49eb) uboot v23.01 就已经重命名了，你代码是 lede 最新的吗

我五一那天拉取的源码呀

closed 2024-05-28 by aiamadeus