coolsnowwolf/lede #12143: tgt不能编绎

Nichunhua, 2024-05-08:
### 详细叙述

版本：1531aa758eed60b556d3e0b2731f7b743c43c51a
tgt版本：1.0.83
默认配置中只选tgt一个，编绎失败

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

X86_X64

### 详细日志

util.c: In function 'backed_file_open':
util.c:88:23: error: storage size of 'st' isn't known
   88 |         struct stat64 st;
      |                       ^~
util.c:96:15: error: implicit declaration of function 'fstat64'; did you mean 'fstatat'? [-Werror=implicit-function-declaration]
   96 |         err = fstat64(fd, &st);
      |               ^~~~~~~
      |               fstatat
util.c:88:23: error: unused variable 'st' [-Werror=unused-variable]
   88 |         struct stat64 st;
      |                       ^~
cc1: all warnings being treated as errors

WYC-2020, 2024-05-09:
这种错误 不是非常多的嘛 也说过方法 自己都不搜索一下，真是懒 https://github.com/coolsnowwolf/lede/issues/11771
[image]
如此

Nichunhua, 2024-05-09:
感谢 [WYC-2020](https://github.com/WYC-2020) 
目前我已在tgt的Makefile中添加了如下代码
[image]

ifeq ($(CONFIG_USE_MUSL),y)
	TARGET_CFLAGS += -D_LARGEFILE64_SOURCE
endif
EXTRA_CFLAGS=$(TARGET_CPPFLAGS)

重新编绎成功

另外由于我的NAS用的是LSI Raid 磁盘阵列卡，目前的包中没有相关的驱动，设备启动后无法识别到磁盘，所以我在block.mk添加了如下代码
[image]

define KernelPackage/megaraid
  SUBMENU:=$(BLOCK_MENU)
  TITLE:=megaraid device support 
  DEPENDS:=@TARGET_x86
  KCONFIG:=CONFIG_MEGARAID_MM \
  CONFIG_MEGARAID_MAILBOX \
  CONFIG_MEGARAID_SAS
  FILES:= $(LINUX_DIR)/drivers/scsi/megaraid/megaraid_sas.ko
  AUTOLOAD:=$(call AutoLoad,40,megaraid_sas,1)
endef

$(eval $(call KernelPackage,megaraid))

经编绎测试后能正常加载成功，不知这样是不是有问题，建议在项目中添加megaraid 的驱动支持

coolsnowwolf, 2026-09-05:
当前 feeds/packages 的 tgt 已为 1.0.97，且 Makefile 补丁与 OpenWrt 主线完全一致（包括移除 tgtimg/文档安装并正确处理 AIO/Sheepdog）。原报告针对 1.0.83 的 stat64/fstat64 编译错误已由版本升级解决，无需新增修改。issue 已过时，现关闭。

closed 2026-09-05 by coolsnowwolf