coolsnowwolf/lede #12144: mwan3在x86固件上不能正常使用了

shu0802, 2024-05-08:
### 详细叙述

负载均衡不能成功使用了

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

3865U软路由

### 详细日志

[image]

392377870, 2024-05-09:
[image]
[image]
我的x86也不能用了

392377870, 2024-05-09:
在启动项中重新启动后正常使用
[image]

shu0802, 2024-05-10:
在启动项中重新启动后还是不能正常使用

King-Of-Knights, 2024-05-13:
这种情况是网口与检测ip没连上

ramondelee, 2024-05-16:
这个问题在21.02里也有，以下给个推测和两个解决方法
推测：应该是mwan3先于wan启动，导致没有识别到，2.8.8版本应该有检测或刷新代码所以正常，2.11应该少了这个
解决方案一：系统-启动项-本地脚本，在exit 0前插入一行代码 mwan3 restart ，重启OK
解决方案一：编译时，修改源码files/etc/init.d/mwan3中START=19为START=96，生成固件重新刷机
我自己测试有效

zuni11, 2024-05-16:
大雕已经回退2.8.8啦，整个世界安静了。

wlshdjj, 2024-05-23:
> 这个问题在21.02里也有，以下给个推测和两个解决方法 推测：应该是mwan3先于wan启动，导致没有识别到，2.8.8版本应该有检测或刷新代码所以正常，2.11应该少了这个 解决方案一：系统-启动项-本地脚本，在exit 0前插入一行代码 mwan3 restart ，重启OK 解决方案二：编译时，修改源码files/etc/init.d/mwan3中START=19为START=96，生成固件重新刷机 我自己测试有效

mwan3不完善，一些东西执行fw3 restart之后，规则全部丧失，所以需要自己维护，加个firewall include。我也是用21.03的2.11.7版本，START改为99，自己加了include，用了近两年了。

ramondelee, 2024-05-27:
> mwan3不完善，一些东西执行fw3 restart之后，规则全部丧失，所以需要自己维护，加个firewall include。我也是用21.03的2.11.7版本，START改为99，自己加了include，用了近两年了。

请教一下“加个firewall include”这个是怎么操作的？

maxjove, 2024-06-14:
这个如何解决的， restart好像没用

coolsnowwolf, 2026-09-05:
已核对。该 issue 针对 mwan3 2.11 的启动顺序/接口未就绪问题，评论中已给出临时调整 START/restart 的办法。当前 packages feed 已同步上游 mwan3 2.12.2，初始化与 hotplug 逻辑也已更新；与 `/home/lean/Desktop/openwrt/feeds/packages/net/mwan3` 对比，核心 init 脚本一致。旧版 2.11 问题不再对应当前包，关闭为已完成。

closed 2026-09-05 by coolsnowwolf