coolsnowwolf/lede #12148: 切换为luci-23.05发现两个luci-app异常

eearphon, 2024-05-09:
### 详细叙述

http://192.168.1.1/cgi-bin/luci/admin/services/hd_idle
TypeError
Cannot read properties of null (reading 'trim')

http://192.168.1.1/cgi-bin/luci/admin/services/frps
RPCError
RPC call to uci/get failed with ubus code 4: 未找到资源
  at ClassConstructor.handleCallReply (http://192.168.1.1/luci-static/resources/rpc.js?v=git-24.129.18378-2e7d866:15:3)

用老版本luci则正常

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

ESxi 8.0 x86_64

### 详细日志

编译正常

kxdn3, 2024-05-09:
怎么切的

eearphon, 2024-05-10:
> 怎么切的
编辑 feeds.conf.default 文件，第二行行首加#，第三行行首#去掉

Wilson-coder, 2024-05-13:
Luci-App-Zerotier干脆就没了。本来想试着编译一下23.05尝尝鲜，没有这个App的话就只能换回旧版Luci了

loydcheung, 2024-05-15:
还有v2ray-server也没了，还好不是经常用

TheLastManT1, 2024-05-15:
同样的问题，frpc和frps都报错

TheLastManT1, 2024-05-16:
> Luci-App-Zerotier干脆就没了。本来想试着编译一下23.05尝尝鲜，没有这个App的话就只能换回旧版Luci了

我从kiddin9的仓库里拉的可以用

yyshell, 2024-05-17:
目前，有好多组件还没兼容，可以去隔壁imm看看。

OldCoding, 2024-05-18:
一些没有的APP可以从隔壁imm拷过来

TheLastManT1, 2024-05-18:
> 目前，有好多组件还没兼容，可以去隔壁imm看看。

> 一些没有的APP可以从隔壁imm拷过来

没用，一样的报错

TheLastManT1, 2024-05-19:
发现好像是git-24这种版本的就会异常，包括luci-app-netdata不显示，换成git-25的就好了

zgc, 2024-06-11:
刚刚尝试了一下，确实是有问题

coolsnowwolf, 2026-09-05:
已核对当前 LuCI 实现。`luci-app-hd-idle` 的磁盘信息读取已对空值做安全处理（`disk[opt]?.trim()`），不会再因 lsblk 返回 null 而触发 `Cannot read properties of null (reading trim)`。`luci-app-frps` 当前仍存在，并明确依赖 `+frps`，同时提供 `/etc/config/frps` 对应的 UCI 表单；因此旧版 LuCI git-24 的异常已被后续实现修复。该 issue 关闭为已完成。

closed 2026-09-05 by coolsnowwolf