coolsnowwolf/lede #12149: R5S无法识别NVMe固态硬盘，友善官方的可以识别

iHub-2020, 2024-05-09:
### 详细叙述

固件编译没有任何报错，系统也可以正常启动，加装了新买的闪迪 PCIe3.0 1TB的NVMe固态硬盘，重启后发现挂载点中没有nvme固态硬盘，使用fdisk -l也看不到，使用make kernel_menuconfig查看了配置，NVMe已经选择：
[image]
将下面的三个子选项全部选择再次编译后仍然无法识别；
于是采用友善官方的固件发现可以正常识别到这块SSD，因此判断SSD应该没有问题，应该是编译过程中缺少了相应组件或者驱动，还请知道的大神帮忙解答，谢谢！

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

友善R5S

### 详细日志

make[2]: Leaving directory '/home/openwrt/lede'
export MAKEFLAGS= ;make -w -r json_overview_image_info
make[2]: Entering directory '/home/openwrt/lede'
WORK_DIR=/home/openwrt/lede/build_dir/target-aarch64_generic_musl/json_info_files /home/openwrt/lede/scripts/json_overview_image_info.py /home/openwrt/lede/bin/targets/rockchip/armv8/profiles.json 
make[2]: Leaving directory '/home/openwrt/lede'
export MAKEFLAGS= ;make -w -r checksum
make[2]: Entering directory '/home/openwrt/lede'
make[2]: Leaving directory '/home/openwrt/lede'
make[1]: Leaving directory '/home/openwrt/lede'
openwrt@reyanmatic:~/lede$ make menuconfig

eearphon, 2024-05-10:
好些NVME兼容性不好的

WYC-2020, 2024-05-10:
kmod-nvme 这个驱动有么

iHub-2020, 2024-05-10:
@WYC-2020 请问如何检查kmod-nvme是否已经安装了呢？

WYC-2020, 2024-05-10:
在openwrt 里面 软件安装哪里搜索是否安装了 然后看下内核日志 nvme 有什么提示没 模块加载成了没

> @WYC-2020 查过了，是有的见截图 <img alt="5-10-2024 10-25-02 AM" width="380" src="https://private-user-images.githubusercontent.com/68175751/329437360-c4030af9-6b7d-4c79-b488-74f2b22eb6a5.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTUzMDk0MTMsIm5iZiI6MTcxNTMwOTExMywicGF0aCI6Ii82ODE3NTc1MS8zMjk0MzczNjAtYzQwMzBhZjktNmI3ZC00Yzc5LWI0ODgtNzRmMmIyMmViNmE1LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MTAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTEwVDAyNDUxM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTEyMTk1YzczNTE4MTFmNTViODhlOWZkMDNjMjg1MWIwNTkwNWZhYzZiZDViMTI5NWU1ZDY0ZDY5NzRiMzA3ZmUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.XOoZONUgZneNQyb9N-8TehRVL82yJZuHfHr8rXgfxTA">

WYC-2020, 2024-05-10:
https://github.com/coolsnowwolf/lede/issues/2728  看这个帖子l说好像要efi模式的才支持 你的是efi么

iHub-2020, 2024-05-10:
EFI是x86才有的选项，R5S里没有这个选项呀
[image]

WYC-2020, 2024-05-10:
> EFI是x86才有的选项，R5S里没有这个选项呀 <img alt="5-10-2024 11-00-26 AM" width="430" src="https://private-user-images.githubusercontent.com/68175751/329443196-ab400b31-4abe-4b72-bf60-edf830d2ac8e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTUzMTAzNzUsIm5iZiI6MTcxNTMxMDA3NSwicGF0aCI6Ii82ODE3NTc1MS8zMjk0NDMxOTYtYWI0MDBiMzEtNGFiZS00YjcyLWJmNjAtZWRmODMwZDJhYzhlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MTAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTEwVDAzMDExNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTM5Zjg4YWQwYmNkNjc3ZmNhMWU1ODQwYWMxYTU0ZDY3MWVlZjNhMzk5ODNiYzdkZjU3OGJmZDJhYmNlMmE3MzUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.pMK41C1ICRdv5l613RjT9pTM0rQTSbBXFG1f74NFUIM">
对哈 你是R5S哈哈忘记了，按我前面那个方法检查一下吧 ，实在没辙 你只能贴内核日志给大雕他们检查了

iHub-2020, 2024-05-10:
好的，谢谢啦

iHub-2020, 2024-05-10:
我就先把内核部分的日志贴出来吧，实在不行就只能放弃用友善官方固件了，不折腾了：

Fri May 10 17:11:34 2024 kern.info kernel: [    1.037104] pci 0002:20:00.0: PME# supported from D0 D1 D3hot
Fri May 10 17:11:34 2024 kern.warn kernel: [    1.042519] pci 0002:20:00.0: Primary bus is hard wired to 0
Fri May 10 17:11:34 2024 kern.info kernel: [    1.043088] pci 0002:20:00.0: bridge configuration invalid ([bus 01-ff]), reconfiguring
Fri May 10 17:11:34 2024 kern.info kernel: [    1.044124] pci_bus 0002:21: busn_res: [bus 21-2f] end is updated to 21
Fri May 10 17:11:34 2024 kern.info kernel: [    1.044808] pci 0002:20:00.0: BAR 0: assigned [mem 0x380000000-0x3bfffffff]
Fri May 10 17:11:34 2024 kern.info kernel: [    1.045527] pci 0002:20:00.0: BAR 1: no space for [mem size 0x40000000]
Fri May 10 17:11:34 2024 kern.info kernel: [    1.046183] pci 0002:20:00.0: BAR 1: failed to assign [mem size 0x40000000]
Fri May 10 17:11:34 2024 kern.info kernel: [    1.046870] pci 0002:20:00.0: BAR 6: assigned [mem 0xf0200000-0xf020ffff pref]
Fri May 10 17:11:34 2024 kern.info kernel: [    1.047587] pci 0002:20:00.0: PCI bridge to [bus 21]
Fri May 10 17:11:34 2024 kern.info kernel: [    1.051485] pcieport 0002:20:00.0: PME: Signaling with IRQ 59
Fri May 10 17:11:34 2024 kern.info kernel: [    1.052512] pcieport 0002:20:00.0: AER: enabled with IRQ 60
Fri May 10 17:11:34 2024 kern.info kernel: [    1.059517] dma-pl330 fe530000.dma-controller: Loaded driver for PL330 DMAC-241330
Fri May 10 17:11:34 2024 kern.info kernel: [    1.060280] dma-pl330 fe530000.dma-controller:    DBUFF-128x8bytes Num_Chans-8 Num_Peri-32 Num_Events-16
Fri May 10 17:11:34 2024 kern.info kernel: [    1.063926] dma-pl330 fe550000.dma-controller: Loaded driver for PL330 DMAC-241330
Fri May 10 17:11:34 2024 kern.info kernel: [    1.064681] dma-pl330 fe550000.dma-controller:    DBUFF-128x8bytes Num_Chans-8 Num_Peri-32 Num_Events-16
Fri May 10 17:11:34 2024 kern.err kernel: [    1.066928] arm-scmi firmware:scmi: Failed. SCMI protocol 22 not active.
Fri May 10 17:11:34 2024 kern.info kernel: [    1.069755] Serial: 8250/16550 driver, 4 ports, IRQ sharing enabled
Fri May 10 17:11:34 2024 kern.info kernel: [    1.072269] printk: console [ttyS2] disabled
Fri May 10 17:11:34 2024 kern.info kernel: [    1.072766] fe660000.serial: ttyS2 at MMIO 0xfe660000 (irq = 65, base_baud = 1500000) is a 16550A
Fri May 10 17:11:34 2024 kern.info kernel: [    1.073960] printk: console [ttyS2] enabled
Fri May 10 17:11:34 2024 kern.info kernel: [    1.074750] printk: bootconsole [uart8250] disabled
Fri May 10 17:11:34 2024 kern.notice kernel: [    1.078136] random: crng init done
Fri May 10 17:11:34 2024 kern.warn kernel: [    1.079973] cacheinfo: Unable to detect cache hierarchy for CPU 0
Fri May 10 17:11:34 2024 kern.info kernel: [    1.085023] loop: module loaded
Fri May 10 17:11:34 2024 kern.info kernel: [    1.085429] mtip32xx Version 1.3.1
Fri May 10 17:11:34 2024 kern.info kernel: [    1.091310] rk_gmac-dwmac fe2a0000.ethernet: IRQ eth_lpi not found
Fri May 10 17:11:34 2024 kern.warn kernel: [    1.092283] rk_gmac-dwmac fe2a0000.ethernet: supply phy not found, using dummy regulator
Fri May 10 17:11:34 2024 kern.info kernel: [    1.093155] rk_gmac-dwmac fe2a0000.ethernet: clock input or output? (output).
Fri May 10 17:11:34 2024 kern.info kernel: [    1.093838] rk_gmac-dwmac fe2a0000.ethernet: TX delay(0x3c).
Fri May 10 17:11:34 2024 kern.info kernel: [    1.094348] rk_gmac-dwmac fe2a0000.ethernet: RX delay(0x2f).
Fri May 10 17:11:34 2024 kern.info kernel: [    1.094862] rk_gmac-dwmac fe2a0000.ethernet: integrated PHY? (no).
Fri May 10 17:11:34 2024 kern.info kernel: [    1.100474] rk_gmac-dwmac fe2a0000.ethernet: init for RGMII
Fri May 10 17:11:34 2024 kern.info kernel: [    1.101243] rk_gmac-dwmac fe2a0000.ethernet: User ID: 0x30, Synopsys ID: 0x51
Fri May 10 17:11:34 2024 kern.info kernel: [    1.101915] rk_gmac-dwmac fe2a0000.ethernet:      DWMAC4/5
Fri May 10 17:11:34 2024 kern.info kernel: [    1.102380] rk_gmac-dwmac fe2a0000.ethernet: DMA HW capability register supported
Fri May 10 17:11:34 2024 kern.info kernel: [    1.103041] rk_gmac-dwmac fe2a0000.ethernet: RX Checksum Offload Engine supported
Fri May 10 17:11:34 2024 kern.info kernel: [    1.103702] rk_gmac-dwmac fe2a0000.ethernet: TX Checksum insertion supported
Fri May 10 17:11:34 2024 kern.info kernel: [    1.104323] rk_gmac-dwmac fe2a0000.ethernet: Wake-Up On Lan supported
Fri May 10 17:11:34 2024 kern.info kernel: [    1.104889] rk_gmac-dwmac fe2a0000.ethernet: TSO supported
Fri May 10 17:11:34 2024 kern.info kernel: [    1.105392] rk_gmac-dwmac fe2a0000.ethernet: Enable RX Mitigation via HW Watchdog Timer
Fri May 10 17:11:34 2024 kern.info kernel: [    1.106111] rk_gmac-dwmac fe2a0000.ethernet: Enabled RFS Flow TC (entries=10)
Fri May 10 17:11:34 2024 kern.info kernel: [    1.106747] rk_gmac-dwmac fe2a0000.ethernet: TSO feature enabled
Fri May 10 17:11:34 2024 kern.info kernel: [    1.107282] rk_gmac-dwmac fe2a0000.ethernet: Using 32/32 bits DMA host/device width
Fri May 10 17:11:34 2024 kern.info kernel: [    1.203665] xhci-hcd xhci-hcd.0.auto: xHCI Host Controller
Fri May 10 17:11:34 2024 kern.info kernel: [    1.204195] xhci-hcd xhci-hcd.0.auto: new USB bus registered, assigned bus number 1
Fri May 10 17:11:34 2024 kern.info kernel: [    1.204833] ehci-platform fd800000.usb: EHCI Host Controller
Fri May 10 17:11:34 2024 kern.info kernel: [    1.204849] ohci-platform fd840000.usb: Generic Platform OHCI controller
[... cut ...]
Fri May 10 17:11:34 2024 kern.info kernel: [   15.616472] PPP MPPE Compression module registered
Fri May 10 17:11:34 2024 kern.info kernel: [   15.617963] NET: Registered PF_PPPOX protocol family
Fri May 10 17:11:34 2024 user.info kernel: [   15.620016] kmodloader: done loading kernel modules from /etc/modules.d/*
Fri May 10 17:11:34 2024 kern.info kernel: [   17.305577] EXT4-fs (mmcblk0p1): mounted filesystem without journal. Quota mode: none.
Fri May 10 17:11:34 2024 kern.info kernel: [   17.376440] EXT4-fs (mmcblk1p1): mounted filesystem with ordered data mode. Quota mode: none.

iHub-2020, 2024-05-12:
看来无解，估计R6S一样，已经申请退货了~

VIKINGYFY, 2024-05-15:
换天灵的IMM试试

yyshell, 2024-05-17:
由于友善官方不友善，好像lede对支持友善的设备也不友善了。

commit cb1f2c6a (2024-06-05): rockchip: fix rk3568 nvme detection issue

Fixes: #12149
closed 2024-06-05 by aiamadeus