coolsnowwolf/lede #12150: 这个报错怎么处理

mmeiwenlong, 2024-05-09:
### 详细叙述

Turn off this advice by setting config variable advice.detachedHead to false

HEAD is now at a2e2717e Ensure read_file() files are considered by "gn analyze"
Packing checkout...
. /home/meiyong/lede/include/shell.sh; xzcat /home/meiyong/lede/dl/gn-2024-01-24-a2e2717e.tar.xz | tar -C /home/meiyong/lede/build_dir/hostpkg/gn-2024-01-24-a2e2717e/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/* /home/meiyong/lede/build_dir/hostpkg/gn-2024-01-24-a2e2717e
touch /home/meiyong/lede/build_dir/hostpkg/gn-2024-01-24-a2e2717e/.prepared28ee227364b6b916065e8c7813042717_6664517399ebbbc92a37c5bb081b5c53
python3 /home/meiyong/lede/build_dir/hostpkg/gn-2024-01-24-a2e2717e/build/gen.py --no-last-commit-position
touch /home/meiyong/lede/build_dir/hostpkg/gn-2024-01-24-a2e2717e/.configured
MAKEFLAGS="" /home/meiyong/lede/staging_dir/host/bin/ninja  -j1 -C /home/meiyong/lede/build_dir/hostpkg/gn-2024-01-24-a2e2717e/out
ninja: Entering directory `/home/meiyong/lede/build_dir/hostpkg/gn-2024-01-24-a2e2717e/out'
[1/287] CXX src/base/command_line.o
FAILED: src/base/command_line.o 
clang++ -MMD -MF src/base/command_line.o.d -I../src -I. -DNDEBUG -O3 -fdata-sections -ffunction-sections -Werror -D_FILE_OFFSET_BITS=64 -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -pthread -pipe -fno-exceptions -fno-rtti -fdiagnostics-color -Wall -Wextra -Wno-unused-parameter -Wextra-semi -Wundef -std=c++20 -Wrange-loop-analysis -Wextra-semi-stmt -c ../src/base/command_line.cc -o src/base/command_line.o
/bin/sh: 1: clang++: not found
ninja: build stopped: subcommand failed.
make[3]: *** [Makefile:58: /home/meiyong/lede/build_dir/hostpkg/gn-2024-01-24-a2e2717e/.built] Error 1
make[3]: Leaving directory '/home/meiyong/lede/feeds/helloworld/gn'
time: package/feeds/helloworld/gn/host-compile#4.01#0.75#8.10
    ERROR: package/feeds/helloworld/gn [host] failed to build.
make[2]: *** [package/Makefile:114: package/feeds/helloworld/gn/host/compile] Error 1
make[2]: Leaving directory '/home/meiyong/lede'
make[1]: *** [package/Makefile:110: /home/meiyong/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/meiyong/lede'
make: *** [/home/meiyong/lede/include/toplevel.mk:231：world] 错误 2
meiyong@meiyong-virtual-machine:~/lede$ 

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

i5 1135G7

### 详细日志

Turn off this advice by setting config variable advice.detachedHead to false

HEAD is now at a2e2717e Ensure read_file() files are considered by "gn analyze"
Packing checkout...
. /home/meiyong/lede/include/shell.sh; xzcat /home/meiyong/lede/dl/gn-2024-01-24-a2e2717e.tar.xz | tar -C /home/meiyong/lede/build_dir/hostpkg/gn-2024-01-24-a2e2717e/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/* /home/meiyong/lede/build_dir/hostpkg/gn-2024-01-24-a2e2717e
touch /home/meiyong/lede/build_dir/hostpkg/gn-2024-01-24-a2e2717e/.prepared28ee227364b6b916065e8c7813042717_6664517399ebbbc92a37c5bb081b5c53
python3 /home/meiyong/lede/build_dir/hostpkg/gn-2024-01-24-a2e2717e/build/gen.py --no-last-commit-position
touch /home/meiyong/lede/build_dir/hostpkg/gn-2024-01-24-a2e2717e/.configured
MAKEFLAGS="" /home/meiyong/lede/staging_dir/host/bin/ninja  -j1 -C /home/meiyong/lede/build_dir/hostpkg/gn-2024-01-24-a2e2717e/out
ninja: Entering directory `/home/meiyong/lede/build_dir/hostpkg/gn-2024-01-24-a2e2717e/out'
[1/287] CXX src/base/command_line.o
FAILED: src/base/command_line.o 
clang++ -MMD -MF src/base/command_line.o.d -I../src -I. -DNDEBUG -O3 -fdata-sections -ffunction-sections -Werror -D_FILE_OFFSET_BITS=64 -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -pthread -pipe -fno-exceptions -fno-rtti -fdiagnostics-color -Wall -Wextra -Wno-unused-parameter -Wextra-semi -Wundef -std=c++20 -Wrange-loop-analysis -Wextra-semi-stmt -c ../src/base/command_line.cc -o src/base/command_line.o
/bin/sh: 1: clang++: not found
ninja: build stopped: subcommand failed.
make[3]: *** [Makefile:58: /home/meiyong/lede/build_dir/hostpkg/gn-2024-01-24-a2e2717e/.built] Error 1
make[3]: Leaving directory '/home/meiyong/lede/feeds/helloworld/gn'
time: package/feeds/helloworld/gn/host-compile#4.01#0.75#8.10
    ERROR: package/feeds/helloworld/gn [host] failed to build.
make[2]: *** [package/Makefile:114: package/feeds/helloworld/gn/host/compile] Error 1
make[2]: Leaving directory '/home/meiyong/lede'
make[1]: *** [package/Makefile:110: /home/meiyong/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/meiyong/lede'
make: *** [/home/meiyong/lede/include/toplevel.mk:231：world] 错误 2
meiyong@meiyong-virtual-machine:~/lede$

Umomboom, 2024-05-10:
sudo apt install clang

WYC-2020, 2024-05-10:
> sudo apt install clang

这根本就不是宿主机的问题，怕不是用了helloworld 里面的gn 吧,那个是有问题的，少了patches，自己用xiaorouji这边的

mmeiwenlong, 2024-05-10:
> sudo apt install clang

谢谢，可以了

closed 2024-05-10 by mmeiwenlong