coolsnowwolf/lede #12152: 切换到23.05自动锁firewall4了

TheLastManT1, 2024-05-10:
### 详细叙述

从去年开始就一直编译的，五一前还能正常编译，后来切换到23.05就自动锁firewall4了，无法取消。已经删除过tmp和.config文件了，没用。用grep查了下，用到firewall4的也就homeproxy，但是我也没勾啊。
我切换回18，firewall4就没有被强制选上了，再切回23，firewall4还是被强制选上的。
我重新拉源码，改成一样的23.05，也加了原来的源，firewall4却没有被强制选上。
到底是哪里的问题啊

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

无，只勾选firewall4一切正常。再勾选firewall就是两个插件冲突

coolsnowwolf, 2024-05-10:
rm -rf tmp ，清理缓存

WYC-2020, 2024-05-10:
刚亲测了一些没有强制锁

TheLastManT1, 2024-05-10:
> rm -rf tmp ，清理缓存

试过了，没有用

TheLastManT1, 2024-05-10:
> 刚亲测了一些没有强制锁

真的无解了，./scripts/feeds uninstall -a删除了全部源，然后再只装官方源·，还是会强制选上firewall4。真是见鬼了

TheLastManT1, 2024-05-10:
> > 刚亲测了一些没有强制锁
> 
> 真的无解了，./scripts/feeds uninstall -a删除了全部源，然后再只装官方源·，还是会强制选上firewall4。真是见鬼了

记错了，不是uninstall -a，是clean

gsblay, 2024-05-11:
Base system里面选firewall，不要选firewall4看看。
我的23.05的确没有锁firewall4

ZqinKing, 2024-05-11:
make menuconfig下使用/n查一下被什么组件依赖了。

TheLastManT1, 2024-05-12:
> make menuconfig下使用/n查一下被什么组件依赖了。

没用啊，这是什么命令？

TheLastManT1, 2024-05-12:
无法解决，已经重开了

closed 2024-05-12 by TheLastManT1