coolsnowwolf/lede #12154: 开启Turbo ACC使用WireGuard系统和内核报日志

YEYEv50, 2024-05-11:
### 详细叙述

使用最新源码编译，开启Turbo ACC 流量分载 Flow Offloading。使用手机网络连接WireGuard组网后出现。
WireGuard接口持续 Keep-Alive为25，其他默认无变动。
目前新编译在用的固件无集成SFE，关闭Turbo ACC Flow Offloading重试组网无类似日志出现。
或将流量分载选成SFE也无此问题（上一版编译的集成了Flow Offloading，SFE）

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

ACRH17

### 详细日志

系统日志
Sat May 11 20:12:57 2024 kern.warn kernel: [  229.062623] test 65534
Sat May 11 20:12:58 2024 kern.warn kernel: [  229.242229] test 65534
Sat May 11 20:12:58 2024 kern.warn kernel: [  229.439688] test 65534
Sat May 11 20:12:58 2024 kern.warn kernel: [  229.444854] test 65534
Sat May 11 20:12:58 2024 kern.warn kernel: [  229.454273] test 65534

内核日志
[  153.407355] test 65534
[  180.693205] test 65534
[  180.895273] test 65534
[  181.099912] test 65534
[  181.193946] test 65534
[  181.203014] test 65534
[  181.223141] test 65534

WYC-2020, 2024-05-13:
应该又是什么调试信息 有时间瞧瞧

YEYEv50, 2024-05-13:
> 应该又是什么调试信息 有时间瞧瞧

我也不知道咋回事，以前好像没出现过。还是我没注意到？以前用immortalwrt 23.05的貌似没发现有

WYC-2020, 2024-05-14:
[image]
吧这个补丁中的 这句打印去掉，应该就不会打印那个 test了，你试试

WYC-2020, 2024-05-14:
看了下应该只有内核5.10才会有这个 问题

WYC-2020, 2024-05-14:
@YEYEv50 试试 可以了给个答复

YEYEv50, 2024-05-14:
> 看了下应该只有内核5.10才会有这个 问题

好的，感谢。只是输出调试日志没什么影响就行。主要以前没注意到过，有些好奇。那关闭issues了。

closed 2024-05-14 by YEYEv50

WYC-2020, 2024-05-15:
> > 看了下应该只有内核5.10才会有这个 问题
> 
> 好的，感谢。只是输出调试日志没什么影响就行。主要以前没注意到过，有些好奇。那关闭issues了。

功能是否正常 这个你的验证了 我只是去掉那个 调试打印信息，如果你原来 打印这个的时候 他是正常的  那去掉这个打印 也正常 那就没问题了，如果原来 就不正常 那就需要具体分析一下原因了