coolsnowwolf/lede #12155: 为什么aria2运行不了呀

Lemon1151, 2024-05-12:
### 详细叙述

为什么aria2运行不了呀

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

xioabao-nas 24.5.1

### 详细日志

[image]

melodicule, 2024-05-12:
我猜你的系统日志里面有这样一行
```
daemon.err aria2: Please create download dir first.
```
查看你的aria2配置->文件和目录，确认指定的下载目录是否存在，目录的所有者是不是aria2

qingtian110, 2024-05-16:
因为 lede 的 feeds 太老旧，没人维护或同步。你的原因是 openssl 升级了 https://github.com/coolsnowwolf/lede/commit/7494eb16185a176de226f55e842cadf94f1c5a11

aria2 则需要  `procd_add_jail_mount "/usr/lib"`   才能运行在 openssl 3 上。

手动改下文件 `/etc/init.d/aria2`

添加这样的内容：https://github.com/sbwml/feeds_packages_net_aria2/commit/cfaa72235df27f764642667e3422a1975f054c01

就可以正常启动。

f1owkang, 2024-05-26:
> 因为 lede 的 feeds 太老旧，没人维护或同步。你的原因是 openssl 升级了 [7494eb1](https://github.com/coolsnowwolf/lede/commit/7494eb16185a176de226f55e842cadf94f1c5a11)
> 
> aria2 则需要 `procd_add_jail_mount "/usr/lib"` 才能运行在 openssl 3 上。
> 
> 手动改下文件 `/etc/init.d/aria2`
> 
> 添加这样的内容：[sbwml/feeds_packages_net_aria2@cfaa722](https://github.com/sbwml/feeds_packages_net_aria2/commit/cfaa72235df27f764642667e3422a1975f054c01)
> 
> 就可以正常启动。

楼上正解，已验证成功

poromony, 2024-06-02:
> 因为 lede 的 feeds 太老旧，没人维护或同步。你的原因是 openssl 升级了 https://github.com/coolsnowwolf/lede/commit/7494eb16185a176de226f55e842cadf94f1c5a11
>
>
>
> aria2 则需要  `procd_add_jail_mount "/usr/lib"`   才能运行在 openssl 3 [...]

我这里还是不行，我是启用了ssl，证书和密钥直接用的uhttpd 的文件，之前都是正常的，最近一次编译后一直报错，系统日志如下：
Sun Jun  2 16:48:43 2024 daemon.info aria2c[27967]: jail: exec-ing /usr/bin/aria2c
Sun Jun  2 16:48:43 2024 daemon.err aria2c[27967]: Parse error in /mnt/nvme0n1p1/software/aria2/run_files/aria2.conf.main
Sun Jun  2 16:48:43 2024 daemon.err aria2c[27967]: Exception: [AbstractOptionHandler.cc:69] errorCode=28 We encountered a problem while processing the option '--rpc-certificate'.
Sun Jun  2 16:48:43 2024 daemon.err aria2c[27967]:   -> [OptionHandlerImpl.cc:556] errorCode=1 Could not get file status: No such file or directory
Sun Jun  2 16:48:43 2024 daemon.err aria2c[27967]: Usage:
Sun Jun  2 16:48:43 2024 daemon.err aria2c[27967]:  --rpc-certificate=FILE       Use the certificate in FILE for RPC server.
Sun Jun  2 16:48:43 2024 daemon.err aria2c[27967]:                               The certificate must be in PEM format.
Sun Jun  2 16:48:43 2024 daemon.err aria2c[27967]:                               Use --rpc-private-key option to specify the
Sun Jun  2 16:48:43 2024 daemon.err aria2c[27967]:                               private key. Use --rpc-secure option to enable
Sun Jun  2 16:48:43 2024 daemon.err aria2c[27967]:                               encryption.
Sun Jun  2 16:48:43 2024 daemon.info aria2c[27967]: jail: jail (27968) exited with exit: 28
Sun Jun  2 16:48:43 2024 daemon.info procd: Instance aria2::aria2.main s in a crash loop 6 crashes, 0 seconds since last crash
Sun Jun  2 16:55:45 2024 daemon.info aria2: Instance "main" disabled.

coolsnowwolf, 2026-09-05:
已修复并推送 packages 独立提交 `c378b22b2db7053b51cfb10f793cc7980bc3fd4a`。根因是 aria2 在 procd jail 中无法访问 OpenSSL 3 的 provider 模块目录；已按 OpenWrt 主线同步，在 `/etc/init.d/aria2` 的 jail 配置中加入 `procd_add_jail_mount "/usr/lib/ossl-modules"`。已完成 shell 语法检查、diff 检查，并确认 origin/master 与本地提交一致。请升级 aria2 包或重新编译固件后验证，现关闭为已完成。

closed 2026-09-05 by coolsnowwolf