coolsnowwolf/lede #12157: 2个无线网卡，其中一个重启或再次开机后不是驱动丢失就是无法开启网络

dragon888888888888, 2024-05-13:
### 详细叙述

一个是板载的无线网卡，又加了一个WIFI6的无线网卡
执行重置后虽然驱动有了，但下面这个有时仍然无法打开，如果是重启或重新开机则是一定无法开启的

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

光影猫

### 详细日志

已完成编译
[image]

wlshdjj, 2024-05-14:
有些网卡不支持多interface，删除只保留一个试试

dragon888888888888, 2024-05-14:
> 有些网卡不支持多interface，删除只保留一个试试

和这个无关，一般就是只开了一个,设置后可以工作，但是不能重启或关机，否则下次重开机下面这个网卡就是无线未开启或未关联
有时甚至驱动丢失，需要执行重置，重置后驱动虽然正常了，但是还是一样不能重开面，否则再次开机还是一样的无线未开启或未关联

loydcheung, 2024-05-15:
是不是存储空间捉襟见肘，刚好保存不了。

dragon888888888888, 2024-05-15:
> 是不是存储空间捉襟见肘，刚好保存不了。

不是，虽然16G 只分了几百M，也不会保存不了。

dragon888888888888, 2024-05-15:
> 是不是存储空间捉襟见肘，刚好保存不了。

[image]

dragon888888888888, 2024-05-15:
[image]
只要不重启就可以正常工作，一重启radio1这个网卡就是无线未开启或未关联（甚至驱动丢失），上面的radio0那个正常

dragon888888888888, 2024-06-05:
已解决，radio1设置成56频道，重启后就可以正常工作，过过比上面那个晚1分钟左右启动

closed 2024-06-11 by dragon888888888888