coolsnowwolf/lede #12171: wifi 客户端 没网

bigDshe, 2024-05-19:
### 详细叙述

刷了openwrt 的小米R3G MT7621，通过wifi 扫描加入普通主路由的方式连接，可以正常获取到ip，但wifi 和网线连接都没网，连主路由都ping不通，换了5.4/5.10内核都一样，使用默认配置不另加插件编译出来也是同样问题，试了官方23.05的wifi客户端模式可以正常上网。就感觉wwan 和wan隔断了一样，虽然添加的时候防火墙选的wan。

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

MT7621

### 详细日志

##其他都没问题，只有wifi客户端方式连接主路由上网，没网

caixin1218, 2024-05-20:
> ### 详细叙述
> 刷了openwrt 的小米R3G MT7621，通过wifi 扫描加入普通主路由的方式连接，可以正常获取到ip，但wifi 和网线连接都没网，连主路由都ping不通，换了5.4/5.10内核都一样，使用默认配置不另加插件编译出来也是同样问题，试了官方23.05的wifi客户端模式可以正常上网。就感觉wwan 和wan隔断了一样，虽然添加的时候防火墙选的wan。
> 
> ### 重复 issue
> * [x]  没有类似的 issue
> 
> ### 具体型号
> MT7621
> 
> ### 详细日志
> ##其他都没问题，只有wifi客户端方式连接主路由上网，没网

可能是配置的问题，你可以先使用本地连接wifi 和有线 看有没有问题，再添加一个无线客户端，这样对比就知道哪里的问题。

bigDshe, 2024-05-20:
换openwrt官方系统 ，顺利解决

bigDshe, 2024-05-21:
测试了几个路由器，不同人编译的固件，发现wifi客户端模式连接主路由，没网络的问题比较普遍

bigDshe, 2024-05-25:
换官方了

closed 2024-05-25 by bigDshe