coolsnowwolf/lede #12172: 无线桥接时，当无线客户端的信道为 AUTO 时，同频段的 WIFI 无法启动

No06, 2024-05-20:
### 详细叙述

[image]

如图桥接的无线信道是52，当设置客户端信道AUTO时，客户端可以连接，但是 AP 启动不了

客户端设置信道52后就都正常

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

CR660x

### 详细日志

```
Mon May 20 11:47:51 2024 daemon.notice hostapd: Remove interface 'wlan1-1'
Mon May 20 11:47:51 2024 daemon.notice hostapd: wlan1-1: interface state ENABLED->DISABLED
Mon May 20 11:47:51 2024 daemon.notice hostapd: wlan1-1: AP-DISABLED
Mon May 20 11:47:51 2024 daemon.notice hostapd: wlan1-1: CTRL-EVENT-TERMINATING
Mon May 20 11:47:51 2024 daemon.notice hostapd: nl80211: deinit ifname=wlan1-1 disabled_11b_rates=0
Mon May 20 11:47:51 2024 kern.info kernel: [423789.374324] device wlan1-1 left promiscuous mode
Mon May 20 11:47:51 2024 kern.info kernel: [423789.379573] br-lan: port 2(wlan1-1) entered disabled state
Mon May 20 11:47:52 2024 daemon.notice netifd: Network device 'wlan1-1' link is down
Mon May 20 11:47:52 2024 kern.info kernel: [423789.682091] wlan1: deauthenticating from 70:42:d3:17:ae:7c by local choice (Reason: 3=DEAUTH_LEAVING)
Mon May 20 11:47:52 2024 daemon.notice wpa_supplicant[2956]: wlan1: CTRL-EVENT-DISCONNECTED bssid=70:42:d3:17:ae:7c reason=3 locally_generated=1
Mon May 20 11:47:52 2024 daemon.err wpa_supplicant[2956]:  Failed to stop hostapd AP interfaces
Mon May 20 11:47:52 2024 daemon.notice netifd: Network device 'wlan1' link is down
Mon May 20 11:47:52 2024 daemon.notice netifd: Interface 'wwan' has link connectivity loss
Mon May 20 11:47:52 2024 daemon.err wpa_supplicant[2956]:  Failed to stop hostapd AP interfaces
Mon May 20 11:47:52 2024 daemon.notice wpa_supplicant[2956]: wlan1: CTRL-EVENT-DSCP-POLICY clear_all
Mon May 20 11:47:52 2024 daemon.notice netifd: wwan (18681): udhcpc: received SIGTERM
Mon May 20 11:47:52 2024 daemon.notice netifd: wwan (18681): udhcpc: unicasting a release of 192.168.110.36 to 192.168.110.1
Mon May 20 11:47:52 2024 daemon.notice netifd: wwan (18681): udhcpc: sending release
Mon May 20 11:47:52 2024 daemon.notice netifd: wwan (18681): udhcpc: entering released state
Mon May 20 11:47:52 2024 daemon.notice wpa_supplicant[2956]: wlan1: CTRL-EVENT-DSCP-POLICY clear_all
Mon May 20 11:47:52 2024 daemon.notice wpa_supplicant[2956]: nl80211: deinit ifname=wlan1 disabled_11b_rates=0
Mon May 20 11:47:52 2024 daemon.notice netifd: wwan (18681): Command failed: ubus call network.interface notify_proto { "action": 0, "link-up": false, "keep": false, "interface": "wwan" } (Permission denied)
Mon May 20 11:47:52 2024 daemon.notice netifd: Interface 'wwan' is now down
Mon May 20 11:47:53 2024 daemon.notice netifd: Interface 'wwan' is disabled
Mon May 20 11:47:53 2024 daemon.notice netifd: Interface 'wwan' is enabled
Mon May 20 11:47:53 2024 daemon.notice netifd: Interface 'wwan' is disabled
Mon May 20 11:47:54 2024 daemon.notice hostapd: Configuration file: /var/run/hostapd-phy1.conf (phy wlan1-1) --> new PHY
Mon May 20 11:47:54 2024 kern.info kernel: [423792.146979] br-lan: port 2(wlan1-1) entered blocking state
Mon May 20 11:47:54 2024 kern.info kernel: [423792.152592] br-lan: port 2(wlan1-1) entered disabled state
Mon May 20 11:47:54 2024 kern.info kernel: [423792.159089] device wlan1-1 entered promiscuous mode
Mon May 20 11:47:54 2024 daemon.notice hostapd: wlan1-1: interface state UNINITIALIZED->COUNTRY_UPDATE
Mon May 20 11:47:54 2024 daemon.notice hostapd: ACS: Automatic channel selection started, this may take a bit
Mon May 20 11:47:54 2024 daemon.notice hostapd: wlan1-1: interface state COUNTRY_UPDATE->ACS
Mon May 20 11:47:54 2024 daemon.notice hostapd: wlan1-1: ACS-STARTED
Mon May 20 11:47:56 2024 daemon.notice netifd: radio1 (20140): sh: out of range
Mon May 20 11:47:56 2024 daemon.notice wpa_supplicant[2956]: wlan1: CTRL-EVENT-SCAN-FAILED ret=-16 retry=1
Mon May 20 11:47:56 2024 daemon.notice netifd: Interface 'wwan' is enabled
Mon May 20 11:47:57 2024 daemon.notice wpa_supplicant[2956]: wlan1: CTRL-EVENT-SCAN-FAILED ret=-16 retry=1
Mon May 20 11:47:58 2024 daemon.notice wpa_supplicant[2956]: wlan1: CTRL-EVENT-SCAN-FAILED ret=-16 retry=1
Mon May 20 11:47:58 2024 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Mon May 20 11:47:59 2024 daemon.notice wpa_supplicant[2956]: wlan1: CTRL-EVENT-SCAN-FAILED ret=-16 retry=1
Mon May 20 11:48:00 2024 daemon.notice wpa_supplicant[2956]: wlan1: CTRL-EVENT-SCAN-FAILED ret=-16 retry=1
Mon May 20 11:48:01 2024 kern.info kernel: [423798.837138] br-lan: port 2(wlan1-1) entered disabled state
Mon May 20 11:48:01 2024 kern.info kernel: [423798.845220] device wlan1-1 left promiscuous mode
Mon May 20 11:48:01 2024 kern.info kernel: [423798.850131] br-lan: port 2(wlan1-1) entered disabled state
Mon May 20 11:48:01 2024 daemon.err hostapd: ACS: Failed to request initial scan
Mon May 20 11:48:01 2024 daemon.err hostapd: ACS: Failed to request scan
Mon May 20 11:48:01 2024 daemon.err hostapd: Interface initialization failed
Mon May 20 11:48:01 2024 daemon.notice hostapd: wlan1-1: interface state ACS->DISABLED
Mon May 20 11:48:01 2024 daemon.notice hostapd: wlan1-1: AP-DISABLED
Mon May 20 11:48:01 2024 daemon.err hostapd: ACS: Failed to start
[... 171 lines cut ...]
Mon May 20 11:48:56 2024 daemon.warn hostapd: Failed to check if DFS is required; ret=-1
Mon May 20 11:48:56 2024 daemon.warn hostapd: Failed to check if DFS is required; ret=-1
Mon May 20 11:48:56 2024 daemon.err hostapd: Wrong coupling between HT and VHT/HE channel setting
Mon May 20 11:48:58 2024 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Mon May 20 11:48:59 2024 daemon.notice netifd: wwan (21839): udhcpc: broadcasting discover
Mon May 20 11:48:59 2024 daemon.notice netifd: wwan (21839): udhcpc: broadcasting select for 192.168.110.36, server 192.168.110.1
Mon May 20 11:48:59 2024 daemon.notice netifd: wwan (21839): udhcpc: lease of 192.168.110.36 obtained from 192.168.110.1, lease time 1800
Mon May 20 11:48:59 2024 daemon.notice netifd: Interface 'wwan' is now up
Mon May 20 11:49:00 2024 daemon.warn hostapd: Failed to check if DFS is required; ret=-1
Mon May 20 11:49:00 2024 daemon.warn hostapd: Failed to check if DFS is required; ret=-1
Mon May 20 11:49:00 2024 daemon.err hostapd: Wrong coupling between HT and VHT/HE channel setting
Mon May 20 11:49:06 2024 daemon.warn hostapd: Failed to check if DFS is required; ret=-1
Mon May 20 11:49:06 2024 daemon.warn hostapd: Failed to check if DFS is required; ret=-1
Mon May 20 11:49:06 2024 daemon.err hostapd: Wrong coupling between HT and VHT/HE channel setting
Mon May 20 11:49:08 2024 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Mon May 20 11:49:12 2024 user.notice firewall: Reloading firewall due to ifup of wwan (wlan1)
Mon May 20 11:49:12 2024 daemon.warn hostapd: Failed to check if DFS is required; ret=-1
Mon May 20 11:49:12 2024 daemon.warn hostapd: Failed to check if DFS is required; ret=-1
Mon May 20 11:49:12 2024 daemon.err hostapd: Wrong coupling between HT and VHT/HE channel setting
Mon May 20 11:49:13 2024 user.notice nlbwmon: Reloading nlbwmon due to ifup of wwan (wlan1)
Mon May 20 11:49:18 2024 daemon.warn hostapd: Failed to check if DFS is required; ret=-1
Mon May 20 11:49:18 2024 daemon.warn hostapd: Failed to check if DFS is required; ret=-1
Mon May 20 11:49:18 2024 daemon.err hostapd: Wrong coupling between HT and VHT/HE channel setting
Mon May 20 11:49:18 2024 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
```

wlshdjj, 2024-05-20:
为什么要有这么骚的操作呢？搜寻SSID加入时会自动设置为与上级AP相同的channel，中继后发射SSID也使用相同的channel，这是牛欢喜炒荷兰豆的事，也就是你设了除自找麻烦之外也没有任何意义。

No06, 2024-05-20:
> 为什么要有这么骚的操作呢？搜寻SSID加入时会自动设置为与上级AP相同的channel，中继后发射SSID也使用相同的channel，这是牛欢喜炒荷兰豆的事，也就是你设了除自找麻烦之外也没有任何意义。

上级AP每天自动重启然后会重设信道呢？

wlshdjj, 2024-05-20:
> > 为什么要有这么骚的操作呢？搜寻SSID加入时会自动设置为与上级AP相同的channel，中继后发射SSID也使用相同的channel，这是牛欢喜炒荷兰豆的事，也就是你设了除自找麻烦之外也没有任何意义。
> 
> 上级AP每天自动重启然后会重设信道呢？

会自动跟随，你完全不用管。

No06, 2024-05-20:
> > > 为什么要有这么骚的操作呢？搜寻SSID加入时会自动设置为与上级AP相同的channel，中继后发射SSID也使用相同的channel，这是牛欢喜炒荷兰豆的事，也就是你设了除自找麻烦之外也没有任何意义。
> > 
> > 
> > 上级AP每天自动重启然后会重设信道呢？
> 
> 会自动跟随，你完全不用管。

如果会我就不会开issus了

wlshdjj, 2024-05-20:
> > > > 为什么要有这么骚的操作呢？搜寻SSID加入时会自动设置为与上级AP相同的channel，中继后发射SSID也使用相同的channel，这是牛欢喜炒荷兰豆的事，也就是你设了除自找麻烦之外也没有任何意义。
> > > 
> > > 
> > > 上级AP每天自动重启然后会重设信道呢？
> > 
> > 
> > 会自动跟随，你完全不用管。
> 
> 如果会我就不会开issus了

openwrt无线中继我用了十年八年了好吧！上级wifi从来都是自动变channel的，只有TP的WPS不能自动跟随，tomato、ddwrt、openwrt、TP台版固件的wisp都可自动跟随，只和ssid有关，和channel、bssid无关，即使你设了，上级改动了，一样正常连接。

wlshdjj, 2024-05-20:
图一，上级13信道
[image]
我把上级AP改为9信道
几秒钟之内就自动跟随，现在也是9信道
[image]

No06, 2024-05-20:
https://github.com/coolsnowwolf/lede/assets/72285529/9416ed59-56ba-4da6-bc9b-cede28354918

然而在我这就是不会自动切，每天早上起来都是没网的状态

caixin1218, 2024-05-20:
> 2024-05-20.13-42-13.mp4 
> 然而在我这就是不会自动切，每天早上起来都是没网的状态

为什么两个BSSID是一样的？ 你这个网卡就不支持同时收发。

No06, 2024-05-20:
> > 2024-05-20.13-42-13.mp4
> > 然而在我这就是不会自动切，每天早上起来都是没网的状态
> 
> 为什么两个BSSID是一样的？ 你这个网卡就不支持同时收发。

意思是要网卡支持全双工才能在桥接自动切换信道？不相干吧

caixin1218, 2024-05-20:
> > > 2024-05-20.13-42-13.mp4
> > > 然而在我这就是不会自动切，每天早上起来都是没网的状态
> > 
> > 
> > 为什么两个BSSID是一样的？ 你这个网卡就不支持同时收发。
小米r3g这款路由器是可以这样设置的，你这款就不清楚了。单频单发，不支持单频多发。

coolsnowwolf, 2026-09-05:
已核对日志与讨论：该问题发生在同一块单频无线网卡同时承担 STA 中继和 AP，STA 连接上级 AP 后信道变化会触发本地 AP/ACS 重建；日志中的扫描占用（ret=-16）、ACS failed 及 hostapd 崩溃表明是硬件并发能力与旧版 hostapd/驱动限制，不是可由 feeds/packages 修复的包问题。当前主线无线栈没有针对 CR660x 单频同频中继的通用修复。建议使用双频/支持并发的无线设备，或固定上级信道。该 issue 已多年无新复现，现关闭。

closed 2026-09-05 by coolsnowwolf