coolsnowwolf/lede #12176: rax300m(nand版)编译固件失败，

yanzhx7, 2024-05-21:
### 详细叙述

[image]

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

rax3000m（nand版）

### 详细日志

mkdir -p /home/user/lede/dl
SHELL= flock /home/user/lede/tmp/.rustc-1.74.1-src.tar.gz.flock -c '    /home/user/lede/scripts/download.pl "/home/user/lede/dl" "rustc-1.74.1-src.tar.gz" "67db3e22fc9921c885baae5953ba144fc474cde29ec69ab56d43ce764206231d" "" "https://static.rust-lang.org/dist/"    '
+ curl -f --connect-timeout 20 --retry 5 --location https://static.rust-lang.org/dist/rustc-1.74.1-src.tar.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  270M  100  270M    0     0   417k      0  0:11:02  0:11:02 --:--:--  137k
. /home/user/lede/include/shell.sh; /home/user/lede/staging_dir/host/bin/libdeflate-gzip -dc /home/user/lede/dl/rustc-1.74.1-src.tar.gz | tar -C /home/user/lede/build_dir/target-aarch64_cortex-a53_musl/host/rustc-1.74.1-src/.. -xf -
libdeflate-gzip: Out of memory
libdeflate-gzip: "/home/user/lede/dl/rustc-1.74.1-src.tar.gz": file is probably too large to be processed by this program
tar: This does not look like a tar archive
tar: Exiting with failure status due to previous errors
make[3]: *** [Makefile:113: /home/user/lede/build_dir/target-aarch64_cortex-a53_musl/host/rustc-1.74.1-src/.prepareda4d01eafa336882699fbf43c1c50d3e4_6664517399ebbbc92a37c5bb081b5c53] Error 2
make[3]: Leaving directory '/home/user/lede/feeds/packages/lang/rust'
time: package/feeds/packages/rust/host-compile#18.67#17.38#663.85
    ERROR: package/feeds/packages/rust [host] failed to build.
make[2]: *** [package/Makefile:114: package/feeds/packages/rust/host/compile] Error 1
make[2]: Leaving directory '/home/user/lede'
make[1]: *** [package/Makefile:110: /home/user/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/user/lede'
make: *** [/home/user/lede/include/toplevel.mk:231：world] 错误 2

WYC-2020, 2024-05-22:
删掉 dl/rustc-1.74.1-src.tar.gz 在重新下载

xayane, 2024-05-23:
2024-05-22T10:38:52.3368204Z downloading https://ci-artifacts.rust-lang.org/rustc-builds/a28077b28a02b92985b3a3faecf92813155f1ea1/rust-dev-1.74.1-x86_64-unknown-linux-gnu.tar.xz
2024-05-22T10:38:52.6343212Z curl: (22) The requested URL returned error: 404
2024-05-22T10:38:52.6365998Z error: failed to download llvm from c

我也是编译这里报错了，很奇怪。

closed 2024-05-23 by coolsnowwolf