coolsnowwolf/lede #12178: X86编译报错，world 2

c455025066, 2024-05-22:
### 详细叙述

各位大佬，求问下面这个报错怎么解决？搜过了没搜到具体方法

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

X86/64

### 详细日志

[image]

c455025066, 2024-05-22:
time: package/feeds/helloworld/gn/host-compile#0.12#0.05#0.16
make[3]: Entering directory '/home/chenxiaoxin/lede/feeds/helloworld/naiveproxy'
mkdir -p /home/chenxiaoxin/lede/dl
SHELL= flock /home/chenxiaoxin/lede/tmp/.clang-llvmorg-19-init-8091-gab037c4f-1.tgz.flock -c '  	/home/chenxiaoxin/lede/scripts/download.pl "/home/chenxiaoxin/lede/dl" "clang-llvmorg-19-init-8091-gab037c4f-1.tgz" "6b797d118cc593eca0f2e568169295a5a9b65ddabed2de34b9a201a4bb7990f5" "clang-llvmorg-19-init-8091-gab037c4f-1.tgz" "https://commondatastorage.googleapis.com/chromium-browser-clang/Linux_x64"    '
+ curl -f --connect-timeout 20 --retry 5 --location https://commondatastorage.googleapis.com/chromium-browser-clang/Linux_x64/clang-llvmorg-19-init-8091-gab037c4f-1.tgz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (35) OpenSSL SSL_connect: Connection reset by peer in connection to commondatastorage.googleapis.com:443 
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location https://sources.cdn.openwrt.org/clang-llvmorg-19-init-8091-gab037c4f-1.tgz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0   146    0     0    0     0      0      0 --:--:--  0:00:01 --:--:--     0
curl: (22) The requested URL returned error: 404
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location https://sources.openwrt.org/clang-llvmorg-19-init-8091-gab037c4f-1.tgz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0   146    0     0    0     0      0      0 --:--:--  0:00:02 --:--:--     0
curl: (22) The requested URL returned error: 404
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location https://mirror2.openwrt.org/sources/clang-llvmorg-19-init-8091-gab037c4f-1.tgz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0   153    0     0    0     0      0      0 --:--:--  0:00:02 --:--:--     0
curl: (22) The requested URL returned error: 404
Download failed.
No more mirrors to try - giving up.
make[3]: *** [Makefile:111: /home/chenxiaoxin/lede/dl/clang-llvmorg-19-init-8091-gab037c4f-1.tgz] Error 2
make[3]: Leaving directory '/home/chenxiaoxin/lede/feeds/helloworld/naiveproxy'
time: package/feeds/helloworld/naiveproxy/compile#0.23#0.14#6.84
    ERROR: package/feeds/helloworld/naiveproxy failed to build.
make[2]: *** [package/Makefile:116: package/feeds/helloworld/naiveproxy/compile] Error 1
make[2]: Leaving directory '/home/chenxiaoxin/lede'
make[1]: *** [package/Makefile:110: /home/chenxiaoxin/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/chenxiaoxin/lede'
make: *** [/home/chenxiaoxin/lede/include/toplevel.mk:231：world] 错误 2

CYaNu, 2024-05-22:
https://commondatastorage.googleapis.com/chromium-browser-clang/Linux_x64/clang-llvmorg-19-init-8091-gab037c4f-1.tgz
没有问题 
请检查你的梯子

closed 2024-05-23 by coolsnowwolf