coolsnowwolf/lede #12179: 请求修复RA5JDC设备

f1owkang, 2024-05-23:
### 详细叙述

听您说是uboot 参数问题，我尝试自己修复了RA5的设备树。但是无法编译通过，又尝试移植了https://github.com/JiaY-shi/openwrt的，也是出现很多问题例如
`  CC      drivers/thermal/thermal_trip.o
drivers/thermal/thermal_trip.c: In function '__thermal_zone_get_trip':
drivers/thermal/thermal_trip.c:118:9: warning: this 'if' clause does not guard... [-Wmisleading-indentation]
  118 |         if (!tz || trip_id < 0 || trip_id >= tz->num_trips || !trip)
      |         ^~
drivers/thermal/thermal_trip.c:121:17: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the 'if'
  121 |                 if (tz->trips) {
      |                 ^~
`
种种无奈请求大佬修复或者指点一下，感激不尽

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

RA50

### 详细日志

lede正常编译，编译固件无法启动

f1owkang, 2024-05-23:
这里是LEDE编译固件刷入之后产生的日志，刚启动之后一会儿死机
[   13.747209] EXT4-fs (loop0): mounted filesystem with ordered data mode. Quota mode: disabled.
[   13.747397] block: attempting to load /tmp/ext4_cfg/upper/etc/config/fstab
[   13.754892] block: unable to load configuration (fstab: Entry not found)
[   13.761536] block: attempting to load /tmp/ext4_cfg/etc/config/fstab
[   13.768399] block: unable to load configuration (fstab: Entry not found)
[   13.774797] block: attempting to load /etc/config/fstab
[   13.783414] block: unable to load configuration (fstab: Entry not found)
[   13.786336] block: no usable configuration
[   13.793605] EXT4-fs (loop0): unmounting filesystem.
[   13.798546] loop0: detected capacity change from 0 to 231464
[   13.851367] loop0: detected capacity change from 231464 to 102568
[   13.855040] EXT4-fs (loop0): mounted filesystem with ordered data mode. Quota mode: disabled.
[   13.979154] block: attempting to load /tmp/ext4_cfg/upper/etc/config/fstab
[   13.979248] block: unable to load configuration (fstab: Entry not found)
[   13.985015] block: attempting to load /tmp/ext4_cfg/etc/config/fstab
[   13.991930] block: unable to load configuration (fstab: Entry not found)
[   13.998150] block: attempting to load /etc/config/fstab
[   14.004854] block: unable to load configuration (fstab: Entry not found)
[   14.009778] block: no usable configuration
[   14.017653] mount_root: overlay filesystem has not been fully initialized yet
[   14.021417] mount_root: switching to ext4 overlay
[   14.030894] overlayfs: null uuid detected in lower fs '/', falling back to xino=off,index=off,nfs_export=off.
[   14.037720] urandom-seed: Seed file not found (/etc/urandom.seed)
Scanning for Btrfs filesystems
[   14.459796] procd: - early -
[   14.459899] procd: - watchdog -
[   15.080391] procd: - watchdog -
[   15.132556] procd: - ubus -
[   15.322555] procd: - init -
Please press Enter to activate this console.
[   15.833179] urngd: v1.0.2 started.
[   16.466189] kmodloader: loading kernel modules from /etc/modules.d/*
[   16.557935] NET: Registered PF_ALG protocol family
[   16.568800] FS-Cache: Loaded
[   16.570006] CacheFiles: Loaded
[   16.572523] Key type dns_resolver registered
[   16.605932] NET: Registered PF_RXRPC protocol family
[   16.605970] Key type rxrpc registered
[   16.609961] Key type rxrpc_s registered
[   16.616170] kAFS: Red Hat AFS client v0.1 registering.
[   16.751369] kAFS: failed to register: -97
[   16.857637] tun: Universal TUN/TAP device driver, 1.6
[   16.868229] sit: IPv6, IPv4 and MPLS over IPv4 tunneling driver
[   16.961197] cryptodev: driver 1.13 loaded.
[   16.964768] ntfs3: Max link count 4000
[   16.964805] ntfs3: Enabled Linux POSIX ACLs support
[   17.001160] Bridge firewalling registered
[   17.001820] Loading modules backported from Linux version v6.1.24-0-g0102425ac76b
[   17.004157] Backport generated by backports.git v5.15.92-1-44-gd6ea70fafd36
[   17.059383] NET: Registered PF_QIPCRTR protocol family
[   17.065238] usbcore: registered new interface driver ums-alauda
[   17.066312] usbcore: registered new interface driver ums-cypress
[   17.071008] usbcore: registered new interface driver ums-datafab
[   17.077195] usbcore: registered new interface driver ums-freecom
[   17.083212] usbcore: registered new interface driver ums-isd200
[   17.089119] usbcore: registered new interface driver ums-jumpshot
[   17.094726] usbcore: registered new interface driver ums-karma
[   17.101121] usbcore: registered new interface driver ums-sddr09
[   17.106786] usbcore: registered new interface driver ums-sddr55
[   17.112674] usbcore: registered new interface driver ums-usbat
[   17.130302] xt_time: kernel timezone is -0000
[   17.150693] PPP generic driver version 2.4.2
[   17.151764] PPP MPPE Compression module registered
[   17.154976] NET: Registered PF_PPPOX protocol family
[   17.166919] ath11k c000000.wifi: ipq6018 hw1.0
[   17.166954] ath11k c000000.wifi: FW memory mode: 1
[   17.170581] remoteproc remoteproc0: powering up cd00000.remoteproc
[   17.175964] remoteproc remoteproc0: Booting fw image IPQ6018/q6_fw.mdt, size 7340
[   17.857692] remoteproc remoteproc0: remote processor cd00000.remoteproc is now up
[... 93 lines cut ...]
B -     66147 - auth_xbl_sec_hash_seg_entry, Start
B -     66289 - auth_xbl_sec_hash_seg_exit, Start
B -     72834 - xbl_sec_segs_hash_verify_entry, Start
B -     72835 - xbl_sec_segs_hash_verify_exit, Start
B -     73764 - PBL, End
B -     65239 - SBL1, Start
B -    205295 - GCC [RstStat:0x2, RstDbg:0x420100] WDog Stat : 0x4
B -    207735 - clock_init, Start
D -      2897 - clock_init, Delta
B -    216367 - boot_flash_init, Start
D -     55022 - boot_flash_init, Delta
B -    274683 - sbl1_ddr_set_default_params, Start
D -       244 - sbl1_ddr_set_default_params, Delta
B -    281118 - boot_config_data_table_init, Start
D -      2165 - boot_config_data_table_init, Delta - (575 Bytes)
B -    290268 - CDT Version:2,Platform ID:8,Major ID:3,Minor ID:2,Subtype:0
B -    295819 - Image Load, Start
D -      6588 - OEM_MISC Image Loaded, Delta - (0 Bytes)
B -    305152 - Image Load, Start
D -      5063 - PMIC Image Loaded, Delta - (0 Bytes)
B -    313021 - sbl1_ddr_set_params, Start
B -    317993 - CPR configuration: 0x366
B -    321195 - Pre_DDR_clock_init, Start
D -       213 - Pre_DDR_clock_init, Delta
D -         0 - sbl1_ddr_set_params, Delta

f1owkang, 2024-05-26:
最近找到内核6.6的可用

closed 2024-05-26 by f1owkang

FreshKai, 2024-05-31:
您好，有RA50 JDC修复后的固件吗？
我看您主页仓库中的[OpenWrt](https://github.com/f1owkang/OpenWrt)中的描述支持JDC-AX5，是这个吗？
 @f1owkang

f1owkang, 2024-05-31:
> 您好，有RA50 JDC修复后的固件吗？ 我看您主页仓库中的[OpenWrt](https://github.com/f1owkang/OpenWrt)中的描述支持JDC-AX5，是这个吗？ @f1owkang

是的，但是需要自己稍加修改

FreshKai, 2024-05-31:
> > 您好，有RA50 JDC修复后的固件吗？ 我看您主页仓库中的[OpenWrt](https://github.com/f1owkang/OpenWrt)中的描述支持JDC-AX5，是这个吗？ @f1owkang
> 
> 是的，但是需要自己稍加修改

指的是配置文件吗？还是需要在代码层面改动呢？

f1owkang, 2024-05-31:
> > > 您好，有RA50 JDC修复后的固件吗？ 我看您主页仓库中的[OpenWrt](https://github.com/f1owkang/OpenWrt)中的描述支持JDC-AX5，是这个吗？ @f1owkang
> > 
> > 
> > 是的，但是需要自己稍加修改
> 
> 指的是配置文件吗？还是需要在代码层面改动呢？

配置文件

FreshKai, 2024-05-31:
> > > > 您好，有RA50 JDC修复后的固件吗？ 我看您主页仓库中的[OpenWrt](https://github.com/f1owkang/OpenWrt)中的描述支持JDC-AX5，是这个吗？ @f1owkang
> > > 
> > > 
> > > 是的，但是需要自己稍加修改
> > 
> > 
> > 指的是配置文件吗？还是需要在代码层面改动呢？
> 
> 配置文件

好的，感谢您的工作