coolsnowwolf/lede #12183: 编译报错

pascalqq, 2024-05-24:
### 详细叙述

使用master，默认config，启用了helloworld。

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86-64

### 详细日志

ake[3]: Leaving directory '/home/tony/lede/feeds/packages/libs/c-ares'
time: package/feeds/packages/c-ares/compile#0.21#0.24#0.39
make[3]: Entering directory '/home/tony/lede/package/libs/mbedtls'
mkdir -p /home/tony/lede/staging_dir/target-x86_64_musl/root-x86/stamp
echo '12' | cmp -s - /home/tony/lede/staging_dir/target-x86_64_musl/pkginfo/libmbedtls.version || { echo '12' > /home/tony/lede/staging_dir/target-x86_64_musl/pkginfo/libmbedtls.version;  } 
SHELL= flock /home/tony/lede/tmp/.root-copy.flock -c 'cp -fpR /home/tony/lede/build_dir/target-x86_64_musl/mbedtls-2.28.5/.pkgdir/libmbedtls/. /home/tony/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/tony/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.libmbedtls_installed
echo "libmbedtls" >> /home/tony/lede/staging_dir/target-x86_64_musl/pkginfo/mbedtls.default.install
make[3]: Leaving directory '/home/tony/lede/package/libs/mbedtls'
time: package/libs/mbedtls/compile#0.25#0.29#0.47
make[3]: Entering directory '/home/tony/lede/package/libs/pcre'
mkdir -p /home/tony/lede/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/tony/lede/tmp/.root-copy.flock -c 'cp -fpR /home/tony/lede/build_dir/target-x86_64_musl/pcre-8.45/.pkgdir/libpcre/. /home/tony/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/tony/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.libpcre_installed
echo "libpcre" >> /home/tony/lede/staging_dir/target-x86_64_musl/pkginfo/pcre.default.install
make[3]: Leaving directory '/home/tony/lede/package/libs/pcre'
time: package/libs/pcre/compile#0.52#0.35#0.79
make[3]: Entering directory '/home/tony/lede/feeds/packages/net/shadowsocks-libev'
mkdir -p /home/tony/lede/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/tony/lede/tmp/.root-copy.flock -c 'cp -fpR /home/tony/lede/build_dir/target-x86_64_musl/shadowsocks-libev-3.3.5/.pkgdir/shadowsocks-libev-ss-server/. /home/tony/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/tony/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.shadowsocks-libev-ss-server_installed
echo "shadowsocks-libev-ss-server" >> /home/tony/lede/staging_dir/target-x86_64_musl/pkginfo/shadowsocks-libev.default.install
make[3]: Leaving directory '/home/tony/lede/feeds/packages/net/shadowsocks-libev'
time: package/feeds/packages/shadowsocks-libev/compile#0.23#0.25#0.42
make[3]: Entering directory '/home/tony/lede/feeds/packages/lang/rust'
CARGO_HOME=/home/tony/lede/dl/cargo TARGET_CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/tony/lede/build_dir/target-x86_64_musl/rust-1.74.1=rust-1.74.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" OPENWRT_RUSTC_BOOTSTRAP_CACHE=/home/tony/lede/dl/rustc python3 /home/tony/lede/build_dir/target-x86_64_musl/host/rustc-1.74.1-src/x.py --build-dir /home/tony/lede/build_dir/target-x86_64_musl/host/rustc-1.74.1-src/build --config /home/tony/lede/build_dir/target-x86_64_musl/host/rustc-1.74.1-src/config.toml dist build-manifest cargo llvm-tools rustc rust-std rust-src
Building bootstrap
    Finished dev [unoptimized] target(s) in 0.16s
downloading https://ci-artifacts.rust-lang.org/rustc-builds/a28077b28a02b92985b3a3faecf92813155f1ea1/rust-dev-1.74.1-x86_64-unknown-linux-gnu.tar.xz
curl: (22) The requested URL returned error: 404                                                                                                     

error: failed to download llvm from ci

    help: old builds get deleted after a certain time
    help: if trying to compile an old commit of rustc, disable `download-ci-llvm` in config.toml:

    [llvm]
    download-ci-llvm = false
    
Build completed unsuccessfully in 0:00:01
make[3]: *** [Makefile:111: /home/tony/lede/build_dir/target-x86_64_musl/host/rustc-1.74.1-src/.built] Error 1
make[3]: Leaving directory '/home/tony/lede/feeds/packages/lang/rust'
time: package/feeds/packages/rust/host-compile#0.59#0.63#2.18
    ERROR: package/feeds/packages/rust [host] failed to build.
make[2]: *** [package/Makefile:114: package/feeds/packages/rust/host/compile] Error 1
make[2]: Leaving directory '/home/tony/lede'
make[1]: *** [package/Makefile:110: /home/tony/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/tony/lede'
make: *** [/home/tony/lede/include/toplevel.mk:231: world] Error 2

lirg-111, 2024-05-24:
梯子问题吧？

pascalqq, 2024-05-24:
> 梯子问题吧？

你试试上面url你访问有东西么？

coolsnowwolf, 2024-05-24:
更新 feeds

pascalqq, 2024-05-24:
已解决

closed 2024-05-24 by pascalqq