coolsnowwolf/lede #12184: 编译出错

Kuron88, 2024-05-25:
### 详细叙述

全新拉取的源码，拉取了passwall的插件，然后编译就出错

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

CR660X

### 详细日志

{standard input}: Assembler messages:
{standard input}:4260: Error: opcode not supported on this processor: mips2 (mips2) `sync'
{standard input}:4535: Error: opcode not supported on this processor: mips2 (mips2) `sync'
make[4]: *** [Makefile:25: libev/ev.o] Error 1
make[4]: Leaving directory '/home/kuron/lede/build_dir/target-mipsel_24kc_musl/dns2tcp-1.1.1'
make[3]: *** [Makefile:45: /home/kuron/lede/build_dir/target-mipsel_24kc_musl/dns2tcp-1.1.1/.built] Error 2
make[3]: Leaving directory '/home/kuron/lede/feeds/passwall/dns2tcp'
time: package/feeds/passwall/dns2tcp/compile#0.40#0.12#0.52
    ERROR: package/feeds/passwall/dns2tcp failed to build.
make[2]: *** [package/Makefile:116: package/feeds/passwall/dns2tcp/compile] Error 1
make[2]: Leaving directory '/home/kuron/lede'
make[1]: *** [package/Makefile:110: /home/kuron/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/kuron/lede'
make: *** [/home/kuron/lede/include/toplevel.mk:231：world] 错误 2

coolsnowwolf, 2024-05-26:
dns2tcp 别用他们的版本就好了

coolsnowwolf, 2024-05-26:
[image]

Asyncload, 2024-05-26:
[image]
好像是差不多的错误 但是好像这个方法不能解决

closed 2024-05-28 by coolsnowwolf