coolsnowwolf/lede #12189: 为rg-EW3200X Pro编译wpad-basic-wolfssl出错

asterwyx, 2024-05-25:
### 详细叙述

我在使用ActionsOpenwrt为我的X32 Pro编译固件，因为device tree中的flash layout发生了变动，现在OpenWRT官方编译的固件已经不能够启动了。现在问题是在编译wpad-basic-wolfssl的时候，我遇到了一个链接问题。我在 #11201 中看到了类似的报错，也看到了有人发布了补丁，但是我看到主线中似乎并未包含该补丁，并且我也不确定该补丁是否可用，因为我找不到diff文件中指定的文件，其他的补丁也是这样，我不是很清楚这些patch是怎样工作的。

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

rg-EW3200X Pro

### 详细日志

```bash
export MAKEFLAGS=""; aarch64-openwrt-linux-musl-gcc -o /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/hostapd-wpad-basic-wolfssl/hostapd-2022-07-29-b704dc72/wpad -Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/hostapd-wpad-basic-wolfssl/hostapd-2022-07-29-b704dc72=hostapd-2022-07-29-b704dc72 -Wformat -Werror=format-security -DPIC -fPIC -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro ./files/multicall.c /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/hostapd-wpad-basic-wolfssl/hostapd-2022-07-29-b704dc72/hostapd/hostapd_multi.a /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/hostapd-wpad-basic-wolfssl/hostapd-2022-07-29-b704dc72/wpa_supplicant/wpa_supplicant_multi.a -L/workdir/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/usr/lib -L/workdir/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/lib -DPIC -fPIC -specs=/workdir/openwrt/include/hardened-ld-pie.specs -znow -zrelro -lwolfssl -lubox -lubus -lm -lnl-tiny
/workdir/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/lib/gcc/aarch64-openwrt-linux-musl/11.3.0/../../../../aarch64-openwrt-linux-musl/bin/ld: /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/hostapd-wpad-basic-wolfssl/hostapd-2022-07-29-b704dc72/wpa_supplicant/wpa_supplicant_multi.a(mbo.o): in function `mbo_build_anqp_buf':
mbo.c:(.text+0xe8c): undefined reference to `gas_anqp_add_element'
/workdir/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/lib/gcc/aarch64-openwrt-linux-musl/11.3.0/../../../../aarch64-openwrt-linux-musl/bin/ld: mbo.c:(.text+0xe94): undefined reference to `gas_anqp_add_element'
/workdir/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/lib/gcc/aarch64-openwrt-linux-musl/11.3.0/../../../../aarch64-openwrt-linux-musl/bin/ld: mbo.c:(.text+0xefc): undefined reference to `gas_anqp_set_element_len'
/workdir/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/lib/gcc/aarch64-openwrt-linux-musl/11.3.0/../../../../aarch64-openwrt-linux-musl/bin/ld: mbo.c:(.text+0xf08): undefined reference to `gas_anqp_set_element_len'
collect2: error: ld returned 1 exit status
make[3]: *** [Makefile:794: /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/hostapd-wpad-basic-wolfssl/hostapd-2022-07-29-b704dc72/.built] Error 1
make[3]: Leaving directory '/workdir/openwrt/package/network/services/hostapd'
time: package/network/services/hostapd/wpad-basic-wolfssl/compile#0.33#0.07#0.38
    ERROR: package/network/services/hostapd failed to build (build variant: wpad-basic-wolfssl).
make[2]: *** [package/Makefile:116: package/network/services/hostapd/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2
Error: Process completed with exit code 2.
```

asterwyx, 2024-05-25:
编译流程在[该仓库](https://github.com/asterwyx/ActionsOpenwrt)的工作流中

coolsnowwolf, 2024-05-26:
已经修复了

missorforget, 2024-05-27:
Applying ./patches/993-fix-mbo-module-build.patch using plaintext: 
patching file wpa_supplicant/Makefile
Hunk #1 FAILED at 339.
1 out of 1 hunk FAILED -- saving rejects to file wpa_supplicant/Makefile.rej
Patch failed!  Please fix ./patches/993-fix-mbo-module-build.patch!
make[3]: *** [Makefile:794: /home/lede/lean/build_dir/target-x86_64_musl/hostapd-wpad-full-openssl/hostapd-2022-07-29-b704dc72/.prepared_40a8310f21e48ad5c524ced2ebfbf31c_6664517399ebbbc92a37c5bb081b5c53] Error 1
make[3]: Leaving directory '/home/lede/lean/package/network/services/hostapd'
time: package/network/services/hostapd/wpad-full-openssl/compile#0.68#0.26#4.76
    ERROR: package/network/services/hostapd failed to build (build variant: wpad-full-openssl).
make[2]: *** [package/Makefile:116: package/network/services/hostapd/compile] Error 1
make[2]: Leaving directory '/home/lede/lean'
make[1]: *** [package/Makefile:110: /home/lede/lean/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2

asterwyx, 2024-05-27:
看起来这个patch和最新的版本是不兼容的。

missorforget, 2024-05-27:
https://github.com/coolsnowwolf/lede/issues/11748

WYC-2020, 2024-05-27:
> Applying ./patches/993-fix-mbo-module-build.patch using plaintext: patching file wpa_supplicant/Makefile Hunk #1 FAILED at 339. 1 out of 1 hunk FAILED -- saving rejects to file [...]

自己看看 是不是有另外一个补丁也修复了这个地方

closed 2024-05-29 by coolsnowwolf