coolsnowwolf/lede #12194: 红米ac2100编译出错

bobandbrony, 2024-05-27:
### 详细叙述

https://y5hwpw3fsi.feishu.cn/docx/UO0wdHReDoUdewxwlpGckSjin3d
使用这个教程部署部署的

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

红米ac2100  cpu是mtk7621at

### 详细日志

make[4]: Entering directory '/workdir/openwrt/build_dir/target-mipsel_24kc_musl/dns2tcp-1.1.1'
mipsel-openwrt-linux-musl-gcc -std=c99 -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -msoft-float -mips16 -minterlink-mips16 -fmacro-prefix-map=/workdir/openwrt/build_dir/target-mipsel_24kc_musl/dns2tcp-1.1.1=dns2tcp-1.1.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DPIC -fpic -c libev/ev.c -o libev/ev.o
libev/ev.c:2139:31: warning: 'ev_default_loop_ptr' initialized and declared 'extern'
   EV_API_DECL struct ev_loop *ev_default_loop_ptr = 0; /* needs to be initialised to make it a definition despite extern */
                               ^~~~~~~~~~~~~~~~~~~
{standard input}: Assembler messages:
{standard input}:4260: Error: opcode not supported on this processor: mips2 (mips2) `sync'
{standard input}:4535: Error: opcode not supported on this processor: mips2 (mips2) `sync'
make[4]: *** [Makefile:25: libev/ev.o] Error 1
make[4]: Leaving directory '/workdir/openwrt/build_dir/target-mipsel_24kc_musl/dns2tcp-1.1.1'
make[3]: *** [Makefile:45: /workdir/openwrt/build_dir/target-mipsel_24kc_musl/dns2tcp-1.1.1/.built] Error 2
make[3]: Leaving directory '/workdir/openwrt/feeds/small/dns2tcp'
time: package/feeds/small/dns2tcp/compile#0.36#0.08#0.42
    ERROR: package/feeds/small/dns2tcp failed to build.
make[2]: *** [package/Makefile:116: package/feeds/small/dns2tcp/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2
Error: Process completed with exit code 2.

WYC-2020, 2024-05-27:
xiaorouji 那边更新有问题 
[image]
PKG_USE_MIPS16:=0这个保留 在编译试试

bobandbrony, 2024-05-27:
https://github.com/bobandbrony/OpenWrt-
在这里编译，该如何操作呢

bobandbrony, 2024-05-27:
云编译依然不行，但本地编译成功了
[image]
谢谢！

Asyncload, 2024-05-27:
> 云编译依然不行，但本地编译成功了
> [image]
> 谢谢！

你本地编译修改了哪里？

closed 2024-05-29 by coolsnowwolf

eearphon, 2024-05-30:
> xiaorouji 那边更新有问题 [...]

谢谢，加回这句编译成功，newifi-d1