coolsnowwolf/lede #12198: Hostapd错误

qqsir-dev, 2024-05-28:
### 详细叙述

make[3] -C package/network/services/hostapd compile
    ERROR: package/network/services/hostapd failed to build (build variant: wpad-full-openssl).
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [/home/runner/work/AutoLede/AutoLede/openwrt/include/toplevel.mk:231: world] Error 1
make[2]: Entering directory '/home/runner/work/AutoLede/AutoLede/openwrt/scripts/config'

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86软路由

### 详细日志

make[3] -C package/network/services/hostapd compile
    ERROR: package/network/services/hostapd failed to build (build variant: wpad-full-openssl).
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [/home/runner/work/AutoLede/AutoLede/openwrt/include/toplevel.mk:231: world] Error 1
make[2]: Entering directory '/home/runner/work/AutoLede/AutoLede/openwrt/scripts/config'

ddnsoft, 2024-05-29:
同款错误：make[3] -C package/network/services/hostapd compile
    ERROR: package/network/services/hostapd failed to build (build variant: wpad-basic).
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [/home/runner/work/OpenWrt/OpenWrt/openwrt/include/toplevel.mk:231: world] Error 1

hzjnet, 2024-05-29:
提示很清楚啊： Please re-run make with -j1 V=s or V=sc  
用单线程编译

ddnsoft, 2024-05-29:
> 提示呃：请使用 -j1 V=s 或 V=sc 用单线程编译重新运行 make

单线程只是能显示详细的信息，出错还是要出错的

qqsir-dev, 2024-05-29:
这个在Arm架构编译不出错，好像只在X86

qqsir-dev, 2024-05-31:
这个咋整啊，X86一直编译出这个错误，我这水平不高，有没有简单粗暴的办法也行

qqsir-dev, 2024-06-02:
@coolsnowwolf 

patching file wpa_supplicant/Makefile
Hunk #1 FAILED at 339.
1 out of 1 hunk FAILED -- saving rejects to file wpa_supplicant/Makefile.rej
Patch failed!  Please fix ./patches/993-fix-mbo-module-build.patch!
make[3]: *** [Makefile:794: /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/hostapd-wpad-full-openssl/hostapd-2022-07-29-b704dc72/.prepared_2c264aaf7e748e32cd7f2287dea1eb70_6664517399ebbbc92a37c5bb081b5c53] Error 1
make[3]: Leaving directory '/home/runner/work/AutoLede/AutoLede/openwrt/package/network/services/hostapd'
time: package/network/services/hostapd/wpad-full-openssl/compile#0.58#0.39#0.81
    ERROR: package/network/services/hostapd failed to build (build variant: wpad-full-openssl).
make[2]: *** [package/Makefile:116: package/network/services/hostapd/compile] Error 1
make[2]: Leaving directory '/home/runner/work/AutoLede/AutoLede/openwrt'
make[1]: *** [package/Makefile:110: /home/runner/work/AutoLede/AutoLede/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/runner/work/AutoLede/AutoLede/openwrt'
make: *** [/home/runner/work/AutoLede/AutoLede/openwrt/include/toplevel.mk:231: world] Error 2
Error: Process completed with exit code 2.

qqsir-dev, 2024-06-04:
我错了，以前写过一个布丁，忘记注释了
# 修复 hostapd 报错
# cp -f $GITHUB_WORKSPACE/script/011-fix-mbo-modules-build.patch package/network/services/hostapd/patches/011-fix-mbo-modules-build.patch
注释后号了，Sorry

closed 2024-06-04 by qqsir-dev

ddnsoft, 2024-06-05:
> 我错误地写过一块布丁，忘记了留念
> 
> # 修复 hostapd 报错
> # cp -f $GITHUB_WORKSPACE/script/011-fix-mbo-modules-build.patch 包/网络/服务/hostapd/patches/011-fix-mbo-modules-build.patch
> 注释号了，抱歉

今早编译还是出错呀，请问具体要怎么操作？