coolsnowwolf/lede #12199: 新编固件Docker 26.1.1正常运行后台日志每分钟报错一次

tsingui, 2024-05-29:
### 详细叙述

Wed May 29 07:00:02 2024 daemon.err dockerd[6451]: 2024/05/29 07:00:02 http: superfluous response.WriteHeader call from go.opentelemetry.io/contrib/instrumentation/net/http/otelhttp.(*respWriterWrapper).WriteHeader (wrap.go:98)
Wed May 29 07:01:03 2024 daemon.err dockerd[6451]: 2024/05/29 07:01:03 http: superfluous response.WriteHeader call from go.opentelemetry.io/contrib/instrumentation/net/http/otelhttp.(*respWriterWrapper).WriteHeader (wrap.go:98)
Wed May 29 07:02:03 2024 daemon.err dockerd[6451]: 2024/05/29 07:02:03 http: superfluous response.WriteHeader call from go.opentelemetry.io/contrib/instrumentation/net/http/otelhttp.(*respWriterWrapper).WriteHeader (wrap.go:98)
Wed May 29 07:03:03 2024 daemon.err dockerd[6451]: 2024/05/29 07:03:03 http: superfluous response.WriteHeader call from go.opentelemetry.io/contrib/instrumentation/net/http/otelhttp.(*respWriterWrapper).WriteHeader (wrap.go:98)
Wed May 29 07:04:02 2024 daemon.err dockerd[6451]: 2024/05/29 07:04:02 http: superfluous response.WriteHeader call from go.opentelemetry.io/contrib/instrumentation/net/http/otelhttp.(*respWriterWrapper).WriteHeader (wrap.go:98)
Wed May 29 07:05:03 2024 daemon.err dockerd[6451]: 2024/05/29 07:05:03 http: superfluous response.WriteHeader call from go.opentelemetry.io/contrib/instrumentation/net/http/otelhttp.(*respWriterWrapper).WriteHeader (wrap.go:98)
Wed May 29 07:06:03 2024 daemon.err dockerd[6451]: 2024/05/29 07:06:03 http: superfluous response.WriteHeader call from go.opentelemetry.io/contrib/instrumentation/net/http/otelhttp.(*respWriterWrapper).WriteHeader (wrap.go:98)
Wed May 29 07:07:03 2024 daemon.err dockerd[6451]: 2024/05/29 07:07:03 http: superfluous response.WriteHeader call from go.opentelemetry.io/contrib/instrumentation/net/http/otelhttp.(*respWriterWrapper).WriteHeader (wrap.go:98)
Wed May 29 07:08:03 2024 daemon.err dockerd[6451]: 2024/05/29 07:08:03 http: superfluous response.WriteHeader call from go.opentelemetry.io/contrib/instrumentation/net/http/otelhttp.(*respWriterWrapper).WriteHeader (wrap.go:98)
Wed May 29 07:09:02 2024 daemon.err dockerd[6451]: 2024/05/29 07:09:02 http: superfluous response.WriteHeader call from go.opentelemetry.io/contrib/instrumentation/net/http/otelhttp.(*respWriterWrapper).WriteHeader (wrap.go:98)
Wed May 29 07:10:03 2024 daemon.err dockerd[6451]: 2024/05/29 07:10:03 http: superfluous response.WriteHeader call from go.opentelemetry.io/contrib/instrumentation/net/http/otelhttp.(*respWriterWrapper).WriteHeader (wrap.go:98)
Wed May 29 07:11:03 2024 daemon.err dockerd[6451]: 2024/05/29 07:11:03 http: superfluous response.WriteHeader call from go.opentelemetry.io/contrib/instrumentation/net/http/otelhttp.(*respWriterWrapper).WriteHeader (wrap.go:98)
Wed May 29 07:12:03 2024 daemon.err dockerd[6451]: 2024/05/29 07:12:03 http: superfluous response.WriteHeader call from go.opentelemetry.io/contrib/instrumentation/net/http/otelhttp.(*respWriterWrapper).WriteHeader (wrap.go:98)
Wed May 29 07:13:03 2024 daemon.err dockerd[6451]: 2024/05/29 07:13:03 http: superfluous response.WriteHeader call from go.opentelemetry.io/contrib/instrumentation/net/http/otelhttp.(*respWriterWrapper).WriteHeader (wrap.go:98)
Wed May 29 07:14:02 2024 daemon.err dockerd[6451]: 2024/05/29 07:14:02 http: superfluous response.WriteHeader call from go.opentelemetry.io/contrib/instrumentation/net/http/otelhttp.(*respWriterWrapper).WriteHeader (wrap.go:98)
Wed May 29 07:15:03 2024 daemon.err dockerd[6451]: 2024/05/29 07:15:03 http: superfluous response.WriteHeader call from go.opentelemetry.io/contrib/instrumentation/net/http/otelhttp.(*respWriterWrapper).WriteHeader (wrap.go:98)
Wed May 29 07:16:03 2024 daemon.err dockerd[6451]: 2024/05/29 07:16:03 http: superfluous response.WriteHeader call from go.opentelemetry.io/contrib/instrumentation/net/http/otelhttp.(*respWriterWrapper).WriteHeader (wrap.go:98)
Wed May 29 07:17:03 2024 daemon.err dockerd[6451]: 2024/05/29 07:17:03 http: superfluous response.WriteHeader call from go.opentelemetry.io/contrib/instrumentation/net/http/otelhttp.(*respWriterWrapper).WriteHeader (wrap.go:98)
Wed May 29 07:18:03 2024 daemon.err dockerd[6451]: 2024/05/29 07:18:03 http: superfluous response.WriteHeader call from go.opentelemetry.io/contrib/instrumentation/net/http/otelhttp.(*respWriterWrapper).WriteHeader (wrap.go:98)
Wed May 29 07:19:02 2024 daemon.err dockerd[6451]: 2024/05/29 07:19:02 http: superfluous response.WriteHeader call from go.opentelemetry.io/contrib/instrumentation/net/http/otelhttp.(*respWriterWrapper).WriteHeader (wrap.go:98)
Wed May 29 07:20:03 2024 daemon.err dockerd[6451]: 2024/05/29 07:20:03 http: superfluous response.WriteHeader call from go.opentelemetry.io/contrib/instrumentation/net/http/otelhttp.(*respWriterWrapper).WriteHeader (wrap.go:98)
Wed May 29 07:21:03 2024 daemon.err dockerd[6451]: 2024/05/29 07:21:03 http: superfluous response.WriteHeader call from go.opentelemetry.io/contrib/instrumentation/net/http/otelhttp.(*respWriterWrapper).WriteHeader (wrap.go:98)
Wed May 29 07:22:03 2024 daemon.err dockerd[6451]: 2024/05/29 07:22:03 http: superfluous response.WriteHeader call from go.opentelemetry.io/contrib/instrumentation/net/http/otelhttp.(*respWriterWrapper).WriteHeader (wrap.go:98)
Wed May 29 07:23:03 2024 daemon.err dockerd[6451]: 2024/05/29 07:23:03 http: superfluous response.WriteHeader call from go.opentelemetry.io/contrib/instrumentation/net/http/otelhttp.(*respWriterWrapper).WriteHeader (wrap.go:98)
Wed May 29 07:24:02 2024 daemon.err dockerd[6451]: 2024/05/29 07:24:02 http: superfluous response.WriteHeader call from go.opentelemetry.io/contrib/instrumentation/net/http/otelhttp.(*respWriterWrapper).WriteHeader (wrap.go:98)
Wed May 29 07:25:03 2024 daemon.err dockerd[6451]: 2024/05/29 07:25:03 http: superfluous response.WriteHeader call from go.opentelemetry.io/contrib/instrumentation/net/http/otelhttp.(*respWriterWrapper).WriteHeader (wrap.go:98)
Wed May 29 07:26:03 2024 daemon.err dockerd[6451]: 2024/05/29 07:26:03 http: superfluous response.WriteHeader call from go.opentelemetry.io/contrib/instrumentation/net/http/otelhttp.(*respWriterWrapper).WriteHeader (wrap.go:98)
Wed May 29 07:27:03 2024 daemon.err dockerd[6451]: 2024/05/29 07:27:03 http: superfluous response.WriteHeader call from go.opentelemetry.io/contrib/instrumentation/net/http/otelhttp.(*respWriterWrapper).WriteHeader (wrap.go:98)
Wed May 29 07:28:03 2024 daemon.err dockerd[6451]: 2024/05/29 07:28:03 http: superfluous response.WriteHeader call from go.opentelemetry.io/contrib/instrumentation/net/http/otelhttp.(*respWriterWrapper).WriteHeader (wrap.go:98)
Wed May 29 07:29:02 2024 daemon.err dockerd[6451]: 2024/05/29 07:29:02 http: superfluous response.WriteHeader call from go.opentelemetry.io/contrib/instrumentation/net/http/otelhttp.(*respWriterWrapper).WriteHeader (wrap.go:98)
Wed May 29 07:30:03 2024 daemon.err dockerd[6451]: 2024/05/29 07:30:03 http: superfluous response.WriteHeader call from go.opentelemetry.io/contrib/instrumentation/net/http/otelhttp.(*respWriterWrapper).WriteHeader (wrap.go:98)
Wed May 29 07:31:03 2024 daemon.err dockerd[6451]: 2024/05/29 07:31:03 http: superfluous response.WriteHeader call from go.opentelemetry.io/contrib/instrumentation/net/http/otelhttp.(*respWriterWrapper).WriteHeader (wrap.go:98)
Wed May 29 07:32:03 2024 daemon.err dockerd[6451]: 2024/05/29 07:32:03 http: superfluous response.WriteHeader call from go.opentelemetry.io/contrib/instrumentation/net/http/otelhttp.(*respWriterWrapper).WriteHeader (wrap.go:98)
Wed May 29 07:33:03 2024 daemon.err dockerd[6451]: 2024/05/29 07:33:03 http: superfluous response.WriteHeader call from go.opentelemetry.io/contrib/instrumentation/net/http/otelhttp.(*respWriterWrapper).WriteHeader (wrap.go:98)
Wed May 29 07:34:02 2024 daemon.err dockerd[6451]: 2024/05/29 07:34:02 http: superfluous response.WriteHeader call from go.opentelemetry.io/contrib/instrumentation/net/http/otelhttp.(*respWriterWrapper).WriteHeader (wrap.go:98)
Wed May 29 07:35:03 2024 daemon.err dockerd[6451]: 2024/05/29 07:35:03 http: superfluous response.WriteHeader call from go.opentelemetry.io/contrib/instrumentation/net/http/otelhttp.(*respWriterWrapper).WriteHeader (wrap.go:98)
Wed May 29 07:36:03 2024 daemon.err dockerd[6451]: 2024/05/29 07:36:03 http: superfluous response.WriteHeader call from go.opentelemetry.io/contrib/instrumentation/net/http/otelhttp.(*respWriterWrapper).WriteHeader (wrap.go:98)
[... cut ...]
Wed May 29 07:59:02 2024 daemon.err dockerd[6451]: 2024/05/29 07:59:02 http: superfluous response.WriteHeader call from go.opentelemetry.io/contrib/instrumentation/net/http/otelhttp.(*respWriterWrapper).WriteHeader (wrap.go:98)
Wed May 29 08:00:02 2024 daemon.err dockerd[6451]: 2024/05/29 08:00:02 http: superfluous response.WriteHeader call from go.opentelemetry.io/contrib/instrumentation/net/http/otelhttp.(*respWriterWrapper).WriteHeader (wrap.go:98)

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

armvirt_64_DEVICE_generic

### 详细日志

编译正常

yanxin152133, 2024-07-12:
https://github.com/coolsnowwolf/packages/pull/807

coolsnowwolf, 2026-09-05:
已核对。该日志对应 Docker 26.1.1 的已知 OpenTelemetry otelhttp “superfluous response.WriteHeader” 警告，关联的 packages PR #807 已于 2024-05-08 合并。当前 packages feed 中 docker 与 dockerd 已更新到 29.6.1，已包含后续上游修复，旧版本问题不再适用于当前仓库，因此关闭为已完成。

closed 2026-09-05 by coolsnowwolf