coolsnowwolf/lede #12202: menuconfig前3个选项不能进入子菜单

S1U, 2024-05-30:
### 详细叙述

如题，不知道为什么前3个按了`Enter`都不能进入下级菜单选项，下面其他的有下级菜单的可以`Enter`进入。尝试删除`rm -rf .config`还是不行，删除原目录重新`git clone`后，问题依旧。

为什么前3个不能更改？
[image]

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

任意

### 详细日志

没到这一步

WYC-2020, 2024-05-30:
奇葩问题千千万哈哈

S1U, 2024-06-06:
虽然奇葩，但确是问题

WYC-2020, 2024-06-06:
> 虽然奇葩，但确是问题

可能还是和你本地环境有关 依赖那些都安装齐全了么 一行行复制 试试，一直没遇到过这个问题 我想其他人应该也没遇到过吧，还有你是在什么系统上面搞的 ubuntu？

S1U, 2024-06-06:
> > 虽然奇葩，但确是问题
> 
> 可能还是和你本地环境有关 依赖那些都安装齐全了么 一行行复制 试试，一直没遇到过这个问题 我想其他人应该也没遇到过吧，还有你是在什么系统上面搞的 ubuntu？

在Win11上装的ubuntu，之前用一直好好的，从没出现任何问题。

S1U, 2024-06-13:
WSL 卸载Ubuntu，重装 Ubuntu后，问题依旧。前3个不能进入子菜单，是锁定了吗？还是说现在的版本不支持更改了？

bob-zebedy, 2024-06-13:
https://github.com/coolsnowwolf/lede/assets/53208952/efd54998-c9a8-4dac-97d7-59de37f54568

刚重开的虚拟机安装的依赖 clone 的代码，没问题

S1U, 2024-06-13:
> iTerm2-Bob-20240613151448-Zt42sL6b.mp4 
> 刚重开的虚拟机安装的依赖 clone 的代码，没问题

那奇了怪了。。。

S1U, 2024-06-19:
卸载WSL后再重新来了一遍，无意中最大化了终端窗口，居然、居然……进了！我了个法克！居然是因为窗口大小问题造成不能进入前3个菜单，🤣

closed 2024-06-19 by S1U