coolsnowwolf/lede #12204: x86编译时 inotify-tools 报错

qzldd, 2024-05-31:
### 详细叙述

make[8]: Leaving directory '/workdir/openwrt/build_dir/target-x86_64_musl/inotify-tools-3.20.11.0/libinotifytools/src/inotifytools'
make[8]: Entering directory '/workdir/openwrt/build_dir/target-x86_64_musl/inotify-tools-3.20.11.0/libinotifytools/src'
/bin/bash ../../libtool  --tag=CC   --mode=compile x86_64-openwrt-linux-musl-gcc -DHAVE_CONFIG_H -I. -I../.. -I../../libinotifytools/src/inotifytools   -I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/include -I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include/fortify -I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include  -std=c99 -Wall -Wshadow -Werror -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/workdir/openwrt/build_dir/target-x86_64_musl/inotify-tools-3.20.11.0=inotify-tools-3.20.11.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -MT inotifytools.lo -MD -MP -MF .deps/inotifytools.Tpo -c -o inotifytools.lo inotifytools.c
OpenWrt-libtool: compile:  x86_64-openwrt-linux-musl-gcc -DHAVE_CONFIG_H -I. -I../.. -I../../libinotifytools/src/inotifytools -I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/include -I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include/fortify -I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include -std=c99 -Wall -Wshadow -Werror -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/workdir/openwrt/build_dir/target-x86_64_musl/inotify-tools-3.20.11.0=inotify-tools-3.20.11.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -MT inotifytools.lo -MD -MP -MF .deps/inotifytools.Tpo -c inotifytools.c  -fPIC -DPIC -o .libs/inotifytools.o
inotifytools.c: In function 'inotifytools_watch_recursively_with_exclude':
inotifytools.c:1335:30: error: storage size of 'my_stat' isn't known
 1335 |         static struct stat64 my_stat;
      |                              ^~~~~~~
inotifytools.c:1342:36: error: implicit declaration of function 'lstat64'; did you mean 'lstat'? [-Werror=implicit-function-declaration]
 1342 |                         if ( -1 == lstat64( next_file, &my_stat ) ) {
      |                                    ^~~~~~~
      |                                    lstat
inotifytools.c:1335:30: error: unused variable 'my_stat' [-Werror=unused-variable]
 1335 |         static struct stat64 my_stat;
      |                              ^~~~~~~
inotifytools.c: In function 'isdir':
inotifytools.c:1621:30: error: storage size of 'my_stat' isn't known
 1621 |         static struct stat64 my_stat;
      |                              ^~~~~~~
inotifytools.c:1621:30: error: unused variable 'my_stat' [-Werror=unused-variable]
inotifytools.c:1630:1: error: control reaches end of non-void function [-Werror=return-type]
 1630 | }
      | ^
cc1: all warnings being treated as errors
make[8]: *** [Makefile:685: inotifytools.lo] Error 1
make[8]: Leaving directory '/workdir/openwrt/build_dir/target-x86_64_musl/inotify-tools-3.20.11.0/libinotifytools/src'
make[7]: *** [Makefile:749: all-recursive] Error 1
make[7]: Leaving directory '/workdir/openwrt/build_dir/target-x86_64_musl/inotify-tools-3.20.11.0/libinotifytools/src'
make[6]: *** [Makefile:351: all-recursive] Error 1
make[6]: Leaving directory '/workdir/openwrt/build_dir/target-x86_64_musl/inotify-tools-3.20.11.0/libinotifytools'
make[5]: *** [Makefile:404: all-recursive] Error 1
make[5]: Leaving directory '/workdir/openwrt/build_dir/target-x86_64_musl/inotify-tools-3.20.11.0'
make[4]: *** [Makefile:336: all] Error 2
make[4]: Leaving directory '/workdir/openwrt/build_dir/target-x86_64_musl/inotify-tools-3.20.11.0'
make[3]: *** [Makefile:106: /workdir/openwrt/build_dir/target-x86_64_musl/inotify-tools-3.20.11.0/.built] Error 2
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/utils/inotify-tools'
time: package/feeds/packages/inotify-tools/compile#0.17#0.08#0.23
    ERROR: package/feeds/packages/inotify-tools failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/inotify-tools/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2
Error: Process completed with exit code 2.

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

X86

### 详细日志

这个错误好象已经修复，大雕能否费心下，引进修复？https://github.com/openwrt/packages/pull/23549[https://github.com/openwrt/packages/pull/23549](url)

qzldd, 2024-05-31:
https://github.com/openwrt/packages/pull/23549

WYC-2020, 2024-05-31:
搜索关键字 stat64  就能找到解决方案的 这个老问题了 遇到先搜搜说不定有意外的惊喜

WYC-2020, 2024-05-31:
[image]
如此修复

qzldd, 2024-05-31:
已成功编译，谢谢！

closed 2024-05-31 by qzldd