coolsnowwolf/lede #12208: 编译出错

Kuron88, 2024-06-01:
### 详细叙述

编译出错

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

CR660X

### 详细日志

mkdir -p /home/kuron/lede/staging_dir/hostpkg/stamp
touch /home/kuron/lede/build_dir/target-mipsel_24kc_musl/host/rustc-1.78.0-src/.built
touch /home/kuron/lede/staging_dir/hostpkg/stamp/.rust_installed
make[3]: Leaving directory '/home/kuron/lede/feeds/packages/lang/rust'
time: package/feeds/packages/rust/host-compile#4201.75#204.55#1437.77
make[2]: Leaving directory '/home/kuron/lede'
make[1]: *** [package/Makefile:110: /home/kuron/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/kuron/lede'
make: *** [/home/kuron/lede/include/toplevel.mk:231：world] 错误 2

Umomboom, 2024-06-02:
神仙看了都得摇头

ganan3917, 2024-06-02:
拉取今天的代码编译同样的错误

lxg2001, 2024-06-02:
这个问题：“Patch failed! Please fix ./patches/030-meson-don-t-use-run.patch!”

WYC-2020, 2024-06-03:
瞎搞，自己乱升级，patch又不修复，别人l这边 明显就是2.38.1 你这2.39.3如何来的？确定是这边的仓库，不是的话 去你拉的仓库 哪里反馈

Kuron88, 2024-06-03:
[image]
这不是2.39.3吗

closed 2024-06-03 by Kuron88

WYC-2020, 2024-06-03:
> ![image](https://private-user-images.githubusercontent.com/29248355/335936101-9df2460f-b1f0-4b82-bfc7-2029f60f4136.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

那也不对啊 别人那个位置的 根本就没有patch  需要打？只有
[image]
这个位置的 才需要打pa'tch 并且这里的版本 根本就不是2.39.3 你确定不是魔改？