coolsnowwolf/lede #12210: 编译错误

dd-bpir3, 2024-06-03:
### 详细叙述

ERROR: toolchain/gcc/initial failed to build.

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

bpi-r3

### 详细日志

make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [/home/yph/lede/include/toplevel.mk:231: world] Error 1
make[2]: Entering directory '/home/yph/lede/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/home/yph/lede/scripts/config'
make[1]: Entering directory '/home/yph/lede'
make[2]: Entering directory '/home/yph/lede'
+ mkdir -p /home/yph/lede/staging_dir/target-aarch64_cortex-a53_musl
+ cd /home/yph/lede/staging_dir/target-aarch64_cortex-a53_musl
+ mkdir -p bin lib stamp usr/include usr/lib
mkdir -p /home/yph/lede/build_dir/target-aarch64_cortex-a53_musl/stamp
touch /home/yph/lede/staging_dir/target-aarch64_cortex-a53_musl/.prepared
make[3]: Entering directory '/home/yph/lede/toolchain/gdb'
make[3]: Nothing to be done for 'compile'.
make[3]: Leaving directory '/home/yph/lede/toolchain/gdb'
time: toolchain/gdb/compile#0.01#0.00#0.02
make[3]: Entering directory '/home/yph/lede/toolchain/binutils'
make[3]: Nothing to be done for 'compile'.
make[3]: Leaving directory '/home/yph/lede/toolchain/binutils'
time: toolchain/binutils/compile#0.00#0.01#0.02
make[3]: Entering directory '/home/yph/lede/toolchain/gcc/initial'
export SHELL="bash"; make CFLAGS="-O2 -I/home/yph/lede/staging_dir/host/include " CFLAGS_FOR_TARGET="-Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CXXFLAGS_FOR_TARGET="-Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" GOCFLAGS_FOR_TARGET="-Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro"  -C /home/yph/lede/build_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/gcc-11.3.0-initial all-build-libiberty all-gcc all-target-libgcc
make[4]: Entering directory '/home/yph/lede/build_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/gcc-11.3.0-initial'
make[5]: Entering directory '/home/yph/lede/build_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/gcc-11.3.0-initial/build-x86_64-pc-linux-gnu/libiberty'
make[6]: Entering directory '/home/yph/lede/build_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/gcc-11.3.0-initial/build-x86_64-pc-linux-gnu/libiberty/testsuite'
make[6]: Nothing to be done for 'all'.
make[6]: Leaving directory '/home/yph/lede/build_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/gcc-11.3.0-initial/build-x86_64-pc-linux-gnu/libiberty/testsuite'
make[5]: Leaving directory '/home/yph/lede/build_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/gcc-11.3.0-initial/build-x86_64-pc-linux-gnu/libiberty'
make[5]: Entering directory '/home/yph/lede/build_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/gcc-11.3.0-initial/libiberty'
make[6]: Entering directory '/home/yph/lede/build_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/gcc-11.3.0-initial/libiberty/testsuite'
make[6]: Nothing to be done for 'all'.
make[6]: Leaving directory '/home/yph/lede/build_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/gcc-11.3.0-initial/libiberty/testsuite'
make[5]: Leaving directory '/home/yph/lede/build_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/gcc-11.3.0-initial/libiberty'
make[5]: Entering directory '/home/yph/lede/build_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/gcc-11.3.0-initial/lto-plugin'
make  all-am
make[6]: Entering directory '/home/yph/lede/build_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/gcc-11.3.0-initial/lto-plugin'
make[6]: Leaving directory '/home/yph/lede/build_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/gcc-11.3.0-initial/lto-plugin'
make[5]: Leaving directory '/home/yph/lede/build_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/gcc-11.3.0-initial/lto-plugin'
make[5]: Entering directory '/home/yph/lede/build_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/gcc-11.3.0-initial/intl'
make[5]: Nothing to be done for 'all'.
make[5]: Leaving directory '/home/yph/lede/build_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/gcc-11.3.0-initial/intl'
make[5]: Entering directory '/home/yph/lede/build_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/gcc-11.3.0-initial/build-x86_64-pc-linux-gnu/fixincludes'
make[5]: Nothing to be done for 'all'.
make[5]: Leaving directory '/home/yph/lede/build_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/gcc-11.3.0-initial/build-x86_64-pc-linux-gnu/fixincludes'
make[5]: Entering directory '/home/yph/lede/build_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/gcc-11.3.0-initial/build-x86_64-pc-linux-gnu/libcpp'
test -f config.h || (rm -f stamp-h1 && make stamp-h1)
make[5]: Leaving directory '/home/yph/lede/build_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/gcc-11.3.0-initial/build-x86_64-pc-linux-gnu/libcpp'
make[5]: Entering directory '/home/yph/lede/build_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/gcc-11.3.0-initial/zlib'
true "AR_FLAGS=rc" "CC_FOR_BUILD=gcc" "CFLAGS=-O2 -I/home/yph/lede/staging_dir/host/include " "CXXFLAGS=-g -O2" "CFLAGS_FOR_BUILD=" "CFLAGS_FOR_TARGET=-Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" "INSTALL=/home/yph/lede/staging_dir/host/bin/install -c" "INSTALL_DATA=/home/yph/lede/staging_dir/host/bin/install -c -m 644" "INSTALL_PROGRAM=/home/yph/lede/staging_dir/host/bin/install -c" "INSTALL_SCRIPT=/home/yph/lede/staging_dir/host/bin/install -c" "LDFLAGS=-static-libstdc++ -static-libgcc " "LIBCFLAGS=-O2 -I/home/yph/lede/staging_dir/host/include " "LIBCFLAGS_FOR_TARGET=-Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" "MAKE=make" "MAKEINFO=/home/yph/lede/build_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/gcc-11.3.0/missing makeinfo --split-size=5000000 " "PICFLAG=" "PICFLAG_FOR_TARGET=" "SHELL=/bin/bash" "EXPECT=expect" "RUNTEST=runtest" "RUNTESTFLAGS=" "exec_prefix=/home/yph/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl" "infodir=/home/yph/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/share/info" "libdir=/home/yph/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/lib" "prefix=/home/yph/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl" "tooldir=/home/yph/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/aarch64-openwrt-linux-musl" "AR=ar" "AS=as" "CC=gcc" "CXX=g++" "LD=ld" "LIBCFLAGS=-O2 -I/home/yph/lede/staging_dir/host/include " "NM=nm" "PICFLAG=" "RANLIB=ranlib" "DESTDIR=" DO=all multi-do # make
make[5]: Leaving directory '/home/yph/lede/build_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/gcc-11.3.0-initial/zlib'
make[5]: Entering directory '/home/yph/lede/build_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/gcc-11.3.0-initial/libbacktrace'
make  all-am
make[6]: Entering directory '/home/yph/lede/build_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/gcc-11.3.0-initial/libbacktrace'
true  DO=all multi-do # make
make[6]: Leaving directory '/home/yph/lede/build_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/gcc-11.3.0-initial/libbacktrace'
make[5]: Leaving directory '/home/yph/lede/build_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/gcc-11.3.0-initial/libbacktrace'
[... 10 lines cut ...]
make[5]: Nothing to be done for 'all'.
make[5]: Leaving directory '/home/yph/lede/build_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/gcc-11.3.0-initial/fixincludes'
make[5]: Entering directory '/home/yph/lede/build_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/gcc-11.3.0-initial/gcc'
g++ -no-pie   -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -static-libstdc++ -static-libgcc  -o cc1 c/c-lang.o c-family/stub-objc.o attribs.o c/c-errors.o c/c-decl.o c/c-typeck.o c/c-convert.o c/c-aux-info.o c/c-objc-common.o c/c-parser.o c/c-fold.o c/gimple-parser.o c-family/c-common.o c-family/c-cppbuiltin.o c-family/c-dump.o c-family/c-format.o c-family/c-gimplif [... cut ...]
make[3]: Leaving directory '/home/yph/lede/toolchain/gcc/initial'
time: toolchain/gcc/initial/compile#13.54#3.33#16.80
    ERROR: toolchain/gcc/initial failed to build.
make[2]: *** [toolchain/Makefile:97: toolchain/gcc/initial/compile] Error 1
make[2]: Leaving directory '/home/yph/lede'
make[1]: *** [toolchain/Makefile:93: /home/yph/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/stamp/.toolchain_compile] Error 2
make[1]: Leaving directory '/home/yph/lede'
make: *** [/home/yph/lede/include/toplevel.mk:231: world] Error 2

WYC-2020, 2024-06-04:
万能大法试试，看你这错误 好像是pthread 没找到，但是不应该会有这样的错误的 ，确定编译环境那些依赖是否装完整了，然后 make distclean  在重新编译试试

xseven121, 2024-06-04:
删除/home/yph/lede/staging_dir/host/lib/libzstd.a

dd-bpir3, 2024-06-09:
解决问题swap分区太小了调大解决问题

closed 2024-07-11 by aiamadeus