coolsnowwolf/lede #12215: 云编译hostapd编译报错

jumpdang, 2024-06-06:
### 详细叙述

hostapd编译报错，搜了类似issue,报错代码好像不一样，有没有大佬看看是怎么回事

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86-64

### 详细日志

2024-06-02T22:48:21.3445291Z Applying ./patches/993-fix-mbo-module-build.patch using plaintext: 
2024-06-02T22:48:21.3455410Z patching file wpa_supplicant/Makefile
2024-06-02T22:48:21.3456723Z Hunk #1 FAILED at 339.
2024-06-02T22:48:21.3461214Z 1 out of 1 hunk FAILED -- saving rejects to file wpa_supplicant/Makefile.rej
2024-06-02T22:48:21.3463716Z Patch failed!  Please fix ./patches/993-fix-mbo-module-build.patch!
2024-06-02T22:48:21.3468938Z make[3]: *** [Makefile:794: /home/runner/work/OpenWrt/OpenWrt/openwrt/build_dir/target-x86_64_musl/hostapd-wpad-full-openssl/hostapd-2022-07-29-b704dc72/.prepared_fa7aafe74add70b1266d32b42dcf5b9c_6664517399ebbbc92a37c5bb081b5c53] Error 1
2024-06-02T22:48:21.3470876Z make[3]: Leaving directory '/home/runner/work/OpenWrt/OpenWrt/openwrt/package/network/services/hostapd'
2024-06-02T22:48:21.3473043Z time: package/network/services/hostapd/wpad-full-openssl/compile#0.59#0.38#0.81
2024-06-02T22:48:21.3480313Z     ERROR: package/network/services/hostapd failed to build (build variant: wpad-full-openssl).
2024-06-02T22:48:21.3483094Z make[2]: *** [package/Makefile:116: package/network/services/hostapd/compile] Error 1
2024-06-02T22:48:21.3487785Z make[2]: Leaving directory '/home/runner/work/OpenWrt/OpenWrt/openwrt'
2024-06-02T22:48:21.3492919Z make[1]: *** [package/Makefile:110: /home/runner/work/OpenWrt/OpenWrt/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
2024-06-02T22:48:21.3498386Z make[1]: Leaving directory '/home/runner/work/OpenWrt/OpenWrt/openwrt'
2024-06-02T22:48:21.3509547Z make: *** [/home/runner/work/OpenWrt/OpenWrt/openwrt/include/toplevel.mk:231: world] Error 2
2024-06-02T22:48:21.3590484Z ##[error]Process completed with exit code 2.
[0_Build.txt](https://github.com/user-attachments/files/15598216/0_Build.txt)

WYC-2020, 2024-06-06:
自己对比下package/network/services/hostapd/patches 这下面的patch 是否有多的 吧多的删除了，怀疑你是不是放了除l以外其他的补丁，导致冲突了

yasuis, 2024-06-21:
可以把diy-script.sh里# 修复 hostapd 报错改为
cp -f $GITHUB_WORKSPACE/script/011-fix-mbo-modules-build.patch package/network/services/hostapd/patches/011-fix-mbo-modules-build.patch

closed 2024-07-10 by coolsnowwolf