coolsnowwolf/lede #12221: 关于RAX3000M “交换机”问题

lgwake, 2024-06-07:
### 详细叙述

用代码可以无误编译出固件，但是发现没有“交换机”选项，是需要自己继续编写不？编译的时候已经勾选了base system里面的swconfig,config.buildinfo如下

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

rax3000m-nand

### 详细日志

CONFIG_TARGET_mediatek=y
CONFIG_TARGET_mediatek_filogic=y
CONFIG_TARGET_mediatek_filogic_DEVICE_cmcc_rax3000m-nand=y
CONFIG_GNUTLS_ALPN=y
CONFIG_GNUTLS_ANON=y
CONFIG_GNUTLS_DTLS_SRTP=y
CONFIG_GNUTLS_HEARTBEAT=y
CONFIG_GNUTLS_OCSP=y
CONFIG_GNUTLS_PSK=y
CONFIG_PACKAGE_alsa-lib=y
CONFIG_PACKAGE_alsa-ucm-conf=y
CONFIG_PACKAGE_alsa-utils=y
CONFIG_PACKAGE_boost=y
CONFIG_PACKAGE_fdk-aac=y
CONFIG_PACKAGE_ffmpeg=y
CONFIG_PACKAGE_kmod-hid=y
CONFIG_PACKAGE_kmod-hid-generic=y
CONFIG_PACKAGE_kmod-input-core=y
CONFIG_PACKAGE_kmod-input-evdev=y
CONFIG_PACKAGE_kmod-ipt-offload=y
CONFIG_PACKAGE_kmod-nf-flow=y
CONFIG_PACKAGE_kmod-sound-core=y
CONFIG_PACKAGE_kmod-tcp-bbr=y
CONFIG_PACKAGE_kmod-usb-audio=y
CONFIG_PACKAGE_kmod-usb-hid=y
CONFIG_PACKAGE_lame-lib=y
CONFIG_PACKAGE_libatomic=y
CONFIG_PACKAGE_libbz2=y
CONFIG_PACKAGE_libcap=y
CONFIG_PACKAGE_libexpat=y
CONFIG_PACKAGE_libfaad2=y
CONFIG_PACKAGE_libffmpeg-full=y
CONFIG_PACKAGE_libflac=y
CONFIG_PACKAGE_libfmt=y
CONFIG_PACKAGE_libgmp=y
CONFIG_PACKAGE_libgnutls=y
CONFIG_PACKAGE_libid3tag=y
CONFIG_PACKAGE_libiwinfo-lua=y
CONFIG_PACKAGE_libltdl=y
CONFIG_PACKAGE_libmad=y
CONFIG_PACKAGE_libmicrohttpd-no-ssl=y
CONFIG_PACKAGE_libmms=y
CONFIG_PACKAGE_libmpdclient=y
CONFIG_PACKAGE_libnettle=y
CONFIG_PACKAGE_libnpupnp=y
CONFIG_PACKAGE_libogg=y
CONFIG_PACKAGE_libopus=y
CONFIG_PACKAGE_libshout=y
CONFIG_PACKAGE_libsndfile=y
CONFIG_PACKAGE_libsoxr=y
CONFIG_PACKAGE_libstdcpp=y
CONFIG_PACKAGE_libuci-lua=y
CONFIG_PACKAGE_libvorbis=y
CONFIG_PACKAGE_libvorbisidec=y
CONFIG_PACKAGE_luci-app-firewall=y
CONFIG_PACKAGE_luci-i18n-firewall-zh-cn=y
CONFIG_PACKAGE_luci-proto-ppp=y
CONFIG_PACKAGE_luci-theme-bootstrap=y
CONFIG_PACKAGE_mpd-full=y
CONFIG_PACKAGE_pdnsd-alt=y
CONFIG_PACKAGE_pulseaudio-daemon=y
CONFIG_PACKAGE_rpcd-mod-rrdns=y
CONFIG_PACKAGE_swconfig=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
CONFIG_PACKAGE_vlmcsd=y
CONFIG_PACKAGE_vsftpd-alt=y
CONFIG_PACKAGE_wol=y
CONFIG_PACKAGE_yajl=y
CONFIG_VSFTPD_USE_UCI_SCRIPTS=y
CONFIG_boost-compile-visibility-hidden=y
CONFIG_boost-runtime-shared=y
CONFIG_boost-static-and-shared-libs=y
CONFIG_boost-variant-release=y

coolsnowwolf, 2024-06-07:
是 DSA 架构的驱动，自然没有交换机

lgwake, 2024-06-08:
> 是 DSA 架构的驱动，自然没有交换机

请问有什么版本改成有“交换机”选项？是需要重新DTS还是选择其他内核不？

closed 2024-06-20 by coolsnowwolf