coolsnowwolf/lede #12223: busybox中的sha256sum不能正常工作

ololllolo, 2024-06-07:
### 详细叙述

拉取最新源码，未作任何配置，编译后，进入系统使用sha256sum工具，提示illegal instruction。
导致一些使用此工具的插件工作不正常。例如我正在配置passwall，发现无法更新geoip，发现geoip文件已经下载成功了，更新脚本利用sha256sum校验下载后的文件，但sha256sum工具不能正常工作，所以更新失败了。为复现，重新拉取源码，以默认配置编译，发现此工具还是不能正常运行。

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

X86

### 详细日志

运行失败时，日志提示：
Sat Jun  8 02:37:43 2024 kern.info kernel: [   49.706712] traps: busybox[5217] trap invalid opcode ip:406049 sp:7fff439ed4f8 error:0 in busybox[406000+47000]

ololllolo, 2024-06-12:
我已经就这个问题卡住了好几天了，这期间做了一些测试，总体情况是这样的：
编译环境是一台Esxi 7.03上的debian12,编译后的固件在同一台esxi系统上创建的虚拟机上运行sha256sum就提示illegal instruction。将系统用vcenter迁移到另外一台esxi 7.03上，运行sha256sum则正常。将固件刷入到其他的主机上直接运行，也都正常，只有在最初那台esxi上不正常。而这台esxi上的虚拟机上原来跑的是23年11月份编译的版本，这个版本在这台机器上也是正常的，所以我想可能是新的系统上运行sha256sum开启了某些cpu指令集无法让当前这台esxi主机支持吧，这个问题是我无法解决的，只能用回以前的版本了。
具体的运行情况是：
Esxi 7.03 CPU 双路 Xeon Silver 4210R   sha256sum运行不正常
其他测试过都正常的设备有：
Esxi 7.03 CPU Xeon E5-2650 v4 
Esxi 7.03 CPU Xeon  E3-1230 v3
Esxi 8.01 Core i5-1235U
N5105
AMD 5700G
所以只有这一台系统无法运行，因此我想也没什么机会解决它了，就这样吧。

bob-zebedy, 2024-06-13:
测试正常
[image]

closed 2024-06-18 by coolsnowwolf

Portan, 2025-01-21:
@ololllolo 我也出现了一样的问题，你后面解决了吗？