coolsnowwolf/lede #12225: 编译报错

2253845067, 2024-06-09:
### 详细叙述

最近两三天云编译一直报错，之前都好好的，大佬们看看啥情况

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

H69K

### 详细日志

make[5]: Leaving directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.160'
make -C /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.160 KCFLAGS="-fmacro-prefix-map=/workdir/openwrt/build_dir/target-aarch64_generic_musl=target-aarch64_generic_musl -fno-caller-saves " HOSTCFLAGS="-O2 -I/workdir/openwrt/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="aarch64-openwrt-linux-musl-" ARCH="arm64" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Sat Jun  8 14:30:09 2024" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/workdir/openwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRELEASE=5.15.160 CC="aarch64-openwrt-linux-musl-gcc" Image dtbs modules
make[5]: Entering directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.160'
  CALL    scripts/checksyscalls.sh
  CALL    scripts/atomic/check-atomics.sh
  CHK     include/generated/compile.h
  GEN     .version
  CHK     include/generated/compile.h
  LD      vmlinux.o
  MODPOST vmlinux.symvers
  MODINFO modules.builtin.modinfo
  GEN     modules.builtin
  LD      .tmp_vmlinux.kallsyms1
  KSYMS   .tmp_vmlinux.kallsyms1.S
  AS      .tmp_vmlinux.kallsyms1.S
  LD      .tmp_vmlinux.kallsyms2
  KSYMS   .tmp_vmlinux.kallsyms2.S
  AS      .tmp_vmlinux.kallsyms2.S
  LD      vmlinux
  SORTTAB vmlinux
  SYSMAP  System.map
  OBJCOPY arch/arm64/boot/Image
  DTC     arch/arm64/boot/dts/rockchip/rk3568-opc-h66k.dtb
Error: arch/arm64/boot/dts/rockchip/rk3568-hinlink-opc.dtsi:1[81](https://github.com/2253845067/Actions-OpenWrt/actions/runs/9430022828/job/25977078682#step:12:82).1-5 Label or path gpu not found
Error: arch/arm64/boot/dts/rockchip/rk3568-hinlink-opc.dtsi:186.1-6 Label or path hdmi not found
Error: arch/arm64/boot/dts/rockchip/rk3568-hinlink-opc.dtsi:192.1-9 Label or path hdmi_in not found
Error: arch/arm64/boot/dts/rockchip/rk3568-hinlink-opc.dtsi:198.1-10 Label or path hdmi_out not found
Error: arch/arm64/boot/dts/rockchip/rk3568-hinlink-opc.dtsi:204.1-12 Label or path hdmi_sound not found
Error: arch/arm64/boot/dts/rockchip/rk3568-hinlink-opc.dtsi:450.1-10 Label or path i2s0_8ch not found
Error: arch/arm64/boot/dts/rockchip/rk3568-hinlink-opc.dtsi:645.1-5 Label or path vop not found
Error: arch/arm64/boot/dts/rockchip/rk3568-hinlink-opc.dtsi:651.1-9 Label or path vop_mmu not found
Error: arch/arm64/boot/dts/rockchip/rk3568-hinlink-opc.dtsi:655.1-5 Label or path vp0 not found
FATAL ERROR: Syntax error parsing input tree
make[7]: *** [scripts/Makefile.lib:347: arch/arm64/boot/dts/rockchip/rk3568-opc-h66k.dtb] Error 1
make[6]: *** [scripts/Makefile.build:552: arch/arm64/boot/dts/rockchip] Error 2
make[5]: *** [Makefile:1450: dtbs] Error 2
make[5]: Leaving directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.160'
make[4]: *** [Makefile:26: /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.160/.modules] Error 2
make[3]: *** [Makefile:11: compile] Error 2
make[4]: Leaving directory '/workdir/openwrt/target/linux/rockchip'
make[3]: Leaving directory '/workdir/openwrt/target/linux'
time: target/linux/compile#10.27#4.57#14.66
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [target/Makefile:23: /workdir/openwrt/staging_dir/target-aarch64_generic_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2

aiamadeus, 2024-06-09:
换 6.1 内核，才加了 6.6 内核而且后面还要重新开放超频啥的，5.15 没精力维护了。

2253845067, 2024-06-09:
> 换 6.1 内核，才加了 6.6 内核而且后面还要重新开放超频啥的，5.15 没精力维护了。
好的大佬。

closed 2024-06-09 by 2253845067