coolsnowwolf/lede #12229: WWAN quectel_Gobinet linux6.1/6.6内核编译失败

dragon888888888888, 2024-06-11:
### 详细叙述

5.15内核编译成功  6.1/6.6内核 quectel_Gobinet都是一样的问题，6.1内核quectel_SRPD_PCIE编译成功，6.6内核quectel_SRPD_PCIE编译失败
quectel_Gobinet已向移远要了1.6.5的驱动也是一样的问题
重复 issue有，是23年的显示被修复已被关闭

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

光影猫

### 详细日志

make[3]: Entering directory '/home/abc/lede2/lede/package/wwan/driver/quectel_Gobinet'
rm -f /home/abc/lede2/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/gobinet-1.6.3/.built
touch /home/abc/lede2/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/gobinet-1.6.3/.built_check
make -C "/home/abc/lede2/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-6.1.92" ARCH="arm64" CROSS_COMPILE="aarch64-openwrt-linux-musl-" CXXFLAGS="-Os -pipe -mcpu=generic -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/abc/lede2/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/gobinet-1.6.3=gobinet-1.6.3 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" M="/home/abc/lede2/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/gobinet-1.6.3"  modules
make[4]: Entering directory '/home/abc/lede2/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-6.1.92'
  CC [M]  /home/abc/lede2/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/gobinet-1.6.3/GobiUSBNet.o
/home/abc/lede2/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/gobinet-1.6.3/GobiUSBNet.c: In function 'GobiNetDriverBind':
/home/abc/lede2/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/gobinet-1.6.3/GobiUSBNet.c:1543:32: error: assignment of read-only location '*pDev->net->dev_addr'
 1543 |         pDev->net->dev_addr[0] |= 0x02; /* set local assignment bit */
      |                                ^~
/home/abc/lede2/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/gobinet-1.6.3/GobiUSBNet.c:1544:32: error: assignment of read-only location '*pDev->net->dev_addr'
 1544 |         pDev->net->dev_addr[0] |= 0xbf; /* clear "IP" bit */
      |                                ^~
/home/abc/lede2/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/gobinet-1.6.3/GobiUSBNet.c:1546:22: warning: passing argument 1 of 'memcpy' discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
 1546 |     memcpy (pDev->net->dev_addr, node_id, sizeof node_id);
      |             ~~~~~~~~~^~~~~~~~~~
In file included from ./include/linux/string.h:60,
                 from ./include/linux/bitmap.h:11,
                 from ./include/linux/cpumask.h:12,
                 from ./include/linux/smp.h:13,
                 from ./arch/arm64/include/asm/arch_timer.h:18,
                 from ./arch/arm64/include/asm/timex.h:8,
                 from ./include/linux/timex.h:67,
                 from ./include/linux/time32.h:13,
                 from ./include/linux/time.h:60,
                 from ./arch/arm64/include/asm/stat.h:12,
                 from ./include/linux/stat.h:6,
                 from ./include/linux/module.h:13,
                 from /home/abc/lede2/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/gobinet-1.6.3/GobiUSBNet.c:55:
./arch/arm64/include/asm/string.h:35:21: note: expected 'void *' but argument is of type 'const unsigned char *'
   35 | extern void *memcpy(void *, const void *, __kernel_size_t);
      |                     ^~~~~~
/home/abc/lede2/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/gobinet-1.6.3/GobiUSBNet.c: In function 'GobiUSBNetProbe':
/home/abc/lede2/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/gobinet-1.6.3/GobiUSBNet.c:2973:14: warning: passing argument 1 of 'QuecPrintHex' discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
 2973 |    PrintHex( &pGobiDev->mpNetDev->net->dev_addr[0], 6 );
      |              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/abc/lede2/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/gobinet-1.6.3/GobiUSBNet.c:75:
/home/abc/lede2/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/gobinet-1.6.3/QMIDevice.h:325:19: note: expected 'void *' but argument is of type 'const unsigned char *'
  325 |    void *         pBuffer,
      |    ~~~~~~~~~~~~~~~^~~~~~~
make[5]: *** [scripts/Makefile.build:250: /home/abc/lede2/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/gobinet-1.6.3/GobiUSBNet.o] Error 1
make[4]: *** [Makefile:2010: /home/abc/lede2/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/gobinet-1.6.3] Error 2
make[4]: Leaving directory '/home/abc/lede2/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-6.1.92'
make[3]: *** [Makefile:50: /home/abc/lede2/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/gobinet-1.6.3/.built] Error 2
make[3]: Leaving directory '/home/abc/lede2/lede/package/wwan/driver/quectel_Gobinet'
time: package/wwan/driver/quectel_Gobinet/compile#1.12#0.47#1.59
    ERROR: package/wwan/driver/quectel_Gobinet failed to build.
make[2]: *** [package/Makefile:116: package/wwan/driver/quectel_Gobinet/compile] Error 1
make[2]: Leaving directory '/home/abc/lede2/lede'
make[1]: *** [package/Makefile:110: /home/abc/lede2/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/abc/lede2/lede'
make: *** [/home/abc/lede2/lede/include/toplevel.mk:231：world] 错误 2
abc@abc:~/lede2/lede$

WYC-2020, 2024-06-11:
[driver.tar.gz](https://github.com/user-attachments/files/15783005/driver.tar.gz)
[5.10-872-export-some-functions-of-the-sched-module.patch](https://github.com/user-attachments/files/15783006/5.10-872-export-some-functions-of-the-sched-module.patch)
[6.6-872-export-some-functions-of-the-sched-module.patch](https://github.com/user-attachments/files/15783007/6.6-872-export-some-functions-of-the-sched-module.patch)
[6.1-872-export-some-functions-of-the-sched-module.patch](https://github.com/user-attachments/files/15783008/6.1-872-export-some-functions-of-the-sched-module.patch)
[5.15-872-export-some-functions-of-the-sched-module.patch](https://github.com/user-attachments/files/15783009/5.15-872-export-some-functions-of-the-sched-module.patch)
用这个替换原来的 ，然后patch放到lede/target/linux/generic/backport-x.x 下面，patch前面的就是对应的linux版本，自己按需放

dragon888888888888, 2024-06-11:
> lede/target/linux/generic/backport-x.x 下面

感谢，用这个解决了

closed 2024-06-11 by dragon888888888888