coolsnowwolf/lede #12231: 系统日志被daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns刷屏

dragon888888888888, 2024-06-11:
### 详细叙述

最早搜到这个问题是2019年的，一直没有解决办法。 5.15虽然也有这个问题，用原版openwrt  package/network/services/umdns/files下的umdns.json和umdns.init替换掉LEDE版里的2个文件，编译后解决。但6.6内核这样做没有作用，系统日志一直被这个错误刷屏。

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

光影猫

### 详细日志

已编译成功

dragon888888888888, 2024-06-11:
按上面的方法重编译又线刷了一次，虽然不再报错。不过仍被daemon.notice hostapd: wlan0: BEACON-REQ-TX-STATUS aa:aa:aa:aa:aa:aa 6 ack=1 daemon.notice hostapd: wlan0: BEACON-RESP-RX aa:aa:aa:aa:aa:aa 1aa 0000000000000000000000008000000000000000000000000000
daemon.notice hostapd: wlan0: BEACON-REQ-TX-STATUS aa:aa:aa:aa:aa:aa 6 ack=1 daemon.notice hostapd: wlan0: BEACON-RESP-RX aa:aa:aa:aa:aa:aa 2aa 0000000000000000000000008000000000000000000000000000
daemon.notice hostapd: wlan0: BEACON-REQ-TX-STATUS aa:aa:aa:aa:aa:aa 6 ack=1 daemon.notice hostapd: wlan0: BEACON-RESP-RX aa:aa:aa:aa:aa:aa 3aa 0000000000000000000000008000000000000000000000000000
daemon.notice hostapd: wlan0: BEACON-REQ-TX-STATUS aa:aa:aa:aa:aa:aa 6 ack=1 daemon.notice hostapd: wlan0: BEACON-RESP-RX aa:aa:aa:aa:aa:aa 4aa 0000000000000000000000008000000000000000000000000000
daemon.notice hostapd: wlan0: BEACON-REQ-TX-STATUS aa:aa:aa:aa:aa:aa 6 ack=1 daemon.notice hostapd: wlan0: BEACON-RESP-RX aa:aa:aa:aa:aa:aa 5aa 0000000000000000000000008000000000000000000000000000
这种刷屏

yanxin152133, 2024-06-12:
是不是这个原因
[image]

dragon888888888888, 2024-06-12:
不是，离路由器2米左右，终端没有掉线

yanxin152133, 2024-06-12:
aa:aa:aa:aa:aa:aa 6 ack=1   这个是蓝牙？我看别人的都是其他的
[image]

dragon888888888888, 2024-06-12:
> aa:aa:aa:aa:aa:aa 6 ack=1 这个是蓝牙？我看别人的都是其他的 [...]

有蓝牙，但是根本就没有选蓝牙驱动

dragon888888888888, 2024-06-13:
找到方法了，把DAWN删除就没问题了，不知道这个有什知用，编译时是必选的

closed 2024-06-14 by dragon888888888888

xuexijiaoben, 2024-06-15:
谢谢啦，1385这个我这儿也一直有，留言学习一下，