coolsnowwolf/lede #12232: 编译错误

aaronchin, 2024-06-14:
### 详细叙述

尝试更换过gcc 12.2 与 binutils版本 无效！使用全新拉取 编译 依旧出现如下错误 

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

X86

### 详细日志

make[3]: Entering directory '/home/aaronchin/lede/toolchain/gcc/initial'
export SHELL="bash"; make CFLAGS="-O2 -I/home/aaronchin/lede/staging_dir/host/include " CFLAGS_FOR_TARGET="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CXXFLAGS_FOR_TARGET="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" GOCFLAGS_FOR_TARGET="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro"  -C /home/aaronchin/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-initial all-build-libiberty all-gcc all-target-libgcc
make[4]: Entering directory '/home/aaronchin/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-initial'
make[5]: Entering directory '/home/aaronchin/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-initial/build-x86_64-pc-linux-gnu/libiberty'
if [ x"" != x ]; then \
  gcc -c -DHAVE_CONFIG_H   -I. -I/home/aaronchin/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0/libiberty/../include  -W -Wall -Wwrite-strings -Wc++-compat -Wstrict-prototypes -Wshadow=local -pedantic  -D_GNU_SOURCE -fcf-protection  /home/aaronchin/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0/libiberty/simple-object-mach-o.c -o pic/simple-object-mach-o.o; \
else true; fi
if [ x"" != x ]; then \
  gcc -c -DHAVE_CONFIG_H   -I. -I/home/aaronchin/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0/libiberty/../include  -W -Wall -Wwrite-strings -Wc++-compat -Wstrict-prototypes -Wshadow=local -pedantic  -D_GNU_SOURCE -fcf-protection   /home/aaronchin/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0/libiberty/simple-object-mach-o.c -o noasan/simple-object-mach-o.o; \
else true; fi
gcc -c -DHAVE_CONFIG_H   -I. -I/home/aaronchin/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0/libiberty/../include  -W -Wall -Wwrite-strings -Wc++-compat -Wstrict-prototypes -Wshadow=local -pedantic  -D_GNU_SOURCE -fcf-protection /home/aaronchin/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0/libiberty/simple-object-mach-o.c -o simple-object-mach-o.o
/home/aaronchin/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0/libiberty/simple-object-mach-o.c: In function 'simple_object_mach_o_write_segment':
/home/aaronchin/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0/libiberty/simple-object-mach-o.c:1231:17: error: passing argument 1 of 'set_32' from incompatible pointer type [-Wincompatible-pointer-types]
 1231 |         set_32 (&index[i], index[i]);
      |                 ^~~~~~~~~
      |                 |
      |                 unsigned int *
/home/aaronchin/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0/libiberty/simple-object-mach-o.c:1231:17: note: expected 'unsigned char *' but argument is of type 'unsigned int *'
make[5]: *** [Makefile:1308: simple-object-mach-o.o] Error 1
make[5]: Leaving directory '/home/aaronchin/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-initial/build-x86_64-pc-linux-gnu/libiberty'
make[4]: *** [Makefile:2756: all-build-libiberty] Error 2
make[4]: Leaving directory '/home/aaronchin/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-initial'
make[3]: *** [Makefile:37: /home/aaronchin/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0-initial/.built] Error 2
make[3]: Leaving directory '/home/aaronchin/lede/toolchain/gcc/initial'
time: toolchain/gcc/initial/compile#0.17#0.05#0.22
    ERROR: toolchain/gcc/initial failed to build.
make[2]: *** [toolchain/Makefile:97: toolchain/gcc/initial/compile] Error 1
make[2]: Leaving directory '/home/aaronchin/lede'
make[1]: *** [toolchain/Makefile:93: /home/aaronchin/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/stamp/.toolchain_compile] Error 2
make[1]: Leaving directory '/home/aaronchin/lede'
make: *** [/home/aaronchin/lede/include/toplevel.mk:231：world] 錯誤 2

yanxin152133, 2024-06-14:
内存多少？
https://github.com/openwrt/openwrt/issues/12217

aaronchin, 2024-06-14:
> 内存多少？ [openwrt/openwrt#12217](https://github.com/openwrt/openwrt/issues/12217)
16G

yanxin152133, 2024-06-14:
> > 内存多少？ [openwrt/openwrt#12217](https://github.com/openwrt/openwrt/issues/12217)
> > 16G

编译的机器还是你电脑的物理内存啊

aaronchin, 2024-06-14:
> > > 内存多少？ [openwrt/openwrt#12217](https://github.com/openwrt/openwrt/issues/12217)
> > > 16G
> 
> 编译的机器还是你电脑的物理内存啊

编译的机器

yanxin152133, 2024-06-14:
> > > > 内存多少？ [openwrt/openwrt#12217](https://github.com/openwrt/openwrt/issues/12217)
> > > > 16G
> > 
> > 
> > 编译的机器还是你电脑的物理内存啊
> 
> 编译的机器

那就万能大法重新编译试下吧

WYC-2020, 2024-06-14:
你这个明显就是一个类型转换报错嘛 我这边没遇到过 可以加入 Wno-error=incompatible-pointer-types进行忽略应该就能过了

aaronchin, 2024-06-16:
> 你这个明显就是一个类型转换报错嘛 我这边没遇到过 可以加入 Wno-error=incompatible-pointer-types进行忽略应该就能过了

GCC 换回 13 结案

closed 2024-06-16 by aaronchin