coolsnowwolf/lede #12233: php升级到8.2.20后可道云报错了。

Fly432, 2024-06-14:
### 详细叙述

L大源码中PHP版本一直是8.2.2版本，6月6日php官网将php8.2版本升级到8.2.20，修复了 Windows 上的缓冲区限制，用 _read 替换了读取调用使用；修复了 GHSA-3qgc-jrrr-25jv 错误（绕过 CVE-2012-1823，PHP-CGI 中的参数注入）。（CVE-2024-4577）。所以试着将L大源码中PHP版本升级到8.2.20，编译过程很顺利，但是打开可道云时却报了一个错”须开启php GD库，否则验证码、缩略图使用将不正常“如图，我看了一下php8-mod-gd是有安装的。当我把php换回8.2.2时，可道云又正常了，不知什么原因，请大佬们不吝指教。
[image]

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

无

WYC-2020, 2024-06-17:
[image]
亲测有效
[1020_workaround-external-libgd-feature-detection.patch](https://github.com/user-attachments/files/15862374/1020_workaround-external-libgd-feature-detection.patch)
放到php8下面去lede/feeds/packages/lang/php8/patches

Fly432, 2024-06-17:
问题解决了，谢谢大佬!