coolsnowwolf/lede #12237: nginx 问题 (Segmentation Fault)

cbkhoo1492006, 2024-06-15:
### 详细叙述

Model: p&w r619ac
Firmware version: openwrt (r23.11.11)

Firmware location:
https://pan.wwang.pw/%F0%9F%94%A5OpenWrt_LEAN/%E7%AB%9E%E6%96%97%E4%BA%912.0/2023-12-15

openwrt-ipq40xx-generic-p2w_r619ac-128m-squashfs-nand-sysupgrade.bin

输入 ubus call system board , 它发出了:
{
        "kernel": "5.10.199",
        "hostname": "OpenWrt_Main",
        "system": "ARMv7 Processor rev 5 (v7l)",
        "model": "P&W R619AC 128M",
        "board_name": "p2w,r619ac-128m",
        "rootfs_type": "squashfs",
        "release": {
                "distribution": "OpenWrt",
                "version": "SNAPSHOT",
                "target": "ipq40xx/generic",
                "revision": "R23.11.11",
                "description": "OpenWrt "
        }
}

为何我安装 opkg install nginx-ssl 过后它发出了 Segmentation Fault?

然后输入 service nginx start 它发出了：
Segmentation fault
nginx_init: NOT using conf file!
show config to be used by: nginx -T -c ''

System log:
Sat Jun 15 21:20:49 2024 daemon.err nginx_init: 2024/06/15 21:20:49 [crit] 12491#0: pread() "/usr/" failed (21: Is a directory)
Sat Jun 15 21:20:49 2024 daemon.err nginx_init: nginx: configuration file /usr/ test failed
Sat Jun 15 21:20:49 2024 daemon.err nginx_init: NOT using conf file!

输入:
sh -x /etc/init.d/nginx start

Terminal output:
+ START=80
+ USE_PROCD=1
+ G_OPTS='daemon off;'
+ NGINX_UTIL=/usr/bin/nginx-util
+ UCI_CONF_TEMPLATE=/etc/nginx/uci.conf.template
+ LATEST_UCI_CONF_VERSION=1.2
+ /usr/bin/nginx-util get_env
Segmentation fault
+ eval
+ CONF=
+ extra_command relog 'Reopen log files (without reloading)'
/etc/init.d/nginx: line 109: extra_command: not found

一些dmesg output:
[ 9837.187486] ath10k_ahb a000000.wifi: mac flush vdev 0 drop 0 queues 0x1 ar->paused: 0x0  arvif->paused: 0x0
[10253.435853] ath10k_ahb a000000.wifi: mac flush vdev 0 drop 0 queues 0x1 ar->paused: 0x0  arvif->paused: 0x0
[10668.172015] ath10k_ahb a000000.wifi: mac flush vdev 0 drop 0 queues 0x1 ar->paused: 0x0  arvif->paused: 0x0
[10979.860131] ath10k_ahb a000000.wifi: mac flush vdev 0 drop 0 queues 0x1 ar->paused: 0x0  arvif->paused: 0x0
[11436.316268] ath10k_ahb a000000.wifi: mac flush vdev 0 drop 0 queues 0x1 ar->paused: 0x0  arvif->paused: 0x0
[12036.862840] ath10k_ahb a000000.wifi: mac flush vdev 0 drop 0 queues 0x1 ar->paused: 0x0  arvif->paused: 0x0
[12838.507684] ath10k_ahb a000000.wifi: mac flush vdev 0 drop 0 queues 0x1 ar->paused: 0x0  arvif->paused: 0x0
[13998.818975] ath10k_ahb a000000.wifi: mac flush vdev 0 drop 0 queues 0x1 ar->paused: 0x0  arvif->paused: 0x0
[15052.047970] ath10k_ahb a800000.wifi: Invalid VHT mcs 15 peer stats
[15395.919507] ath10k_ahb a000000.wifi: mac flush vdev 0 drop 0 queues 0x1 ar->paused: 0x0  arvif->paused: 0x0
[15935.846067] ath10k_ahb a000000.wifi: mac flush vdev 0 drop 0 queues 0x1 ar->paused: 0x0  arvif->paused: 0x0
[16320.442038] ath10k_ahb a000000.wifi: mac flush vdev 0 drop 0 queues 0x1 ar->paused: 0x0  arvif->paused: 0x0

dmesg 出来的信息跟这个问题无关。

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

ARMv7 Processor rev 5 (v7l) 

### 详细日志

在 详细叙述 已经写了。

yanxin152133, 2024-06-26:
daemon.err nginx_init: 2024/06/15 21:20:49 [crit] 12491#0: pread() "/usr/" failed (21: Is a directory)
/usr/是啥

coolsnowwolf, 2026-09-05:
已核对。该报告使用 2023 年固件与 nginx-util 1.6-r21；日志中的 `/usr/` 配置路径是 nginx-util 未输出有效环境变量后的空配置结果，而 `sh -x /etc/init.d/nginx start` 直接用 sh 执行 rc.common 脚本，因此出现 `extra_command: not found` 不能作为 nginx 服务缺陷证据。当前 packages 已保留 nginx/nginx-util 的配套 UCI 配置生成逻辑，并要求 nginx-ssl-util 与 nginx-util 版本匹配；未能在现行代码复现该历史二进制/混装问题。请使用当前 feed 成套升级并提供 `nginx-util get_env` 输出后再报告，本条关闭为 not planned。

closed 2026-09-05 by coolsnowwolf